springfall2008/batpred #2127: Planned charge rate is low

lewiswatt, 2025-03-19:
**Describe the bug**
The plan that Predbat generates assumes an extremely slow charge rate, regardless of switch.predbat_set_charge_low_power being off or on. 

Additional options changed with no affect;
switch.predbat_carbon_enable 
input_number.predbat_carbon_metric 
switch.predbat_calculate_second_pass
switch.predbat_calculate_tweak_plan
switch.predbat_calculate_import_low_export
switch.predbat_calculate_export_high_import

**Expected behaviour**
Previous plans have charged at full rate from end of day percent > 100% > ~60% > 100% over the 5hr Octopus GO window. This was the behavior I was seeing for weeks, however since the sun's been out, this isn't the case.

**Predbat version**

v8.16.6

**Environment details**

- 13.5kWh AIO (6kW Inverter)
- Standard HAOS installer

**Screenshots**

Planned behaviour

[image]

Actual behaviour

[image]

**Log file**

[predbat_log.txt](https://github.com/user-attachments/files/19339755/predbat_log.txt)

**Predbat debug yaml file**

[predbat_debug_10_50_00.txt](https://github.com/user-attachments/files/19339764/predbat_debug_10_50_00.txt)

lewiswatt, 2025-03-20:
Resolved.

Had Predbat configured to automatically calculate the charge and discharge values for the storage. The charge curve from 89% to 100% was an accurate slowdown from 1.0 to 0.19, however for some reason it was 0.22 for any value below 89%. Switched it to manual values. 

Honestly no idea why this error occured.

gcoan, 2025-03-20:
Good spot, the auto charge curve is good if you do full rate charges and discharges, but if you have slow rate charging turned on it can confuse matters.  Maybe you did some manual slow rate charging?

Its best really to let Predbat work out the curve and then copy it into apps.yaml so it works off that.  Its also quicker as predbat doesn't have to create the curve on each startup