springfall2008/batpred #2130: Compare function giving consistently inaccurate data

discodancerstu, 2025-03-21:
Hi Trefor,

The compare function does not appear to be providing accurate results (for me anyway!); as you can see from the graphs and data below it is consistently under-predicting the cost.

[image]

Our usage is pretty consistent.

[image]

System setup
-GivTCP v3 beta
-PredAI
-Dual Gen3 inverters, each with 9.5kw battery

[apps(yaml).txt](https://github.com/user-attachments/files/19387858/apps.yaml.txt)
[predbat.log](https://github.com/user-attachments/files/19387859/predbat.log)
[cc2581ab_predai_2025-03-21T11-02-21.052Z.log](https://github.com/user-attachments/files/19387872/cc2581ab_predai_2025-03-21T11-02-21.052Z.log)

gcoan, 2025-03-21:
I wouldn't say mine is perfect, but its a reasonably good prediction, and with the heat pump the usage varies day to day anyway:

[image]

Would be checking your losses firstly to check whether they are sensible, and look at load_scaling because if this is set it will put the prediction off.  Have a look at how accurate your solar prediction is and your blend of PV10/PV50

The compare is just running the predbat prediction engine for each tariff so if your prediction factors are off then that would explain the issue

discodancerstu, 2025-03-21:
Thanks, yes I agree the usage varies a little but mineis very consistent.

I'm unsure how to look at losses etc, please can you explain?

Also PredAI is regularly adjusting the forecast to 0.0 on the plan for a whole day, I can't seem to understand why.

springfall2008, 2025-03-22:
Something is wrong with your PredAI data as the predicted load goes down

discodancerstu, 2025-03-22:
> Something is wrong with your PredAI data as the predicted load goes down

Here is my PredAI yaml:

[image]

And this is the sensor it is looking at:

[image]

As I have dual inverters, this sensor only increments (as the GivTCP sensor decreases for some reason), it does not decrease.

So why would the PredAI predicted load reduce?

SwiftRR, 2025-03-23:
I have run predai.
It has generally performed well but I alternate its use with the HA standard historical data.

Like you, my predai.yaml is based on that provided my Trefor.
I notice that your starting entity isn't a direct lift from the GE load energy entity. 
But I have 1 inverter.
I obtained an AI prediction chart similar to yours.
I also obtained a temperature chart using my weather compensation sensor for my HP as the source. 
```
sensors:
  - name: sensor.givtcp_ce2218g352_load_energy_today_kwh
    #subtract: sensor.myenergi_zappi_20464051_energy_used_today
    subtract: sensor.myenergi_zappi_20464051_charge_added_session
    days: 14
    incrementing: True
    reset_daily: True
    interval: 30
    units: kWh
    future_periods: 100
    database: True
    reset_low: 1.0
    reset_high: 2.0
    country: UK
    max_age: 365
  - name: sensor.e3_tcu10_x07_outside_temperature
    days: 14
    interval: 30
    incrementing: False
    units: c
```

Rob