springfall2008/batpred #2146: No export being planned when clipping will happen.

theholyfish, 2025-03-24:
**Describe the bug**
No export being planned when clipping will occur.

**Expected behaviour**
Export to happen automatically via predbat planning. I'm forcing export so I don't clip. [

[predbat-2.log](https://github.com/user-attachments/files/19430985/predbat-2.log)
[image]

[image]
[apps.yaml.txt](https://github.com/user-attachments/files/19431075/apps.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19431076/predbat_debug.yaml.txt)

](url)

**Predbat version**
8.16.7

**Environment details**

- solaredge system. 
- 6kw homehub inverter 
- 10 kW battery. 
- 
- Standard HAOS. Version 15.0
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.
###

gcoan, 2025-03-24:
See #1206

springfall2008, 2025-03-24:
I'm looking at 1206

springfall2008, 2025-04-04:
The version on main with 'calculate_export_oncharge' set on appears to be better but not perfect. I'll continue to review this

[image]

raldred, 2025-04-09:
This approach doesn't work for DC coupled battery systems. 
We need to be exporting _before_ the clipping period, not during.
So that the battery can be charged slowly to soak up as much of the generation as possible, since trickle charging the battery from solar is not subject to the inverter limits, so we can maximise export.

I think the only solution seems to be as suggested in other threads, schedule exporting early morning, and limit charge current to give room in the battery to cover periods that would otherwise result in clipping.

chalfontchubby, 2026-09-17:
It would be interesting to hear whether the recent fixes to the cloud model which attempt to duplicate the on-off nature of the solar helps at all.

raldred, 2026-09-17:
Agreed. Although I think that might difficult to test as the nights draw in.

chalfontchubby, 2026-09-17:
WE have an advanced cloud model that has 6 months drizzle/overcast, 6 months heatwave....

closed 2026-10-28 by springfall2008