springfall2008/batpred #2147: "Set Reserve Min" to 15% predbat constantly loops from default 4% to user set value 15%

dieselee, 2025-03-25:
**Describe the bug**
When setting "Set Reserve Min" to 15% predbat constantly loops the Prog1 on the inverter from default 4% to the 15% i have input in a loop.

**Expected behaviour**
When setting "Set Reserve Min" to 15% use 15% without looping between 4% and 15% constantly.

**Predbat version**
Predbat
Current version: 1.2.4

**Environment details**
- Inverter & Battery Setup
Sunsynk 8KW with Seplos 280L V2 BMS Communicating on CAN to inverter.
- HAOS
Core - 2025.3.4
Supervisor - 2025.03.3
Operating System - 15.0
Frontend - 20250306.0
 - Anything else?
Intergrated with HA via ESPHOME via RS485

**Screenshots**

[image]

**Log file**
2025-03-25 10:20:17.253953: Inverter 0 write_and_poll_value: Wrote 15 to charge_limit, successfully now 15.0
2025-03-25 10:20:12.227430: Inverter 0 Current charge limit is 4% and new target is 15%
2025-03-25 10:15:29.749470: Inverter 0 write_and_poll_value: Wrote 4 to charge_limit, successfully now 4.0
2025-03-25 10:15:24.297320: Inverter 0 Current charge limit is 15% and new target is 4%

**Predbat debug yaml file**
If we need the debug file i can set debug and upload later if required.

WyndStryke, 2025-03-26:
What is your input_number.predbat_best_soc_min set to?  If it is 4% then change it so that it is >= the reserve min (you will need to translate from % into kWh).   Likewise, best_soc_keep should probably be fractionally higher than best_soc_,min.

dieselee, 2025-03-26:
I just did a quick test and set 2.3kw in best soc and 15% in set reserve min and it was the same outcome cycles from 4% - 15% and repeats in a loop

springfall2008, 2025-03-29:
Fix on main, I think the confusion is set reserve min is meant to be the minimum value to configure to and you should use instead battery_min_soc in apps.yaml if you want to change this with an inverter that has no reserve configuration option.
However I've changed the logic to account for this in the defaults

dieselee, 2025-03-30:
Tested and looks like it’s stable now. It certainly isn’t cycling anymore staying steady at 15%

Thank you for the fix.

closed 2025-03-30 by dieselee