springfall2008/batpred #2149: alphaess smile g3 setup -

LeeTaylorX12, 2025-03-26:
Trying to setup using alphaess smile g3 inverter and batteries.

when i switch to control charge and discharge i get the following error in the logs

`AttributeError: 'NoneType' object has no attribute 'split'
2025-03-26 14:18:10.130153: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'split'
2025-03-26 14:18:10.130234: Error: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/config/hass.py", line 190, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1076, in run_time_loop
    raise e
  File "/config/predbat.py", line 1071, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 626, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 350, in execute_plan
    inverter.adjust_force_export(False)
  File "/config/inverter.py", line 1815, in adjust_force_export
    self.adjust_inverter_mode(force_export)
  File "/config/inverter.py", line 1636, in adjust_inverter_mode
    self.write_and_poll_option("inverter_mode", entity_id, new_inverter_mode)
  File "/config/inverter.py", line 1469, in write_and_poll_option
    entity_base = entity_id.split(".")[0]`

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version** - version: v8.16.7 

xxxx

**Environment details**

- alphaess smile g3 2 x 10 batteries
- Home assistant addon
- modbus integration via https://projects.hillviewlodge.ie/alphaess/

`pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  days_previous:
  - 7
  - 14

  forecast_hours: 48

  load_today:
    - sensor.electicity_load_consumption_kwh
  import_today:
      #- sensor.givtcp_{geserial}_import_energy_today_kwh
      - sensor.alphaess_today_s_energy_consumption_from_grid_meter
  export_today:
      #- sensor.givtcp_{geserial}_export_energy_today_kwh
      - sensor.alphaess_today_s_energy_feed_to_grid_meter
  pv_today:
      - sensor.alphaess_today_s_energy_from_pv_meter

  pv_forecast_today: sensor.solcast_pv_forecast_forecast_today
  pv_forecast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow
  pv_forecast_d3: sensor.solcast_pv_forecast_forecast_day_3
  pv_forecast_d4: sensor.solcast_pv_forecast_forecast_day_4
  pv_forecast_d5: sensor.solcast_pv_forecast_forecast_day_5

#Import rates from the Octopus Energy integration, should point to the sensor sensor.octopus_energy_electricity_METER_NUMBER_current_rate
  metric_octopus_import: sensor.octopus_energy_electricity_current_rate
#Export rates from the Octopus Energy integration, should point to the sensor sensor.octopus_energy_electricity_METER_NUMBER_export_current_rate
  metric_octopus_export: sensor.octopus_energy_electricity_export_current_rate
#Gas rates from the Octopus Energy integration, should point to the sensor sensor.octopus_energy_gas_METER_NUMBER_current_rate
#  metric_octopus_gas: sensor.octopus_energy_gas_current_rate - NO NEED FOR GAS

  metric_standing_charge: sensor.octopus_energy_electricity_current_standing_charge
  octopus_free_session: 

  inverter_type: AlphaESS
  inverter:
    name : "AlphaESS Smile G3"
    has_rest_api: False
    has_mqtt_api: False
    
    num_inverters: 1

    output_charge_control: "power"
    timed_charge_current: sensor.alphaess_battery_max_charge_current
    timed_discharge_current: sensor.alphaess_battery_max_discharge_current

# battery_power (%)
    battery_power: sensor.alphaess_soc_battery
# _soc_kwh
    soc_kw: sensor.alphaess_today_s_energy_charge_battery
# battery_capacity_kwh - non modbus integration
    soc_max: sensor.alphaess_installed_capacity

    #When True the inverter timed_charge_current or timed_discharge_current is used to control charging or discharging as/when it starts and stops rather than using a timed method.
    has_charge_enable_time: False
    has_discharge_enable_time: False
    has_target_soc: False
    
    # When True the inverter has a reserve soc setting in apps.yaml called reserve
    has_reserve_soc: true
    reserve: 10

    charge_time_format: "S"
    charge_time_entity_is_option: False
    soc_units: "%"
    num_load_entities: 1
    has_ge_inverter_mode": False

    # When true the inverter has a button press which is needed to update the inverter registers from the Home Assistant values.
    time_button_press: false
    
    clock_time_format: "%Y-%m-%d %H:%M:%S"
    write_and_poll_sleep: 2
    has_time_window: False
    support_charge_freeze: False
    support_discharge_freeze": False

  # Services to control charging/discharging
    charge_start_service:
      service: input_button.press
      entity_id: "input_button.alphaess_helper_charging_now"
      #option: "Charge from Solar Power and Grid"

    #charge_stop_service:
    #  service: select.select_option
    #  entity_id: "select.solaredge_i1_storage_command_mode"
    # option: "Charge from Solar Power"

    discharge_start_service:
      service: input_button.press
      entity_id: "input_button.alphaess_helper_force_discharge"
      #option: "Maximize Self Consumption"

# Alert feeds - customise to your country and the alert types, severity and keep value
  alerts:
    url: "https://feeds.meteoalarm.org/feeds/meteoalarm-legacy-atom-united-kingdom"
    event: "(Amber|Yellow|Orange|Red).*(Wind|Snow|Fog|Rain|Thunderstorm|Avalanche|Frost|Heat|Coastal event|Flood|Forestfire|Ice|Low temperature|Storm|Tornado|Tsunami|Volcano|Wildfire)"
    severity: "Moderate|Severe|Extreme"
    certainty: "Possible|Likely|Expected"
    keep: 40
`

LeeTaylorX12, 2025-03-27:
> can you edit your github issue and put three back quotes (```) on a line both before and after the apps.yaml code as its impossible to read at the moment.

Ooops - sorry.

Reasonably sure the issue relates to battery entities - hopefully used missed something

gcoan, 2025-03-27:
there's a few things that look wrong with your apps.yaml which might not cause the error you are seeing, but won't help:

days_previous the - 7 and - 14 should be indented 2 spaces to the right.  Make sure you have extended your HA history in configuration.yaml if you want to average the same day last week/a week ago as by default only 10 days are retained

import_today/export_today/pv_today entities are all indents too far to the right, they should be 2 spaces then the dash, lining up with load_today

num_inverters is at the moment a sub-entity within the inverter definition, it should be on a line earlier so its not part of the inverter definition

likewise battery_power, soc_kw and soc_max are all sub-children within the inverter definition,  they shouldn't be.  Only the elements in the documentation that define the inverter type should be https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

charge_start_service/discharge start are also indented too far

I think you need to go back to the template apps.yaml and build it up again, ensuring you don't change the indentation of elements.  its quite likely the way you have the elements nested in the yaml is causing some elements to be ignored by predbat

LeeTaylorX12, 2025-04-01:
One step forward and maybe two back.  I have reformatted the apps.yaml (My apps.yaml has 0 errors ) but still getting **Error: Exception raised 'NoneType' object has no attribute 'split'**

'''2025-04-01 14:00:43.334051: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'split'
2025-04-01 14:00:43.334158: Error: 'NoneType' object has no attribute 'split'
Traceback (most recent call last):
  File "/config/hass.py", line 190, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1372, in run_time_loop
    raise e
  File "/config/predbat.py", line 1367, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 632, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 243, in execute_plan
    inverter.disable_charge_window()
  File "/config/inverter.py", line 1967, in disable_charge_window
    self.enable_charge_discharge_with_time_current("charge", False)
  File "/config/inverter.py", line 2052, in enable_charge_discharge_with_time_current
    self.set_current_from_power(direction, 0)
  File "/config/inverter.py", line 2059, in set_current_from_power
    self.write_and_poll_value(f"timed_{direction}_current", self.base.get_arg(f"timed_{direction}_current", indirect=False, index=self.id), new_current, fuzzy=1)
  File "/config/inverter.py", line 1427, in write_and_poll_value
    domain, entity_name = entity_id.split(".")
AttributeError: 'NoneType' object has no attribute 'split'
'''
Not sure if I am missing an option or have conflicting options?  My integration has an 'input_button' press to start charge or discharge which I have set run for 30min to align with the octopus times.  If you press the button again within the 30 mins the timer resets to 30

this is my current apps.yaml
'''
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  days_previous:
  - 7
  - 14

  forecast_hours: 48

  load_today:
    - sensor.electicity_load_consumption_kwh
  import_today:
      #- sensor.givtcp_{geserial}_import_energy_today_kwh
    - sensor.alphaess_today_s_energy_consumption_from_grid_meter
  export_today:
      #- sensor.givtcp_{geserial}_export_energy_today_kwh
    - sensor.alphaess_today_s_energy_feed_to_grid_meter
  pv_today:
    - sensor.ess_solar_production_today

  pv_forecast_today: sensor.solcast_pv_forecast_forecast_today
  pv_forecast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow
  pv_forecast_d3: sensor.solcast_pv_forecast_forecast_day_3
  pv_forecast_d4: sensor.solcast_pv_forecast_forecast_day_4
  pv_forecast_d5: sensor.solcast_pv_forecast_forecast_day_5

[... 67 lines cut ...]
    has_discharge_enable_time: False
    has_target_soc: False
# When True the inverter has a reserve soc setting in apps.yaml called reserve
    has_reserve_soc: true
    reserve: 10
    charge_time_format: "S"
    charge_time_entity_is_option: False
    soc_units: "%"
    num_load_entities: 1
    has_ge_inverter_mode: False

    timed_charge_current: sensor.alphaess_battery_max_charge_current
    timed_discharge_current: sensor.alphaess_battery_max_discharge_current

# When true the inverter has a button press which is needed to update the inverter registers from the Home Assistant values.
    time_button_press: false
    clock_time_format: "%Y-%m-%d %H:%M:%S"
    write_and_poll_sleep: 2
    has_time_window: False
    support_charge_freeze: False
    support_discharge_freeze: False

'''

[predbat.log](https://github.com/user-attachments/files/19551403/predbat.log)

gcoan, 2025-04-01:
can you re-paste your apps.yaml with the right quote symbol at the start and end please its ``` I think you have put '''

Can't read it at the moment !

I've not come across anyone with an inverter that needs a button press to activate the charge, that might be new ground you are treading....

LeeTaylorX12, 2025-04-01:
it should be fine?  i noticed my mistake and corrected

doing the button press isn't a massive issue - if i have a something to trigger I can do the automation to do the actual press.

gcoan, 2025-04-02:
I haven't dug through the predbat code to work out precisely where it is crashing, but from the traceback and looking at your apps.yaml, and looking at the modbus integration page, I am wondering whether you have/are trying to set things up slightly wrong

On the modbus integration I see that there are start and end charge and discharge time slots (indeed slot 1 and slot 2), this is very common to the way predbat works and would I think be a lot easier than trying to get the 30 minute button to work.

Also in your apps.yaml you have a charge start service and a discharge start service, but no stop services.  I think this could well be why predbat is crashing because it is in 'disable charge window' which (without checking the code) is to stop charging, but you haven't defined a stop charge service - so hence the 'nontype' error

LeeTaylorX12, 2025-04-03:
HI,

I don't believe its anything to do with the services.  Where its crashing is in write_and_poll_switch function which makes me think its more a clash (or missing) of options I have selected.  To try and work this out; i have reset the inverter definition back to the template with the exception of reserve.  Still getting the error.

What I have noticed is start/end times for charge and discharge aren't being updated in home assistant (i.e. dash values are being updated per plan but HA has not changed).  And when i look at the dash a number of 'predicted power' values are zero e.g. grid_power_best.  Probably completely unrelated.

As an aside binary_sensor.predbat_charging and binary_sensor.predbat_exporting are these setup to a specific inventor install or should they always be available?

On the plus side; the button press is working! :)  Maybe too early to call victory but the charge service is calling the button press as I had hoped; i.e. it looks to be pressing the button inline with the half hour slots.  In a 3hr charge window now so have to see if it continues to press. Have to wait a few hours to see if the export button press is working also.

i note your comment on the button press; not sure why that approach is so radically different to the switch shown in the examples?  If doing this by syncing start/stop times is the preferred way I am happy to progress that.  However as per earlier comment need some guidance on predbat ha time entities not being updated.

gcoan, 2025-04-03:
> What I have noticed is start/end times for charge and discharge aren't being updated in home assistant (i.e. dash values are being updated per plan but HA has not changed). 

Can you share the YAML of your charge/discharge times and screenshots of what isn't being updated and I can try to help

> And when i look at the dash a number of 'predicted power' values are zero e.g. grid_power_best. Probably completely unrelated.

Possibly, but grid_power_best for example is always zero per the documentation https://springfall2008.github.io/batpred/output-data/#best-prediction-data

There are lots of sensors and some of them only use the attributes.

> As an aside binary_sensor.predbat_charging and binary_sensor.predbat_exporting are these setup to a specific inventor install or should they always be available?

Yes these are generic to predbat for all inverters and should be available and work.  Some other inverter scripts use these to automate charge and discharge activity.

> On the plus side; the button press is working! :) Maybe too early to call victory but the charge service is calling the button press as I had hoped; i.e. it looks to be pressing the button inline [...]

Woo !

> i note your comment on the button press; not sure why that approach is so radically different to the switch shown in the examples? If doing this by syncing start/stop times is the preferred way I am [...]

Its just most people's inverters use start/end times rather than push buttons.  If if its now working for you then great, it was just a suggestion to go down that route if you couldn't get it to work

chalfontchubby, 2026-10-18:
_Posted by Claude on behalf of @chalfontchubby._

Closing as stale (no activity since April 2025). The crash in the traceback came from `inverter_mode` having no entity configured, which is a setup issue rather than a code bug. Predbat now has a built-in AlphaESS Cloud integration with its own template, so you no longer need a hand-built Modbus setup: see [AlphaESS Cloud](https://springfall2008.github.io/batpred/inverter-setup/#alphaess-cloud). If you still have problems on a current release, please open a new issue and attach a full log, `apps.yaml` and `predbat_debug.yaml`.

closed 2026-10-18 by chalfontchubby