springfall2008/batpred #2153: predbat_manual_api doesn't allow inverter_limit_charge to be changed once set via the API

gcoan, 2025-03-29:
**Describe the bug**
I am trying to use predbat_manual_api to set inverter_limit_charge to avoid the clipping issues I get with #1206 

When I set select.predbat_manual_api to inverter_limit_charge(0)=700 it correctly sets the inverter charge limit to 700W.  When I set the second inverter charge rate to 0 with select.predbat_manual_api=inverter_limit_charge(0)=0 it also works.

Predbat manual api is currently set to

+inverter_limit_charge(0)=700,inverter_limit_charge(1)=0

So far so good.

But subsequently I want to increase the inverter charge rate to 1700W, so set select.predbat_manual_api=inverter_limit_charge(0)=1700 I am expecting the manual API to now be set to:

+inverter_limit_charge(0)=1700,inverter_limit_charge(1)=0

 but predbat appends the new command as an additional manual overwrite command and doesn't change the original limit.

So inverter_limit_charge now says:

+inverter_limit_charge(0)=700,inverter_limit_charge(1)=0,inverter_limit_charge(0)=1700

the 3rd entry should have overwritten the first, instead it is appended, and predbat ignores it.

[image]

If I select [inverter_limit_charge(0)=700] from the select, to remove the earlier charge limit of 700, this is correctly deleted, leaving:

+inverter_limit_charge(1)=0,inverter_limit_charge(0)=1700

And the inverter 0 limit is applied correctly

**Expected behaviour**
Should be able to independently and repeatedly set each inverter_limit_charge as per the documentation without having to delete the prior setting first

**Predbat version**
8.16.7

**Environment details**
HAOS 14.2
HA 2025.3.5
2x Gen 1 hybrids each with their own battery

**Log file**

[predbat.log](https://github.com/user-attachments/files/19516844/predbat.log)

springfall2008, 2025-03-30:
Fix on main

gcoan, 2025-03-30:
Thanks this is now working fine. 

I am seeing issues with changes being reflected in the manual api but not getting changed on the inverter. I'll do some more testing of this and raise a separate issue when I can see it clearly

gcoan, 2025-03-31:
Had to reopen it as I think the version on main has a step backwards

I had had a problem with inverter 1 (H) charge rate being wrong, but couldn't see why that was.

Just now, H was on 678W:

[image]

despite manual API saying it should be 0:

[image]

deleted the manual API entry completely, both inverters went back to 2600

re-set inv 0 (G) to 700 and 1 (H) to 0 and they changed to 666 and 678W again so looks like the iteration of inverter number isn't right as inv 1 is picking up what inv 0 is set to

deleted inv 1 override:

[image]

but inv 1 still unchanged:

[image]

givtcp log shows the same, inv1 is taking the same setting as inv0 even though it never has been set to 700 (the 72 was me manipulating the control directly)
[image]

reverting back to 8.17.1

springfall2008, 2025-04-04:
Fix on 'main' for testing

closed 2025-04-05 by gcoan