springfall2008/batpred #2155: FR: Set battery_pause_mode intelligently based on inverter_limit_charge/discharge

gcoan, 2025-03-29:
**Is your feature request related to a problem? Please describe.**
I am trying to use predbat_manual_api to automate my battery charging whilst reducing solar and grid clipping #1206 I get on sunny days

I can set inverter_limit_charge to 0 via the predbat_manual_api but this does not totally prevent charging of the battery:

[image]

The battery continues to trickle charge at about 150-200W

**Describe the solution you'd like**
I would like predbat to intelligently set battery_pause_mode when inverter_limit_charge and inverter_limit_discharge is set via the manual API.  Its similar to how predbat already sets the mode when doing a force charge or export:

If inverter_limit_charge = 0 AND inverter_limit_discharge = 0 then set battery_pause_mode=Both

else if inverter_limit_charge = 0 then set battery_pause_mode=PauseCharge

else if inverter_limit_discharge = 0 then set battery_pause_mode=PauseDischarge

else set battery_pause_mode=Disabled

This will enable predbat to properly stop the battery charging and discharging when inverter_limit_charge/discharge is used to control charge and discharge rates

gcoan, 2026-08-20:
@chalfontchubby Brilliant work with clearing the backlog of predbat issues

One of the ones you closed, I made mention of this FR that I raised, which I self assigned and then never implemented....

If you are looking for something to do, this would be very useful, every night I set one of my inverters to 0 discharge rate (via predbat manual api) but it continues to trickle discharge because battery pause mode is not set to pause the discharge:

[image]

since raising the issue I have introduced the 'Demand (freeze export)' mode which I use overnight to stop the batteries cross charging.  So the condition logic I expressed above would need to take account of this, i.e. in this case because Freeze Export is set, and battery discharge rate is zero, battery pause mode would need to be set to Pause Both.

Note that if predbat decides it wants to freeze export or freeze charge then it sets the pause mode correctly (as shown in the screenshot, all inverters are on Pause Charge), its when predbat sets the discharge or charge rate (e.g via a manual override) that the pause mode doesn't get set

chalfontchubby, 2026-08-20:
I note you have a lot of issues open - I think 53 more than a year old. I’m trying not to close your issues since you clearly know what you are doing - but it you are able to have a sweep and we can close / raise priority of etc as needed. There is a lot of good stuff in there - but a lot has also changed and triage is taking a while.

chalfontchubby, 2026-08-20:
Had a proper look at this (including your follow-up about Freeze Export) - here's where it landed, design-wise.

**The gap is real and exactly where you said**: `inverter_limit_charge`/`inverter_limit_discharge` (`inverter.py`) only cap the rate ceiling fed into `adjust_charge_rate`/`adjust_discharge_rate` - they never touch `battery_pause_mode`. `adjust_pause_mode()` (`inverter.py`) is the only thing that genuinely stops a `has_timed_pause` inverter; a near-zero rate still lets a few hundred W trickle through. Your automatic Freeze Export path already gets this right (`execute.py`, the "Demand mode, turning on Freeze Export" block) - it calls `adjust_pause_mode(pause_charge=True)` directly. The manual-limit path just never makes an equivalent call at all.

**The wrinkle, and why it's not a one-line fix**: `execute.py`'s pause-mode handling is a small state machine across several blocks (Freeze Export, car-charging hold, iBoost hold, and a final reset-to-Disabled), gated by a `resetPause` flag that lets only the *first* block each cycle actually call `adjust_pause_mode()` - there's no merging of `pause_charge`/`pause_discharge` across blocks today, so whichever fires first wins outright. Your Freeze Export block only ever passes `pause_charge=True`. To get "Pause Both" when Freeze Export is active *and* you've manually zeroed the discharge rate, the fix has to change that block to check the manual discharge limit and pass both flags together - not just add a second independent call site that would get silently skipped by the same gate.

So the shape of the fix is: derive a manual-limit-driven pause requirement (your original charge=0/discharge=0/both logic), then have it combine with - not race against - the existing Freeze Export block's own pause_charge call, so the two express one final call instead of two competing ones.

Not picking this up right now - flagging it as understood and scoped rather than parking it silently. Happy to build it properly if it becomes a priority, or if you fancy a crack at it yourself with the pointers above (sounds like you've been thinking about giving Claude Code a go - this one's a good size for it: contained to `execute.py`/`inverter.py`, a clear existing pattern to follow, and `tests/test_execute.py`'s `set_freeze_export_during_demand` cases are a solid template to extend). You're also honestly the best-placed person to dogfood it either way - you're the one already running manual API limits alongside Freeze Export overnight and watching the trickle live, so you'd know immediately whether a fix actually kills it rather than just looking right in a test. Either way it's on the radar.

chalfontchubby, 2026-08-20:
You have the advantage as well of a system on which you can dogfood it.

gcoan, 2026-08-20:
> note you have a lot of issues open - I think 53 more than a year old

I have been through my open issues.  Not sure it was 53, but was certainly a lot.  Some of them were because I had converted a discussion into an issue for someone else, so they were down as being raised by me even though nothing to do with me...

Anyway, have been through, closed a number that were dormant or I believe already fixed.

Now down to (just!) 28 created by me.

If you wanted some to look at:
3460 still a bug
2726 is a useful enhancement, causes people random issues when Octopus retire product codes
2578 is an annoying one still occurring (observed it today)
2687 is a requested feature for a number of people to stop repeated charge/discharge cycles overnight

I'll see what I can do with progressing this specific issue myself

gcoan, 2026-10-26:
Am working on this now