springfall2008/batpred #2164: Validation errors in apps.yaml

jrviz, 2025-03-29:
**Describe the bug**
I am getting some validation errors in apps.yaml for entries I have not chnaged for a long time. Errors saying items such as a time must contain a dot. This is a recent behaviour.

**Expected behaviour**
Not throwing errors. 

**Predbat version**
8.17.1

**Environment details**

- Solaredge Inverter and batteries
- Standard HAOS 

**Log file**
2025-03-29 17:47:09.155576: Warn: Validation of apps.yaml found configuration item 'charge_start_time' element 00:00:00 is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155669: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element 00:01:00 is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155718: Warn: Validation of apps.yaml found configuration item 'charge_limit' element 100 is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155749: Warn: Validation of apps.yaml found configuration item 'scheduled_charge_enable' element off is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155777: Warn: Validation of apps.yaml found configuration item 'scheduled_discharge_enable' element off is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155806: Warn: Validation of apps.yaml found configuration item 'discharge_start_time' element 00:00:00 is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.155835: Warn: Validation of apps.yaml found configuration item 'discharge_end_time' element 00:01:00 is not a valid entity_id (must contain a dot)
2025-03-29 17:47:09.156262: Error: Validation of apps.yaml found 7 configuration errors

springfall2008, 2025-03-29:
The validation is now, can you share what is in apps.yaml for these settings? These are meant to be entities that Predbat can modify and not fixed values.

jrviz, 2025-03-29:
I changed to a fixed value as I had an entity not available for the charge limit so I chnaged to a fixed value to see if that was an issue, here is my original app.yaml entries for the items throwing an error.

  charge_start_time:
    -  "00:00:00"
  charge_end_time:
    -  "00:01:00"
  charge_limit:
    -  number.solaredge_i1_ac_charge_limit
  scheduled_charge_enable:
    -  off
  scheduled_discharge_enable:
    -  off
  discharge_start_time:
    -  "00:00:00"
  discharge_end_time:
    -  "00:01:00"

The solaredge modbus multi integration is currently showing the entity unavailable which I need to look into. Hence the log item is now:

: Warn: Validation of apps.yaml found configuration item 'charge_limit' element number.solaredge_i1_ac_charge_limit is not a float got unavailable

Which makes sense. 

The time entries and enable entries are unchanged. Should these be entities? I have not really used those and I think those came in from the inverter template.

Didn't quite get what you meant by "validation is now". Did you mean new? If so, that would be why errors are being thrown and in settings I have not used before.

gcoan, 2025-03-29:
> Didn't quite get what you meant by "validation is now". Did you mean new? If so, that would be why errors are being thrown and in settings I have not used before.

yes typo from Trefor,  in 8.17.x Predbat introduces validation of apps.yaml

your apps.yaml entries are not what Predbat is expecting.  As Trefor said, charge_start_time, etc should point to entities that predbat can change with the time it wants to start charging at

see the documentation https://springfall2008.github.io/batpred/apps-yaml/#schedule

So these are valid errors

jrviz, 2025-03-30:
Thanks, that helps. I don't think I need these for Solaredge. I have never used them and they are clearly in error. They came in from the template:

[https://github.com/springfall2008/batpred/blob/main/templates/solaredge.yaml ]

I'm happy to contribute to updating this template to change this and add some information on adapting the sensors for multiple batteries.

springfall2008, 2025-03-30:
You can just delete the lines if they are not needed

gcoan, 2025-03-30:
> I'm happy to contribute to updating this template to change this and add some information on adapting the sensors for multiple batteries.

please do.  Either here and I can merge it in or create a PR

jrviz, 2025-03-30:
I had those entries from the template in the documentation. I have deleted them and all seems well.

@buttash @cyberkryten @leacho73 @ashpork @richadlam @piebeard Did you have the above entries in apps.yaml from the solaredge template which now throws errors? Have you removed them? If that works for all we can change the documentation.

buttash, 2025-03-30:
I’ve saw those errors in my apps.yaml. I removed the entries and all seems ok.

leacho73, 2025-03-31:
Same - I've commented out all 7 items and no further errors.

jrviz, 2025-03-31:
I think what is shown as `number.solaredge_i1_ac_charge_limit` in the template should possibly have been `sensor.solaredge_i1_ac_charge_limit`

However, it seems that none of them are needed and I'll propose a change to remove them from the documented template

jrviz, 2025-03-31:
> > I'm happy to contribute to updating this template to change this and add some information on adapting the sensors for multiple batteries.
> 
> please do. Either here and I can merge it in or create a PR

These lines should be removed from the solaredge template as part of the documentation:
`
  charge_start_time:
    - "00:00:00"
  charge_end_time:
    - "00:01:00"
  charge_limit:
    - number.solaredge_i1_ac_charge_limit
  scheduled_charge_enable:
    - off
  scheduled_discharge_enable:
    - off
  discharge_start_time:
    - "00:00:00"
  discharge_end_time:
    - "00:01:00"
`

[https://github.com/springfall2008/batpred/blob/main/templates/solaredge.yaml](url)

gcoan, 2025-04-01:
I have updated the template, please check you are happy with the change, it'll be in the next PR I do 

https://github.com/gcoan/batpred/blob/main/templates/solaredge.yaml

jrviz, 2025-04-01:
Yes, leave it there. It looks like only needs a value if there is a charging policy which is not disabled or Always Allowed.  Fixed Energy Limit is kWh while Percent of Production is a %age :-) I guess only a %age maps to the Predbat charge_limit

gcoan, 2025-04-01:
Done, let me know if there's anything else on this, or I can close the issue

jrviz, 2025-04-01:
Thanks. Nothing more.

closed 2025-04-01 by gcoan