springfall2008/batpred #2168: Validation error with predheat load forecast

s0129, 2025-03-30:
I have just updated to v8.17.1 from v8.16.7 and I have found it flags an error with the predheat load_forecast in  apps.yaml. I have predheat installed as part of batpred.

My apps.yaml contains this line.

```
  load_forecast:
   - predheat.heat_energy$external
```

```
Warn: Validation of apps.yaml found configuration item 'load_forecast' element predheat.heat_energy$external is not a dict got [{'last_updated': '2025-03-30T16:55:00+0100', 'energy': 78.37}, {'last_updated': '2025-03-30T17:05:00+0100', 'energy': 78.37}, ....
```

The documentation says:
load_forecast - this should be configured to point to a sensor and attribute. The attribute must be in either
- The format of 'last_updated' timestamp and 'energy' in incrementing kWh.
- The format of a dictionary of timestamps and energy data in incremental KWh.

Is this a bug in the new apps.yaml validation?

springfall2008, 2025-03-30:
No it's just a bug in the validation code, please ignore it for now

springfall2008, 2025-04-04:
Fix on main

closed 2025-04-12 by springfall2008