springfall2008/batpred #2171: Home Assistant API

alz41, 2025-03-30:
**Describe the bug**
Predbat fails to connect to Home Assistant via inbuilt Api 

**Predbat version**

version v8.17.1

**Environment details**

- Standard HAOS installer
- Home Assistant Verson
Core 2025.4.0b9
Supervisor 2025.03.4
Operating System 15.1.rc1
Frontend 20250328.0

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
025-03-30 21:43:58.637692: Stopping Predbat
Shutdown, sleeping 20 seconds before restarting
[21:44:19] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: 096b6c8f2a3fc9eea22581cd191ea95d208e68ac1cf9f7198c699d6ef8b3c8d2301bf4ac42cfb41994366f7e36a6c8051954cf8b74cdce4d
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.17.1
**** Starting Standalone Predbat ****
2025-03-30 21:44:20.726912: Loading apps.yaml
2025-03-30 21:44:20.790029: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/
2025-03-30 21:44:20.790177: Warn: Unable to connect directly to Home Assistant at http://supervisor/core, please check your configuration of ha_url/ha_key
2025-03-30 21:44:20.790213: Error: Exception raised 
2025-03-30 21:44:20.790873: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-30 21:44:20.790986: Error: set_state_wrapper - No HA interface available
2025-03-30 21:44:20.791003: Info: record_status Error: Exception raised 
2025-03-30 21:44:20.791018: Error: Exception raised 
2025-03-30 21:44:20.791072: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1182, in initialize
    raise e
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-30 21:44:20.791134: Error: set_state_wrapper - No HA interface available
2025-03-30 21:44:20.791148: Info: record_status Error: Exception raised 
Error: Failed to initialize predbat 
Traceback (most recent call last):
  File "/config/hass.py", line 49, in main
    p_han.initialize()
  File "/config/predbat.py", line 1226, in initialize
    raise e
  File "/config/predbat.py", line 1182, in initialize
    raise e
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-30 21:44:20.791312: Stopping Predbat
Shutdown, sleeping 20 seconds before restarting
[21:44:41] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is:

gcoan, 2025-03-30:
Per the error message in the logfile, what do you have set to ha_url and ha_key in apps.yaml?

alz41, 2025-03-30:
@gcoan both are commented out as they have always been

gcoan, 2025-03-30:
> @gcoan both are commented out as they have always been

Very strange. It's the same as mine. 

You running the predbat add on?

Try restarting the add on?

alz41, 2025-03-30:
> > [@gcoan](https://github.com/gcoan) both are commented out as they have always been
> 
> Very strange. It's the same as mine.
> 
> You running the predbat add on?
> 
> Try restarting the add on?

Yes the Add on version, Already tried restarting also tried uninstalling predbat rebooted Ha and reinstall predbat

gcoan, 2025-03-30:
You can safely ignore that warning, there's a minor issue with working out the slug id when ha_url is defined (which I have a fix for)

But doesn't explain why it crashed without ha_url defined 

Just the confirm your setup:

- standard HAOS
- predbat add on 
- no ha_url or_ha_key defined

alz41, 2025-03-30:
> You can safely ignore that warning, there's a minor issue with working out the slug id when ha_url is defined (which I have a fix for)
> 
> But doesn't explain why it crashed without ha_url defined
> 
> Just the confirm your setup:
> 
> * standard HAOS
> * predbat add on
> * no ha_url or_ha_key defined

Correct

pacemaker82, 2025-03-31:
I have this same error since trying to update to latest predbat version https://github.com/springfall2008/batpred/issues/2177 - installation is stuck and predbat down. Restarted HA hard reboot, add on too - nothing works.

gittyup444, 2025-03-31:
Thanks - I suffered from this today too. I was worried it was something I had broken myself!

Thanks for the above fix, can confirm it fixes it for me too.

Does anyone know if this will be a permanent solution (well at least 10 years until the token expires...) or is it something that could cause a conflict in future releases?

pacemaker82, 2025-03-31:
Just adding that this fixed it for me too. I am not sure how the installation got unstuck though, just seemed to happen randomly over an hour later.

AberDino, 2025-03-31:
Same symptoms here, and the workaround got Predbat back up and running again. Thanks!

LeeFarm11, 2025-03-31:
> Same symptons here too. Above fix - creating Long-Lived Access Token has me up and running again.

This was with v8.17.2 and I use fixed IP address.

pacemaker82, 2025-03-31:
I now have this error after adding the ha_url and ha_key:

`aiohttp.client_exceptions.WSServerHandshakeError: 404, message='Invalid response status', url=URL('http://homeassistant.local:8123//api/websocket')`

lxalln, 2025-03-31:
Rebooted HA this afternoon, and now Predbat is dead 😔

gcoan, 2025-03-31:
I'll post a change on this thread to ha.py which might fix the problem without needing ha_url and ha_key to be defined.  Be good if some can test it.

Its weird because I am running the latest Predbat on the latest HAOS and HA version and its working fine without these having to be defined.

But clearly lots of people are getting issues.

alz41, 2025-03-31:
> I'll post a change on this thread to ha.py which might fix the problem without needing ha_url and ha_key to be defined. Be good if some can test it.
>
> Its weird because I am running the latest [...]

Will give it a try.  I Presume predbat auto update need disabling?

gcoan, 2025-03-31:
> Will give it a try. I Presume predbat auto update need disabling?

No, you'll just be replacing one file in the predbat addon folder.  Predbat won't see it as a new version, and when you next do a predbat upgrade it'll get overwritten

I am stil on HA supervisor 2025.03.3, at least one person has reported the issue on supervisor 2025.03.4

Looking at the supervisor release it mentions a breaking change on the websocket handling https://github.com/home-assistant/supervisor/pull/5788

The failures seem to be occurring at the point when Predbat initiates the websocket error so I wonder if this is somehow related?

alz41, 2025-03-31:
> > Will give it a try. I Presume predbat auto update need disabling?
> 
> No, you'll just be replacing one file in the predbat addon folder. Predbat won't see it as a new version, and when you next do a predbat upgrade it'll get overwritten
> 
> I am stil on HA supervisor 2025.03.3, at least one person has reported the issue on supervisor 2025.03.4
> 
> Looking at the supervisor release it mentions a breaking change on the websocket handling [home-assistant/supervisor#5788](https://github.com/home-assistant/supervisor/pull/5788)
> 
> The failures seem to be occurring at the point when Predbat initiates the websocket error so I wonder if this is somehow related?

Dont know if it will help, If i comment out ha_path & ha_key then restart predbat. HA Supervisor log shows the following errors

2025-03-31 18:51:58.967 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on 6adb4f0d/aarch64-addon-predbat with version 1.2.4
2025-03-31 18:51:59.893 ERROR (MainThread) [aiohttp.server] Error handling request from 172.30.33.6
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/integrations/aiohttp.py", line 157, in sentry_app_handle
    reraise(*_capture_exception())
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/utils.py", line 1718, in reraise
    raise value
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/integrations/aiohttp.py", line 140, in sentry_app_handle
    response = await old_handle(self, request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 198, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 212, in system_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 229, in token_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 293, in core_proxy
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/proxy.py", line 120, in api
    path = request.match_info["path"]
           ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'path'
2025-03-31 18:52:21.809 ERROR (MainThread) [aiohttp.server] Error handling request from 172.30.33.6
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/integrations/aiohttp.py", line 157, in sentry_app_handle
    reraise(*_capture_exception())
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/utils.py", line 1718, in reraise
    raise value
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/integrations/aiohttp.py", line 140, in sentry_app_handle
    response = await old_handle(self, request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 198, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
[... 128 lines cut ...]
  File "/usr/local/lib/python3.13/site-packages/sentry_sdk/integrations/aiohttp.py", line 140, in sentry_app_handle
    response = await old_handle(self, request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 198, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 212, in system_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 229, in token_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 293, in core_proxy
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/proxy.py", line 120, in api
    path = request.match_info["path"]
           ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'path'

gpsmith, 2025-03-31:
Mine did the same this afternoon. I did the token thing, and it's come up. But back to complaining about some app.yaml issues like it did the other day despite nothing changing - but they stopped appearing after a day! 

```
2025-03-31 19:01:52.068413: Warn: Validation of apps.yaml found configuration item 'charge_start_time' element 23:30:00 is not a valid entity_id (must contain a dot)
2025-03-31 19:01:52.068704: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element 05:30:00 is not a valid entity_id (must contain a dot)
2025-03-31 19:01:52.068976: Warn: Validation of apps.yaml found configuration item 'charge_limit' element sensor.solaredge_i1_b1_maximum_energy can not be modified
2025-03-31 19:01:52.069575: Warn: Validation of apps.yaml found configuration item 'scheduled_charge_enable' element False is not a valid entity_id (must be a string)
2025-03-31 19:01:52.069949: Warn: Validation of apps.yaml found configuration item 'scheduled_discharge_enable' element False is not a valid entity_id (must be a string)
2025-03-31 19:01:52.070258: Warn: Validation of apps.yaml found configuration item 'discharge_start_time' element 00:00:00 is not a valid entity_id (must contain a dot)
2025-03-31 19:01:52.070647: Warn: Validation of apps.yaml found configuration item 'discharge_end_time' element 00:01:00 is not a valid entity_id (must contain a dot)
2025-03-31 19:01:52.073225: Error: Validation of apps.yaml found 7 configuration errors
```

gcoan, 2025-03-31:
Just been through the same journey myself.  Earlier I was on supervisor 2025.03.3,  I saw on the upgrade list there was a supervisor upgrade available to 2025.03.4,  HA automatically installed it, and its broken Predbat.

The fix I tried to ha.py doesn't work unfortunately, you are welcome to try it, its at https://drive.google.com/file/d/10bd2zR1EpperUBUyh3ANrb2K5qQ_96Z6/view?usp=sharing

Copy this to your Predbat folder.  The original change was to eliminate the warning message you get when running predbat in a docker container that it can't talk to the HA Supervisor (because there isn't one),  and for those who run predbat in HAOS and do define ha_url and ha_key, then there was the same warning message occurring.  This latter part is fixed now so there is value in the change, but it doesn't solve the original problem reported.

I believe the issue is with HAOS supervisor 2025.03.4, but so far haven't found anybody else reporting an issue.

What Predbat does is it opens a connection to http://supervisor/core/api to check that it can talk to HA, and this is returning error code 500:

```
2025-03-31 19:37:49.033828: Checking Predbat can talk to HA
2025-03-31 19:37:49.059617: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/
```

This API *should* work, it is a published API that the HA supervisor proxies API calls to HA https://developers.home-assistant.io/docs/api/supervisor/endpoints/#core

The workaround is to define ha_url so that predbat talks directly to the HA and not go via the supervisor, e.g. mine is 

ha_url: 'http://192.168.1.69:8123'

You can use homeassistant.local:8123 but using the IP avoids a DNS lookup for every API call.

rkflyer, 2025-03-31:
Same here, adding url and long lived token  got me back up and running again

Died with same log errors as per OP, occured when I updated predbat this afternoon to 8.17.2

Running HAOS in a VM on Proxmox, predbat as an Add on

Core 2025.3.4
Supervisor 2025.03.4
Operating System 15.0
Frontend 20250306.0

gcoan, 2025-03-31:
I don't think the issue is Predbat 8.17.2, I'm running 8.17.1 and it occurred when I updated ha.py to try to fix the issue for others.
So would seem Predbat starts failing once it goes through the startup routine, e.g. after an upgrade, reboot, etc.

There is suggestion it is linked to HAOS 15 as well as Supervisor 2025.03.4 because there are others on the GivEnergy community group https://community.givenergy.cloud/d/5411-second-year-live-on-predbat/1263 who are on Supervisor 2025.03.4 but HAOS 14.2 and not seeing the issue.

If it is a core/supervisor problem then I'd hope it gets reported and fixed soon as it will be affecting other people that use the core API

stevedundee2, 2025-03-31:
I have this problem too. I'd just installed haos 15.1. So I reverted to 15.0 (ha os boot-slot A) but the problem remained and so I am back on 15.1.

For now I'm leaving predbat off. My battery is 83% and a sunny day tomorrow so I don't need it charging from the grid by predbat tonight.

I'm on the latest predbat. I think that updated earlier today but it may have been yesterday. The original haos update to 15.0 was a day or two earlier. There were no problems with that.

Other add ons (eg givtcp, mqtt) seem to be ok.

rkflyer, 2025-03-31:
Just installed 8.17.3 and all is back to normal

Not sure what time it was released this evening but have removed the url and token and it is again connecting via supervisor/core.

Which is the prefered method? Via Core or Token?  I presume Core as it was before?

gcoan, 2025-03-31:
> I have this problem too. I'd just installed haos 15.1. So I reverted to 15.0 (ha os boot-slot A) but the problem remained and so I am back on 15.1.
> 
> I'm on the latest predbat. I think that updated earlier today but it may have been yesterday. The original haos update to 15.0 was a day or two ago. Update to supervisor 2025.3.4 was sometime before that. No problems until today's update to 15.1.

the problem seems to be HAOS supervisor 2025.03.4 and HAOS 15+.  Once predbat reboots it will fail to start due to HA supervisor changes

Upgrade to Predbat 18.7.3 or define ha_URL and ha_key in apps.yaml and it should all work again

LeeFarm11, 2025-04-01:
> Upgrade to Predbat 18.7.3 or define ha_URL and ha_key in apps.yaml and it should all work again

Upgrade to 18.7.3 happened at 23.45 last night. This morning I have commented the ha_URL and ha_key, predbat has restarted and all is OK :)

Boffinboy, 2025-04-04:
Finding I’m having the same error.

Boffinboy, 2025-04-04:
I wasn’t able to but have managed to now. The issue seems to have been fixed. Unfortunately having my usual other issue with REST not working, but I think that’s to do with my Unifi networking gear. I think I might give up and go to web based using Givenergy cloud.