springfall2008/batpred #2172: apps.yaml validation error: car_charging_battery_size

DJBenson, 2025-03-30:
The validator is saying that my value for car_charging_battery_size should be a sensor yet the documentation says;

> car_charging_battery_size - Set this value in apps.yaml to the car's battery size in kWh which must be entered with one decimal place, e.g. 50.0. If not set, Predbat defaults to 100.0kWh. This will be used to predict when to stop car charging.

and my value is set to 77.4 (fixed value, not a sensor, as no such sensor exists).

chalfontchubby, 2026-08-28:
No further activity since our notice, so closing this as stale. If this is still a problem for you, feel free to reopen with a fresh reproduction case and we'll take another look.

cjt105, 2026-09-09:
This issue is still ongoing - whilst the error has no impact, it'd be ideal to fix it so the app can load without error

springfall2008, 2026-09-09:
Automated first-pass triage (a maintainer will review before any action is taken).

**Classification:** bug (validator rejects a fixed value the docs still describe) · **Priority:** low (status-only impact, easy workaround)

I reproduced this against current main (v8.55.1-9-gd48cb223) with a probe test in the validator's own test module (`tests/test_validate_config.py`, run via `tools/triage_test.sh`):

- `car_charging_battery_size: 77.4` (scalar float) → `'car_charging_battery_size' is not of type 'sensor' value was 77.4` — exactly the reported error
- `car_charging_battery_size: 100` (scalar int) → same error
- `car_charging_battery_size: [77.4]` (one-element list) → passes
- `car_charging_battery_size: sensor.xxx` (entity returning 77.4) → passes

**Root cause:** in `validate_config()` the `sensor`/`sensor_list` branch (apps/predbat/predbat.py:1645-1653) only wraps a scalar value into a list when it is a string (predbat.py:1648). A numeric scalar is neither a string nor a list, so it never reaches the "fixed integer/float values are allowed" checks inside the list loop (predbat.py:1667-1672) and falls through to the generic "not of type 'sensor'" error (predbat.py:1744). The sibling `float` branch wraps numeric scalars correctly (predbat.py:1530) — the sensor branch was written to mirror the `string` branch instead, which is why only strings get through. A likely fix is to wrap any non-list scalar there; the fixed-float/int paths would then validate it as documented.

This isn't specific to this one key — every item typed `"sensor"` in APPS_SCHEMA (13 of them, e.g. `pv_forecast_today`, `car_charging_soc`, `car_charging_limit`) rejects a fixed numeric scalar the same way.

**Impact:** status-only — it keeps the `config_valid` sensor at 0 (apps/predbat/predbat.py:711) and logs "Error: Validation of apps.yaml found N configuration errors"; the value itself is read fine, so car charging is unaffected (matching what @cjt105 observed).

**Workaround today:** write the value as a one-element list, e.g.
```yaml
car_charging_battery_size:
  - 77.4
```
That form passes validation (and is what the example in docs/car-charging.md has shown since 2026-03-27). Note the prose on that page still says "must be entered with one decimal place, e.g. 50.0", which invites the failing scalar form — worth aligning with the code either way (fix the branch or the prose).

No duplicate found; possibly related to #1314 (the umbrella "add sensor & data type validation" request), but this is a concrete defect rather than a feature request. Bug-report template items (version/debug yaml/log) weren't attached, but I was able to reproduce without them, so nothing further is needed from the reporters.

chalfontchubby, 2026-09-09:
For the bot - several of those entities marked as sensor really should not be fixed values. E.g. pv_forecast_today, 
Car_charging_soc. 
Some are borderline - car_charging_limit.
Allowing scalars across the board would invite further issues being raised. 
Rather than allow scalars for any sensor, better in this case to fix the documentation to match the workaround.

closed 2026-09-10 by springfall2008