springfall2008/batpred #2173: apps.yaml validation errors - inverter_limit_charge, inverter_limit_discharge and export_limit

DJBenson, 2025-03-30:
More of a question than a bug but these three entities are thowing validation errors as the validator expects an integer but I'm passing input_number sensors so I can dynamically change these values.

apps.yaml simply references the sensor - do I need to wrap the sensor in something within apps.yaml to parse it as an integer or should I just ignore these.

Example from apps.yaml;

```
  inverter_limit_charge:
    - input_number.givenergy_aio_max_charge_rate

  inverter_limit_discharge:
    - input_number.givenergy_aio_max_discharge_rate
```

gcoan, 2025-03-30:
I'm surprised this works for you, last time I tried using an entity in apps.yaml for the charge rate it didn't work

Am trying to do similar with predbat manual api to overwrite inverter_limit_charge programatically

DJBenson, 2025-03-31:
It was something which @springfall2008 advised me to do a looooonnnng time ago in the early days of predbat back when I was having issues with my AIO crashing when charging at full power (in fact I think these were added at my request). I don't routinely use it now so it might be broken. Will try it later.

DJBenson, 2025-04-01:
@gcoan FYI it's still working;

[image]

That sensor is ONLY read by Predbat, it in turn sets the max charge rate of the GivTCP entity - also works for the other two I mentioned.

WyndStryke, 2025-04-01:
I might try this myself.  In my own homebrew scheduler, I reduce the overnight charge/discharge rate during heatwaves & when a lot of PV is expected, just to keep the battery cells at a reasonable temperature & to reduce waste heat (it gets very hot indoors and I don't want to add to it unnecessarily).  Would be good to be able to do the same thing in Predbat too.

closed 2026-08-20 by chalfontchubby