springfall2008/batpred #2175: PredBat doesn't start since GE Inverter Upgrade

mb695, 2025-03-31:
**Describe the bug**
Predate doesn't start

**Predbat version**
8.16.7

**Environment details**
- GivEnergy Hybrid Gen 2 - Firmware D0.920-A0.920
- HomeAssistant OS

**Log file**
Startup
Predbat files are installed correctly for version v8.16.7
**** Starting Standalone Predbat ****
2025-03-31 08:22:36.282516: Loading apps.yaml
2025-03-31 08:22:36.480729: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/
2025-03-31 08:22:36.480948: Error: Exception raised 'HAInterface' object has no attribute 'api_errors'
2025-03-31 08:22:36.483752: Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/requests/models.py", line 974, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 5 (char 4)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/ha.py", line 539, in api_call
    data = response.json()
  File "/usr/local/lib/python3.10/dist-packages/requests/models.py", line 978, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 5 (char 4)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/predbat.py", line 884, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 87, in __init__
    check = self.api_call("/api/")
  File "/config/ha.py", line 543, in api_call
    self.api_errors += 1
AttributeError: 'HAInterface' object has no attribute 'api_errors'
2025-03-31 08:22:36.484610: Error: set_state_wrapper - No HA interface available
2025-03-31 08:22:36.484768: Info: record_status Error: Exception raised 'HAInterface' object has no attribute 'api_errors'
Error: Failed to initialize predbat 'HAInterface' object has no attribute 'api_errors'
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/requests/models.py", line 974, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 5 (char 4)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/ha.py", line 539, in api_call
    data = response.json()
  File "/usr/local/lib/python3.10/dist-packages/requests/models.py", line 978, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 5 (char 4)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/hass.py", line 49, in main
    p_han.initialize()
  File "/config/predbat.py", line 932, in initialize
    raise e
  File "/config/predbat.py", line 884, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 87, in __init__
    check = self.api_call("/api/")
  File "/config/ha.py", line 543, in api_call
    self.api_errors += 1
AttributeError: 'HAInterface' object has no attribute 'api_errors'
2025-03-31 08:22:36.485948: Stopping Predbat

gcoan, 2025-03-31:
#2171 has reported a similar problem

as a workaround can you try setting ha_url and ha_key in apps.yaml - details in the apps.yaml part of the documentation

mb695, 2025-03-31:
Thanks - that works

springfall2008, 2025-04-03:
Now if you update Predbat it should fix the real issue and you don't need the api key anymore (although it can be a good idea to have in some cases)

closed 2025-04-03 by springfall2008