springfall2008/batpred #2178: Solcast PV API calls happening too often?

pacemaker82, 2025-03-31:
**Describe the bug**
You can see the value from the PV forecast from SOLCAST are updated multiple times in quick succession:

[image]

**Expected behaviour**
Call API once every few hours

**Predbat version**

8.17.2

pacemaker82, 2025-04-01:
API limit for solcast is also set to 0, with 0 used. Taken from `sensor.predbat_pv_today`

[image]

gcoan, 2025-04-01:
Just to confirm, solcast_poll_hours is set in apps.yaml? 
https://springfall2008.github.io/batpred/apps-yaml/#solcast-solar-forecast

pacemaker82, 2025-04-01:
Yes, I have:

```
  solcast_host: 'https://api.solcast.com.au/'
  solcast_api_key: 'blah'
  solcast_poll_hours: 6
```

pacemaker82, 2025-04-01:
Also, not to derail, but the documentation @gcoan seems to be slightly misleading:

> pv_forecast_today - Entity name for today's Solcast forecast
pv_forecast_tomorrow - Entity name for tomorrow's Solcast's forecast
pv_forecast_d3 - Entity name for Solcast's forecast for day 3
pv_forecast_d4 - Entity name for Solcast's forecast for day 4 (also d5, d6 & d7 are supported, but not that useful)

However `pv_forecast_d2` is actually day 3 here (depending on if you consider today as day 1) and so on - otherwise `pv_forecast_d2` would reflect same values as "tomorrow"

My current estimates:

`pv_forecast_today` = 33.25 kWh
`pv_forecast_tomorrow` = 32.56 kWh
`pv_forecast_d2` = 28.14 kWh
`pv_forecast_d3` = 26.51 kWh

gcoan, 2025-04-01:
> Also, not to derail, but the documentation @gcoan seems to be slightly misleading:

I'll update the documentation.  Predbat follows the same naming convention as the solcast integration (and possibly the Solcast API's).  Whether you agree with that convention and how its named, debateable !

gcoan, 2025-04-01:
@pacemaker82 I have changed the apps.yaml documentation to now say:

> - **pv_forecast_today** - Entity name for today's Solcast forecast
> - **pv_forecast_tomorrow** - Entity name for tomorrow's Solcast's forecast
> - **pv_forecast_d3** - Entity name for Solcast's forecast for the day after tomorrow
> - **pv_forecast_d4** - Entity name for Solcast's forecast for two days after tomorrow
> 
> Sensors for d5, d6 & d7 are supported, but not that useful so are not pre-defined in the template.

Which aligns with the terminology used in the output data documentation.  Actually realised that Solcast uses today/tomorrow/d3/d4 whereas the Predbat PV forecast output entities are today/tomorrow/d2/d3, so they're not consistent.

Anyway, I have ensured they are at least described more clearly

pacemaker82, 2025-04-01:
If `pv_forecast_d3` is "Entity name for Solcast's forecast for the day after tomorrow" then what is `pv_forecast_d2` if `pv_forecast_tomorrow` has a different value?

pacemaker82, 2025-04-01:
Confusing answer. Predbat clearly provides a `d2` entity whether that's aligned with the spec above or not, and as the docs are for predbat, it should explain what `d2` is ...

gcoan, 2025-04-01:
I'll explain my logic again although I am getting a bit annoyed by this as I don't think there is anything wrong

Predbat has INPUT ENTITY configuration in apps.yaml - i.e. what entities it takes IN for Predbat's processing, the givtcp template looks like this:

```
  # Set these to match solcast sensor names if not using the cloud interface
  # The regular expression (re:) makes the solcast bit optional
  # If these don't match find your own names in Home Assistant
  pv_forecast_today: re:(sensor.(solcast_|)(pv_forecast_|)forecast_today)
  pv_forecast_tomorrow: re:(sensor.(solcast_|)(pv_forecast_|)forecast_tomorrow)
  pv_forecast_d3: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3))
  pv_forecast_d4: re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4))
```

There is no d2 entry in the apps.yaml template and (per the above) no Solcast entity named d2.

The predbat apps.yaml documentation describes what the INPUT configuration items are.  There is no mention of d2.

In solcast.py Predbat gets the Solcast forecast from the INPUT entities defined in apps.yaml.

The appropriate code in function fetch_pv_forecast is:
```
            self.log("Using Solcast integration from inside HA for solar forecast")

            # Fetch data from each sensor
            for argname in ["pv_forecast_today", "pv_forecast_tomorrow", "pv_forecast_d3", "pv_forecast_d4"]:
```

Predbat OUTPUTS a series of entities including the Solar prediction.  I can't easily find the code for this but these are described in the Output Data part of the documentation:

> The following sensors give the forecast Solar data from Solcast. Predbat populates these sensors irrespective of whether you are using the Predbat direct Solcast or Solcast integration method to get your Solar forecast, but if you are using the Solcast integration then the Predbat sensors mirror the similarly named Solcast integration sensors so could be disabled if you so wish.
>
> - sensor.predbat_pv_today - Tracks the PV forecast in kWh for today, attributes give the remaining amount today and the half-hourly data
> - sensor.predbat_pv_tomorrow - Tracks the PV forecast in kWh for tomorrow, attributes give the remaining amount today and the half-hourly data
> - sensor.predbat_pv_d2 - Similar to the above, but tracking the PV forecast for the day after tomorrow
sensor.predbat_pv_d3 - PV forecast for two days after tomorrow
> - sensor.predbat_pv_forecast_h0 - Tracks the PV 'power now' forecast in Watts, attributes give the 10% and 90% power now forecast

What issue is there with this?

pacemaker82, 2025-04-01:
ah my confusion was INPUT vs OUTPUT, I thought the first description was describing the entities outputted by predbat. My mistake

closed 2025-04-02 by pacemaker82