springfall2008/batpred #2180: Predbat Fails to start after  18.7.2 Update

Sycamore967, 2025-03-31:
**Describe the bug**
Upgraded and it looped for over an hour - eventually did something but now says its started but I have nothing.

**Expected behaviour**
Upgrade and restart

v18.7.2

Log has this entry over and over

[20:39:43] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: 2f435bd005476c557a9516423257bbb4734f9c61ff2d20c0dfb149cd6edf3e0afb6a1081b368aa7a8332fd55ca8123ebfd346e553db69728
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.17.2
**** Starting Standalone Predbat ****
2025-03-31 20:39:45.598249: Loading apps.yaml
2025-03-31 20:39:45.763690: Warn: Failed to decode response <Response [500]> from http://supervisor/core/api/
2025-03-31 20:39:45.764255: Warn: Unable to connect directly to Home Assistant at http://supervisor/core, please check your configuration of ha_url/ha_key
2025-03-31 20:39:45.764392: Error: Exception raised 
2025-03-31 20:39:45.765950: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-31 20:39:45.766400: Error: set_state_wrapper - No HA interface available
2025-03-31 20:39:45.766507: Info: record_status Error: Exception raised 
2025-03-31 20:39:45.766569: Error: Exception raised 
2025-03-31 20:39:45.766860: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1182, in initialize
    raise e
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-31 20:39:45.767231: Error: set_state_wrapper - No HA interface available
2025-03-31 20:39:45.767345: Info: record_status Error: Exception raised 
Error: Failed to initialize predbat 
Traceback (most recent call last):
  File "/config/hass.py", line 49, in main
    p_han.initialize()
  File "/config/predbat.py", line 1226, in initialize
    raise e
  File "/config/predbat.py", line 1182, in initialize
    raise e
  File "/config/predbat.py", line 1177, in initialize
    self.ha_interface = HAInterface(self)
  File "/config/ha.py", line 92, in __init__
    raise ValueError
ValueError
2025-03-31 20:39:45.768159: Stopping Predbat
Shutdown, sleeping 20 seconds before restarting

craig-nicolson, 2025-03-31:
Exact same here. I think it's from the HAOS update I just did.

I am seeing this in the Supervisor logs so not sure if it's a predbat or HAOS issue?

```
2025-03-31 21:02:34.979 ERROR (MainThread) [aiohttp.server] Error handling request from ***.***.***.***
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 198, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 212, in system_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 229, in token_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 293, in core_proxy
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/proxy.py", line 120, in api
    path = request.match_info["path"]
           ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'path'
```

bfayers, 2025-03-31:
After a few restarts I also suddenly had issues with it connecting to HASS. Fix was to manually configure ha_url and ha_key for now. Same issue in #2171 I think.

Sycamore967, 2025-03-31:
I have a system back up from this morning for my installed and running version of v8.16.7 but I've never run a restore! I'm assuming I can just restore PredBat and get back to a running version?

gcoan, 2025-03-31:
> I have a system back up from this morning for my installed and running version of v8.16.7 but I've never run a restore! I'm assuming I can just restore PredBat and get back to a running version?

yes you can just restore back to an older predbat version and it should run fine

but defining a ha_key and ha_url in apps.yaml as described in #2171 seems to be a viable workaround for everyone so far

gcoan, 2025-03-31:
> I am seeing this in the Supervisor logs so not sure if it's a predbat or HAOS issue?
>
> ```
> 2025-03-31 21:02:34.979 ERROR (MainThread) [aiohttp.server] Error handling request from ***.***.***.*** [...]

I see the same in my supervisor logs, it starts appearing immediately after the supervisor upgraded itself:

```
2025-03-31 18:57:23.642 INFO (MainThread) [supervisor.supervisor] Update Supervisor to version 2025.03.4
2025-03-31 18:57:23.643 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/amd64-hassio-supervisor with tag 2025.03.4.
```

then 2 minutes later the same stream of errors, tracing back to:

```
  File "/usr/src/supervisor/supervisor/api/proxy.py", line 120, in api
    path = request.match_info["path"]
           ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'path'
```

this would concur with the error 500 that predbat gets when it tries to make a call via the supervisor through to HA core

gcoan, 2025-03-31:
Trefor has fixed it in the above patch 8.17.3

For those that care, the line in ha.py to change is:

check = self.api_call("/api/")

to

check = self.api_call("/api/services")

And predbat will now work without ha_url and ha_key being defined.

For those that are able to, my updated ha.py (which includes Trefor's fix) is https://drive.google.com/file/d/10bd2zR1EpperUBUyh3ANrb2K5qQ_96Z6/view?usp=drive_link be worth others on docker or HAOS confirming it works for them as I will PR this in soon

stevedundee2, 2025-03-31:
The patch works for me. Thanks!

nargotik, 2025-03-31:
Update doesn't apear to me on HA using addon

DJBenson, 2025-04-01:
I couldn't get the Docker addon to pull in the fix so I did the following;

* Ensured Predbat container was running
* SSH'd into my HA machine
* Grabbed the has of the Predbat container with`docker ps | grep predbat`
* Connected to the container terminal with `docker exec -it PREDBATHASH /bin/bash`
* Installed nano in the container with `apt install -y nano`
* Opened the file at the correct line with `nano +88 /config/ha.py`
* Amended line 88 from /api to /api/services
* Saved/exited nano with ctrl+o and ctrl+x
* Restarted the add-on

Sycamore967, 2025-04-01:
Editing ha.py didn't work for me, but waiting for v8.17.3 to be available did.  Installed it from HA and PredBat restarted ok, with all setting back how they were.

nargotik, 2025-04-01:
@DJBenson thank you it worked flawless

closed 2025-04-03 by springfall2008