springfall2008/batpred #2183: Fails to Install predbat addon or appdaemon-predbat addon

bmadgett, 2025-04-01:
**Describe the bug**
Installing the Predbat addon via Addon store gives error message:

Failed to install add-on

The command '/bin/bash -o pipefail -c apt-get install -y python3-pip' returned a non-zero code: 100

Additionally, trying to install the appdaemon-predbat addon gives error message:

Failed to install add-on

The command '/bin/bash -o pipefail -c apk add --no-cache --virtual .build-dependencies build-base=0.5-r3 python3-dev=3.11.9-r0 && echo "Step1"' returned a non-zero code: 60

Previously (until last night) I had a working appdaemon-predbat addon. Noticed it was "dead" late last night, tried reboot, tried updating HA to latest version, now the appdaemon-predbat addon is no longer listed as installed, and get the above error messages when try to reinstall or switch to Predbat addon.

I note today's changes with ha_key and ha_url and supervisor things but since my addon is no longer even listed as installed, I can't restart it to test adding these items, and I can't reinstall due to above errors.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

Was on appdaemon-predbat addon and 'latest' version, can't easily see precisely which now it's listed as uninstalled for some reason.

**Environment details**

- GE AIO battery via GivTCP
- HAOS on RaspPi

**Screenshots**
na

**Log file**
na

**Predbat debug yaml file**
na

bmadgett, 2025-04-01:
Have tried removing the repos from the Addon Store, rebooting, then re-adding and trying install again - no change.

bmadgett, 2025-04-01:
The actual errors in the system log when trying to install the addons look like:

ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/6adb4f0d_predbat/install - The command '/bin/bash -o pipefail -c apt-get install -y python3-pip' returned a non-zero code: 100

and

Failed to to call /addons/46f69597_appdaemon-predbat/install - The command '/bin/bash -o pipefail -c apk add --no-cache --virtual .build-dependencies         build-base=0.5-r3         python3-dev=3.11.9-r0     && echo "Step1"' returned a non-zero code: 60

I've tried reverting HAOS to 14.2 (no better), running 'ha su repair' and 'ha core rebuild', and booting in safe mode. No changes to this failure to install the addons.

bmadgett, 2025-04-02:
This morning I've completely reset the HA instance using the "ha os datadisk wipe" command so it resets HA completely, then restored from a backup to get my config back (note, predbat wasn't listed in the addons available to backup or restore since it "thinks" it's not installed). Then tried installing predbat addon again and it still gives the same failure error during the install. Running out of ideas now.

gcoan, 2025-04-02:
I don't have any ideas on the predbat addon, sorry, beyond my expertise on docker install

Assume that google search for the error code/description doesn't find anything?

Have you tried installing appdaemon and running predbat under that? It should still work

bmadgett, 2025-04-02:
Funny that, it's exactly what I've just got working - appdaemon standalone addon and then predback via HACS inside appdaemon. That appears to be working (literally just going a minute ago). Still not happy that there's obviously something broken with installing it the "proper" way, at least for my setup. But at least it's working again for now, thanks.

bmadgett, 2025-04-02:
> Appdaemon-predbat is no longer supported.
> 
> Have you tried the predbat-addon?
> […](#)

Yes, that was what I tried first, the error message I get is quoted up top. It's similar to but not identical to the one I get trying to install appdaemon-predbat.

Any ideas? If you don't know how to fix the do you have any pointers where I should be looking to "clean up" any detritus that might be left from failed broken installs of the add-ons that might be causing these errors? (Full wipe and restore from backup excluding these add-ons hasn't worked)

springfall2008, 2025-04-02:
What system do you have, perhaps is a CPU/OS that wasn't supported by the versions of tools?

springfall2008, 2025-04-02:
Google says the error with predbat addon maybe just a network issue, did you try a second time later?

bmadgett, 2025-04-02:
Thanks for thinking about this. Standard RPi4 with HAOS, the usual setup. Networking was fine as far as I can tell, but I've tried numerous times now anyway so I'm pretty confident it's not that. Done a heap of googling about the error. 

I think the sequence of events, with the benefit of hindsight and new knowledge gained since, was this: 

11:30pm I notice I can't set a manual charge slot via my dashboard, no response from predbat, "oh no it's broken". Try restarting predbat, no better, I know I'll do an update, might as well do the OS while I'm at it since it looks like there's one available. Mistake, in hindsight. 

I think predbat had stopped due to the (unknown to me at the time) ha_url/key thing you need to add to the config now. But I didn't know this. (I've since added them to the config). And then somehow updating the OS removed the predbat (and appdaemon) add-ons from the installed list and they get these install errors if I try to reinstall.

springfall2008, 2025-04-03:
Can you try this:

1. Restart Home Assistant so its from a clean boot
2. Install Predbat add on again (not the appdaemon one)
3. Check the supervisor log and report what it does (assuming it fails)

bmadgett, 2025-04-03:
Ok... so, I just restarted HA again. And installed predbat addon. And it's worked, this time, of course!  At least, the install hasn't failed, I've not moved the config from the existing appdaemon install over yet to test it's actually "working". I've not changed anything else as far as I know since the last time this same install was failing. So now I don't know what to say! Speaking as a software engineer myself, isn't this the most frustrating bug? When it suddenly fixes itself and you're left with the nagging "but what if it comes back again and I never knew what I did to fix it"... Anyway, thanks for your time. Predbat is invaluable btw :)

closed 2025-04-03 by gcoan