springfall2008/batpred #2184: Documentation - LuxPower

brickatius, 2025-04-01:
I've tried to create 'input_text.xxxxxxx' helpers for charge and discharge start and end times - for example as in the 'luxpower.yaml' template as follows:

charge_start_time:
    - input_text.lux_batpred_charge_start

but unless I've got something completely wrong these 'input_text'  don't seem to be writable by Predbat and an error occurs. Wouldn't it be better to use the actual predbat sensor names like this?

charge_start_time:
    - sensor.predbat_LuxPower_0_charge_start_time

You can then write a Home Assistant automation to target the LuxPower inverter start and end time sensors, as follows, whenever the predbat sensor changes. 

alias: Charge Time Change - Start
description: ""
triggers:
  - entity_id:
      - sensor.predbat_LuxPower_0_charge_start_time
    trigger: state
conditions: []
actions:
  - metadata: {}
    data:
      time: "{{ states('sensor.predbat_LuxPower_0_charge_start_time')}}"
    target:
      entity_id: time.lux_ac_charge_start1
    enabled: true
    action: time.set_value
mode: single

It works a treat and you can see the new charge and discharge times in the LuxPower app and online monitoring.

gcoan, 2025-04-01:
There's certainly something incomplete with the documentation because the template defines using input_text.lux_batpred_discharge_start etc as you say, but then there's no guidance as to what then needs to happen when these input_text's are written to.

Predbat should be able to write to input_text's

But looking at your automation, did you try setting the time.lux_ac_charge_start1 entity id's directly in apps.yaml so predbat can write directly to that entity

Per the documentation, predbat can write to time entities for charge/discharge start/end time https://springfall2008.github.io/batpred/apps-yaml/#schedule although this is a fairly new feature and maybe wasn't there when the luxpower template was developed.

Can you give it a go and if it works, let me know what changes are required to the documentation

Also, if there are other automations required or changes to the template, please flag it up

brickatius, 2025-04-01:
> gcoan

Hi Geoffrey, 

I’ll certainly give it a go. 
I did have Predbat set up very briefly on another machine which for reasons unrelated to Predbat died without backup. I bought another one and started from scratch. As it was only a couple of weeks ago I do remember having this:

charge_start_time:
   - time.lux_ac_charge_start1

I don’t remember the entities in the integration and consequently the values in the monitoring app or online being changed to reflect the values in Predbat. As a result I just set the AC Charge Start1 to 00:00 and the End1 to 23:59 so that when the Charge Enable switch was turned on it was within a charge window. Likewise for discharge settings. I’ll try with the code above in apps.yaml as above tomorrow in order to confirm whether it works or not. 

LuxPower also has start and end times for what it calls ‘Charge First / Charge Priority  which I believe Predbat calls ’discharge_freeze_service’. As there are no times for this in Predbat I set the start and end times to 00:00 and 23:59 respectively. I think that also needs to be reflected in the documentation. 

Kind regards. 
Richard Wall

brickatius, 2025-04-02:
Hi Geoffrey, 

 Further to yesterday - apps.yaml settings.
 With this from the LuxPower.yaml template

charge_start_time:
    - input_text.lux_batpred_charge_start
  charge_end_time:
    - input_text.lux_batpred_charge_end

2025-04-02 08:37:42.170735: Warn: Validation of apps.yaml found configuration item 'charge_start_time' element input_text.lux_batpred_charge_start can not be modified
2025-04-02 08:37:42.170828: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element input_text.lux_batpred_charge_end can not be modified
2025-04-02 08:37:42.171511: Error: Validation of apps.yaml found 2 configuration errors

And with this, using the LuxPower entities. 
charge_start_time:
    - time.lux_ac_charge_start1
  charge_end_time:
    - time.lux_ac_charge_end1

2025-04-02 08:49:02.385591: Warn: Validation of apps.yaml found configuration item 'charge_start_time' element time.lux_ac_charge_start1 can not be modified
2025-04-02 08:49:02.385698: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element time.lux_ac_charge_end1 can not be modified
2025-04-02 08:49:02.386402: Error: Validation of apps.yaml found 2 configuration errors

The Web Ui shows this on both occasions 
⚠charge_start_time	
- sensor.predbat_LuxPower_0_charge_start_time = 23:59:00
⚠charge_end_time	
- sensor.predbat_LuxPower_0_charge_end_time = 23:59:00

But.with this and the automations I mentioned yesterday Predbat is happy.
charge_start_time:
  - sensor.predbat_LuxPower_0_charge_start_time
charge_end_time:
  - sensor.predbat_LuxPower_0_charge_end_time

The same also applies to the forced discharge start and end times. 

It would appear that Predbat is not picking up the entities and is always producing the Predbat ones. Is this a bug or should the LuxPower.yaml just be modified to show the predbat sensors?

More: LuxPower does not have any SOC max or state of charge entities in kWh. I created these as Template helpers. 

soc_max_kWh (Maximum capacity = battery capacity in ah x battery voltage)
{{ states("sensor.lux_battery_capacity_ah") |float * states("sensor.lux_battery_voltage_live") | float  / 1000}}

soc_kWh (Maximum capacity as above x battery charge as a %)
{{ states("sensor.lux_battery_capacity_ah") |float * states("sensor.lux_battery_voltage_live") | float * states("sensor.lux_battery") | float / 100000}}

I also created these populated with values from the Inverter Datasheet and then modified with observations from the Inverter flow chart. In my case 4000w and accurate to within 50w from observation.

charge_rate:
    - input_number.lux_target_charge_current_w
  discharge_rate:
    - input_number.lux_target_discharge_current_w

Hope this helps.

Kind regards. 
Richard

brickatius, 2025-04-02:
Hi Geoffrey, 

If the apps,yaml configuration was working as expected the entries would look like this:-

charge_start_time:
time.lux_ac_charge_start1
charge_end_time:
time.lux_ac_charge_end1
discharge_start_time:
  - time.lux_force_charge_start1
discharge_end_time:
  - time.lux_force_charge_end1

Regards, 
Richard

gcoan, 2025-04-02:
Hi Richard, thanks for all your many messages !

In the luxpower template it has:

```
  inverter_type: LuxPower
  inverter:
...
    has_charge_enable_time: False
    has_discharge_enable_time: False
```

so this is telling predbat that the luxpower inverter doesn't have a charge and discharge time sensors which I think is why predbat is creating its own entities (e.g. predbat_LuxPower_0_charge_start_time)  and setting them to dummy values

but as identified there are time.lux_ac... sensors so I think we should try using them if we can.  if it doesn't work then we'll go back to the automations you have working and just update the documentation to reflect

so in apps.yaml can you set:

```
    has_charge_enable_time: True
    has_discharge_enable_time: True
```
and

```
  charge_start_time:
    - time.lux_ac_charge_start1
  charge_end_time: 
    - time.lux_ac_charge_end1
discharge_start_time: 
  - time.lux_force_charge_start1
discharge_end_time:
  - time.lux_force_charge_end1
```

and lets see if that works.  If you get apps.yaml validation errors and it works, then great, its just a false validation error.

gcoan, 2025-04-02:
> > and lets see if that works. If you get apps.yaml validation errors and it works, then great, its just a false validation error.
> > 
> This is essentially how I have mine set up and I still get the validation error. However the start and end times do appear to be getting set by predbat at the appropriate times.
> 
> I have just watched them update and the times set match a planned charge in the plan due to start at 11:30pm. I've also checked the LuxPower Device Monitor and can see those times have also been set on the device itself.
> 
> So despite the validation error it does appear to be working.

OK good news, I thought they should work.  We can sort out the false validation error separately.

If @brickatius confirms he is happy with the changes I will update the template - plus anything else that you both spot as wrong

> I should also add that I didn't previously have the soc_kw and soc_max using helpers, mine were hard coded, so I thought it a good idea to create those helpers. The updated docs say to set them as Power and Measurement, however as they are kWh and not kW they should be created as Energy and Total.

Sorry, late at night, corrected now

raldred, 2025-04-02:
Sorry I authored the original template, I may have missed providing some entities and automations in the docs, sorry about that. The lux inverter is expecting a different format for start and end times then predbat provides. 

You need input text entities for the charge/discharge start and end times
[image]

Automation to set lux inverter times when predbat changes the inputs
```
alias: Lux Set Charge Start (predbat)
description: ""
mode: single
triggers:
  - entity_id:
      - input_text.lux_batpred_charge_start
    trigger: state
conditions: []
actions:
  - action: time.set_value
    metadata: {}
    data:
      time: "{{ states('input_text.lux_batpred_charge_start')}}"
    target:
      entity_id: time.lux_ac_charge_start1

```

I'll look more in the morning

raldred, 2025-04-02:
You also need an automation to set the discharge SOC on the inverter because it has independent controls. 

My setup works perfectly with these entries and automations the template changes aren't necessary.

```
alias: Lux Set Discharge Target SOC (predbat)
description: ""
triggers:
  - entity_id:
      - binary_sensor.predbat_exporting
      - number.lux_ac_battery_charge_level
    trigger: state
conditions:
  - condition: state
    entity_id: binary_sensor.predbat_exporting
    state: "on"
actions:
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ states('number.lux_ac_battery_charge_level') }}"
    target:
      entity_id: number.lux_forced_discharge_battery_level
mode: single

```

owinwood, 2025-04-03:
> Sorry I authored the original template, I may have missed providing some entities and automations in the docs, sorry about that. The lux inverter is expecting a different format for start and end [...]

I thought the helpers weren't required anymore with LuxPython_DEV adding native time settings? I don't recall what version it was added in, but I believe it's the reason I'm now able to use the time entities directly.

From the readme
"@maegibbons has invested significant time refining time settings. In LUXPython for HA 2023.06, we now have native time settings, so any custom helpers/automations can be removed. We won’t remove the old method immediately, but it will be phased out. Going forward, just add time entities like time.lux_ac_charge_start1."

gcoan, 2025-04-03:
Predbat has added support for setting time entities directly fairly recently (which wasn't there before), so my take on what you LuxPower users are reporting is that these automations are now no longer required

raldred, 2025-04-03:
Ahh yes, that's right @gcoan. I've just tested this now. Thanks

As @owinwood suggests, this now works without the need for the text_input entities. Works as expected.

```
    has_charge_enable_time: True
    has_discharge_enable_time: True
```

```
  charge_start_time:
    - time.lux_ac_charge_start1
  charge_end_time:
    - time.lux_ac_charge_end1
  discharge_start_time:
    - time.lux_force_discharge_start1
  discharge_end_time:
    - time.lux_force_discharge_end1
```

There's warnings and validation errors, but they seem to work

[image]

```
11237 | 2025-04-03 11:04:40.832840: Inverter 0 Turning on scheduled export
-- | --
11236 | 2025-04-03 11:04:40.832726: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
11235 | 2025-04-03 11:04:40.832465: Switch switch.lux_force_discharge_enable is now on
11234 | 2025-04-03 11:04:28.696807: Inverter 0 Wrote discharge_end_time to 11:31:00 successfully
11233 | 2025-04-03 11:04:17.566132: Inverter 0 Set new export end time to 11:31:00 was 00:00:00
11232 | 2025-04-03 11:04:17.565935: Inverter 0 Wrote discharge_start_time to 11:00:00 successfully
11231 | 2025-04-03 11:04:06.531690: Inverter 0 set new export start time to 11:00:00
11230 | 2025-04-03 11:04:06.531561: Inverter 0 write_and_poll_value: Wrote 11:31:00 to idle_start_time, successfully now 11:31:00
11229 | 2025-04-03 11:03:56.507240: Inverter 0 set new demand (idle) start time to 11:31:00 was 05:30:00
11228 | 2025-04-03 11:03:56.507173: Adjust demand (idle) time computed is 11:31:00-23:59:00
11227 | 2025-04-03 11:03:56.507018: Adjust idle time, charge 00:00:00-00:00:00 discharge 11:00:00-11:31:00
11226 | 2025-04-03 11:03:56.506984: Inverter 0 Adjust force export to True, change times from 00:00:00 - 00:00:00 to 11:00:00 - 11:31:00

```

gcoan, 2025-04-03:
I've updated the documentation to now include the LuxPower time entities in luxpower.yaml and added @raldred 's discharge SoC automation:
https://github.com/gcoan/batpred/blob/main/docs/inverter-setup.md#lux-power
https://github.com/gcoan/batpred/blob/main/templates/luxpower.yaml

Other than the spurious warnings about apps.yaml time.luxxxx sensors, is this all OK now?

raldred, 2025-04-03:
Looks good, I dont think the custom entity for `soc_kw` is required.
Since we're using `soc_percent`

https://springfall2008.github.io/batpred/apps-yaml/
> Battery SoC
soc_kw - Entity name of the battery SOC in kWh, should be the inverter one not an individual battery
or
> soc_percent Entity name of the battery SOC in percent of the maximum battery size, should be the inverter one not an individual battery

brickatius, 2025-04-03:
I did the SOC in kWh because Predbat couldn’t compute the charge and discharge curves without it.

gcoan, 2025-04-03:
> Since we're using soc_percent

yes I think you are correct, either soc_kw or soc_percent (fairly new option in predbat) should be specified, doesn't need both

@brickatius you happy I remove this?

and I will add the template sensor to correct for not reporting 100% SoC

gcoan, 2025-04-03:
> I did the SOC in kWh because Predbat couldn’t compute the charge and discharge curves without it.

Thanks for the clarification 

Let's raise this as a separate issue because both should be able to be used I'd suggest

brickatius, 2025-04-03:
Hi all, 
Yes, apart from the validation warnings all seems to be working. I had had this set all the time

has_charge_enable_time: True
has_discharge_enable_time: True

I must admit that when I saw these warnings I didn't actually check whether things were working or not because the Web UI was still showing the sensor.predbatxxxxxx entities. I don't know what has changed but the Web UI is now showing the LuxPower time entities and these are now reflected in the LuxPower app and online settings.

One other thing to mention is that as Predbat does not have start time or end time settings for 'Charge First' - Charge Freeze/Hold in Predbat if you want to use 

charge_freeze_service:

I set the Charge First Start Time 1 = 00:00 and Charge First End Time 1 = 23:59 so that when the charge_freeze_service is called it just acts as an ON/OFF switch as no matter what time it is called it is within the 'window'. 

discharge_freeze_service:

This is LuxPower Charge Last - no need to do anything special for this as it is just an Enable/Disable switch.

gcoan, 2025-04-03:
@brickatius I've raised #2196 separately for Trefor to resolve the spurious time entity validation warnings

Also raised #2195 to enable battery charge/discharge curves to use soc_percent and have removed your custom soc_kw from the documentation

Added @raldred 's soc corrected template sensor

@brickatius can you give me the full YAML for charge and discharge freeze services/calls please?  At the moment both of these are disabled in the luxpower config.  Happy to add them

brickatius, 2025-04-03:
> Hi, thanks for the updates. Made a few modifications based on the info presented in this thread.
> I've not looked at the the luxPython template for a while and am curious about the following:
> I have never had charge/discharge_rate set in my apps.yaml and things seem to have been working fine. Am curious how these helpers are set up?

Hi,
This is another one that doesn’t have a LuxPower entity. Look in your inverter’s data sheet for the basic value. My inverter’s an LXP 5kw and the charge and discharge current is stated as 4000w. You can double check this by observing the inverter flow chart. Mine’s within about 10w. As far as I can tell Predbat will need this in order to work out how long to charge or discharge for in order to get to the desired state of charge.

owinwood, 2025-04-03:
> Hi,
> This is another one that doesn’t have a LuxPower entity. Look in your inverter’s data sheet for the basic value. My inverter’s an LXP 5kw and the charge and discharge current is stated as 4000w. [...]

I have the LXP 3.6kw and according to the data sheet it's 3600w. I've never had an issue with that not being set, it's always managed to reached the targets in the plan pretty well. But have set those values now just to be sure.
Thanks

raldred, 2025-04-03:
Those entities are not required as we're using

```
  timed_charge_current: 
    - number.lux_charge_current_limit
  timed_discharge_current:
    - number.lux_discharge_current_limit
```

brickatius, 2025-04-03:
To be honest, I don’t know about the target charge and discharge targets. 
The entity that @raldred mentions is measured in Amps, whereas the targets are in Watts as can be seen in the inverter flow chart. Don’t these figures have an impact on the time needed to reach a target percentage SOC as per the plan? My 5kW inverter will only charge and discharge the batteries at 4000w. Perhaps this is one for @springfall2008 to clarify.

raldred, 2025-04-03:
So you need to define `battery_rate_max`, I've hard coded mine as. know my system charges at about 6.2kW (120amps) 

```
  battery_rate_max:
    - 6200
```

You should define yours at about `4000`

raldred, 2025-04-03:
This my plan for this evening.
It always reaches target perfectly timed as it also syncs with my car charging on intelligent octopus.

[image]

owinwood, 2025-04-03:
@raldred, everything above mirrors how I have had mine set up, which explains why eveything looks like it was working fine (because it was). Going to remove the charge_rate and discharge_rate from my apps.yaml.

brickatius, 2025-04-03:
OK I get it. I do have this. 

``` battery_rate_max: 4000 ```

So this is doing the job of the 2 input_number entities rendering them unnecessary. I never had a problem with the force charge or discharge slots not reaching their intended targets. 
@gcoan I can confirm that they are not needed.

gcoan, 2025-04-03:
> So this is doing the job of the 2 input_number entities rendering them unnecessary

charge_rate and discharge_rate removed from luxpower.yaml

> The charge level targets is using SOC via the `charge_limit` param and `number.lux_ac_battery_charge_level` on the inverter, along with the automation I mentioned earlier that [@gcoan](https://github.com/gcoan) has added to the setup doc, that automation syncs `number.lux_discharge_current_limit` when the the batpred plan is discharging, because batpred doesn't (maybe it does now?) have the ability to define separate charge targets for charge and discharge as is required by lux inverters.

I was a bit confused by your automation to be honest.

Predbat sets whatever is configured to charge_limit (number.lux_ac_battery_charge_level) when it is doing a forced grid charge, but as far as I know it didn't set this when it was doing a forced grid discharge.  What predbat does set though is the discharge target to whatever entity is configured to 'reserve' in apps.yaml

For luxpower has_reserve_soc is set to False and there is no reserve set,  I suggest try changing this and adding

```yaml
reserve:
  - number.lux_forced_discharge_battery_level
```

and then I think your automation won't be needed

#2197 has been raised for luxpower freeze charge/discharge

brickatius, 2025-05-13:
You know that we removed charge_rate and discharge_rate from apps.yaml - For completeness I think that we need to put them back because of this:
```
Note: Cannot find battery charge curve (settings missing), one of the required settings for soc_kw, battery_power and charge_rate are missing from apps.yaml
Note: Cannot find battery discharge curve (settings missing), one of the required settings for soc_kw, battery_power and discharge_rate are missing from apps.yaml
```
I did reinstate these in apps.yaml but Predbat is complaining about 2 errors in the apps.yaml file.

raldred, 2025-05-13:
Those aren't errors, it's just a warning, it will predict charge discharge based on the max charge/discharge rate.
It probably depends on your connected batteries, for mine (Hanchu), there's barely a curve, there's no curve for discharge.
```
  battery_charge_power_curve:
    97 : 0.5
    98 : 0.2
    99 : 0.1
  #battery_discharge_power_curve:
  #  4 : 1.0
```

owinwood, 2025-05-13:
I have Pylontech US5000 batteies with a Lux Power 3600 Hybrid Inverter, setting the following in my apps.yaml (estimated by observation)
battery_rate_max:
    - 3250

Checked my logs and I'm not seeing those notes.

gcoan, 2025-05-13:
> Those aren't errors, it's just a warning, it will predict charge discharge based on the battery_rate_max.

yes it is a warning and can be ignored as predbat will predict the charging and discharging based upon battery rate max, but if you do create a battery charge curve then the prediction will just be a bit more accurate,  so for a bit of setup and one less warning to look at, its worth it I think

Looking back, we had:

```
  # charge_rate:
  #   - input_number.lux_target_charge_current_w
  # discharge_rate:
  #   - input_number.lux_target_discharge_current_w
```

but elsewhere in apps.yaml, the charge and discharge was using number.lux_target_charge_current_w not input_number:

```
  timed_charge_current:
    - number.lux_charge_current_limit
  timed_discharge_current:
    - number.lux_discharge_current_limit
```

which is correct?

I'll add your example curves to the template @raldred

brickatius, 2025-05-14:
I recreated the 2 'input_number' helpers for charge_rate and discharge_rate (Lux Charge Rate W & Lux Discharge Rate W) populated with values for my inverter of 4010 and 3950 respectively. It would appear that these are not being read but are being replaced by dummy sensors by Predbat from the battery_rate_max (4,000 in my case) as you can see below 
I created a template helper for 'soc_kw' as follows:
```
{{ states("sensor.lux_battery_capacity_ah") |float * states("sensor.lux_battery_voltage_live") | float * states("sensor.lux_battery") | float / 100000}}
```
we then get this from predbat.log. A proper discharge power curve was calculated - obviously unique to my inverter and battery setup.  (The charge curve was lacking data so I used the figure from when it was working before)
```
charge_rate	
- sensor.predbat_LuxPower_0_charge_rate = 4000
battery_charge_power_curve	
100	: 1
discharge_rate	
- sensor.predbat_LuxPower_0_discharge_rate = 4000.0
battery_discharge_power_curve	
20	: 1.0
19	: 0.98
18	: 0.96
17	: 0.97
16	: 0.99
15	: 0.97
14	: 0.96
13	: 0.95
12	: 0.95
11	: 0.95
10	: 0.95
9	: 0.95
8	: 0.95
7	: 0.95
6	: 0.95
5	: 0.95
4	: 0.95
```
We're kind of going round in circles here. I've no idea why the input_number helpers are not being read
```
charge_rate:
    - input_number.lux_charge_rate_w
discharge_rate:
    - input_number.lux_discharge_rate_w
```
and are replaced by the predbat sensor value but if we don't have something there the charge and discharge curves are not calculated because Predbat is specifically looking for soc_kw, battery_power and charge_rate / discharge_rate.  You can't hard code a value for the charge and discharge rates either - apps.yaml complains about 2 errors. 
TBH I'm at a loss as to what to suggest.

raldred, 2025-05-14:
I really dont think these thigns are going to make a great deal of difference to the overall plan accuracy.
But this adds more complication to the documentation for new users having to create a bunch of additional sensors.

I think you need to use template sensors, not input numbers, since these are values to read, they're not being set since we're using `timed_charge_current` and `timed_discharge_current` for batpred controlling the charging power, in Amps.

I notice the array in your example is indented too far, could this be the issue?

```
charge_rate:
  - input_number.lux_charge_rate_w
discharge_rate:
  - input_number.lux_discharge_rate_w
```

gcoan, 2025-05-14:
The problem may be using input_numbers instead of a proper sensor (template sensor is OK) for charge & discharge rate as predbat needs to load the state history when working out the curves

Ignore my comment above querying the sensor names, was late at night, I misread.

Agree this adds only limited value, and its not essential for beginners, but if we have a solution then happy to include it as optional

brickatius, 2025-05-14:
I don't think there's an ideal solution which is a shame.  I tried a template sensor for battery_rate_max and  charge and discharge rates - example here:
```
{{states("number.lux_charge_current_limit") | float (1) * states("sensor.lux_battery_voltage_live") | float (1)}}
```
even though it gives an accurate picture of the battery_rate_max predbat didn't like it when entered as the charge and discharge rates. 
As you said this is only of limited value. I think the example charge and discharge curves in the luxpower yaml are probably fine. If anyone wants to dig around they can always find this issue and can add a temporary solution to calculate their own specific charge/discharge curves.

gcoan, 2025-11-05:
Closing as the predbat documentation now includes all the above Luxpower inverter configuration