springfall2008/batpred #2185: Solcast API calls happening too often

pacemaker82, 2025-04-02:
**Describe the bug**
I have solcast set to refresh every 6 hours in the predbat config yaml

```
  solcast_host: 'https://api.solcast.com.au/'
  solcast_api_key: 'REDACTED'
  solcast_poll_hours: 6
```

I am seeing that the solcast API calls from Predbat are happening much more often, usually in quick succession after the 6 hour refresh. Sometimes 1-2 times more within 5 minutes of the initial refresh of the call. You can see here the slight changes to the API response after each refresh.

[image]

In addition, within the `sensor.predbat_pv_today` the number of API calls used and limit are both set to zero:

[image]

**Expected behaviour**

API calls to Solcast should be every 6 hours per the apps.yaml config only as they are limited.

**Predbat version**

8.17.3

gcoan, 2025-04-02:
just as a comment on this, predbat will need to take account of the 'consumed' API's itself as Solcast removed this API last year

The Solcast integration has to count its own API calls now

pacemaker82, 2025-04-04:
[solcast_forecast_history.csv](https://github.com/user-attachments/files/19601371/solcast_forecast_history.csv)

Providing a CSV dump of the last 3 days where this is happening. I have highlighted YES where I am seeing calls in somewhat quick succession (Where time difference is less than 15000 seconds)

Just to visualise the issue:

[image]

springfall2008, 2025-04-12:
This won't be due to Solcast API calls as you only get 10 a day, somehow predbat is changing the data when it refreshes from the save data - I need to check why this is a bug

springfall2008, 2025-04-12:
Interestingly I don't see this myself, can you check the logfiles as you will see the solcast refresh and then subsequent calls using cached data:

2025-04-12 01:41:12.087838: Fetching https://api.solcast.com.au/rooftop_sites
2025-04-12 01:41:12.715719: Writing cache data for https://api.solcast.com.au/rooftop_sites to cache file /config/cache/https___api.solcast.com.au_rooftop_sites_385d1b497c8bff2a7263dc8f76bb28dc.json
2025-04-12 01:41:12.716772: Fetch data for resource id 13fe-386c-1934-1436
2025-04-12 01:41:12.718474: Fetching https://api.solcast.com.au/rooftop_sites/13fe-386c-1934-1436/forecasts
2025-04-12 01:41:14.259456: Writing cache data for https://api.solcast.com.au/rooftop_sites/13fe-386c-1934-1436/forecasts to cache file /config/cache/https___api.solcast.com.au_rooftop_sites_13fe-386c-1934-1436_forecasts_ea49d81b418ad4a3d47001702a015a4b.json
2025-04-12 01:41:14.350745: Solcast returned 340 data points
2025-04-12 01:41:14.360867: PV Forecast for today is 10.23 (6.79 10% 14.62 90%) kWh and left today is 10.23 (6.79 10% 14.62 90%) kWh
2025-04-12 01:41:14.382209: PV Forecast for day tomorrow is 11.75 (5.74 10% 16.16 90%) kWh
2025-04-12 01:41:14.399610: PV Forecast for day d2 is 9.34 (2.25 10% 16.24 90%) kWh
2025-04-12 01:41:14.422589: PV Forecast for day d3 is 11.55 (1.14 10% 17.1 90%) kWh
2025-04-12 01:41:14.445602: PV Forecast for day d4 is 13.57 (2.7 10% 17.45 90%) kWh
2025-04-12 01:41:14.468562: PV Forecast for day d5 is 10.86 (0.85 10% 16.4 90%) kWh
2025-04-12 01:41:14.491586: PV Forecast for day d6 is 9.01 (0.7 10% 15.28 90%) kWh

---

2025-04-12 01:42:37.233360: Obtaining solar forecast from Solcast API
2025-04-12 01:42:37.234255: Return cached data for https://api.solcast.com.au/rooftop_sites age 1.4 minutes
2025-04-12 01:42:37.234281: Fetch data for resource id 13fe-386c-1934-1436
2025-04-12 01:42:37.234866: Return cached data for https://api.solcast.com.au/rooftop_sites/13fe-386c-1934-1436/forecasts age 1.4 minutes
2025-04-12 01:42:37.266865: Solcast returned 340 data points
2025-04-12 01:42:37.277111: PV Forecast for today is 10.23 (6.79 10% 14.62 90%) kWh and left today is 10.23 (6.79 10% 14.62 90%) kWh
2025-04-12 01:42:37.299790: PV Forecast for day tomorrow is 11.75 (5.74 10% 16.16 90%) kWh
2025-04-12 01:42:37.311139: PV Forecast for day d2 is 9.34 (2.25 10% 16.24 90%) kWh
2025-04-12 01:42:37.322445: PV Forecast for day d3 is 11.55 (1.14 10% 17.1 90%) kWh
2025-04-12 01:42:37.333823: PV Forecast for day d4 is 13.57 (2.7 10% 17.45 90%) kWh
2025-04-12 01:42:37.345045: PV Forecast for day d5 is 10.86 (0.85 10% 16.4 90%) kWh
2025-04-12 01:42:37.356349: PV Forecast for day d6 is 9.01 (0.7 10% 15.28 90%) kWh
2025-04-12 01:42:37.407577: Today's load divergence 100.0 % in-day adjustment 100.0 % damping 0.95x

pacemaker82, 2025-04-12:
it seems to have stopped doing it on my end at the moment so no log to show. I'll pull one when it happens again.

closed 2026-08-28 by chalfontchubby