springfall2008/batpred #2186: Compare function graph not updating since ?8.17.1

PianSom, 2025-04-02:
**Describe the bug**
On the Compare chart the lines showing historic actual usage no longer update. This started a few days ago, I think when I updated to 8.17.1, and has continued since

Aberrent behaviour confirmed by two other users.

**Expected behaviour**
Updating line

**Predbat version**

8.17.3

**Environment details**

- AIOx2 + Gateway
- Docker containers for Predbat + HA

**Screenshots**

[image]

**Log file**
On request

**Predbat debug yaml file**
On request

springfall2008, 2025-04-02:
This is a timezone issue..

springfall2008, 2025-04-02:
Fix on main

PianSom, 2025-04-02:
Thank you Trefor

springfall2008, 2025-04-03:
Now released

closed 2025-04-03 by springfall2008