springfall2008/batpred #2187: Validation of apps.yaml error

MarekKubacak, 2025-04-02:
After the latest update, the system throws validation warnings and an error during apps.yaml validation. Several sensor values (e.g., sensor.solis_house_load_today, sensor.solis_pv_total_power, etc.) are expected to be floats but are returning unknown, causing 8 configuration errors in total.

LOG 
`2025-04-02 12:18:31.471405: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-04-02 12:18:31.471467: Warn: Validation of apps.yaml found configuration item 'load_today' element sensor.solis_house_load_today is not a float got unknown
2025-04-02 12:18:31.471500: Warn: Validation of apps.yaml found configuration item 'import_today' element sensor.solis_grid_import_today is not a float got unknown
2025-04-02 12:18:31.471517: Warn: Validation of apps.yaml found configuration item 'export_today' element sensor.solis_grid_export_today is not a float got unknown
2025-04-02 12:18:31.471542: Warn: Validation of apps.yaml found configuration item 'pv_today' element sensor.solis_power_generation_today is not a float got unknown
2025-04-02 12:18:31.471577: Warn: Validation of apps.yaml found configuration item 'battery_power' element sensor.solis_battery_power_2 is not a float got unknown
2025-04-02 12:18:31.471600: Warn: Validation of apps.yaml found configuration item 'pv_power' element sensor.solis_pv_total_power is not a float got unknown
2025-04-02 12:18:31.471640: Warn: Validation of apps.yaml found configuration item 'load_power' element sensor.solis_house_load is not a float got unknown
2025-04-02 12:18:31.471677: Warn: Validation of apps.yaml found configuration item 'reserve' element number.solis_backup_mode_soc is not a float got unknown
2025-04-02 12:18:31.471839: Error: Validation of apps.yaml found 8 configuration errors
2025-04-02 12:18:31.471911: --------------- PredBat - update at 2025-04-02 12:15:00+01:00 with clock skew 0 minutes, minutes now 735`

springfall2008, 2025-04-03:
Surely this is correct if these values are unknown, have you checked the sensors?

MarekKubacak, 2025-04-03:
Sensor looks fine here are screenshots 

[image]

[image]

springfall2008, 2025-04-04:
Do these errors not clear after a little while, it strikes me like a reboot issue (Solis not being up yet)?

MarekKubacak, 2025-04-04:
Usually, in the opposite way, after a full reboot/power cycle, there is no error. After a couple of hours/days, I got an error.

MarekKubacak, 2025-04-04:
I just updated to v8.17.5, and the SolaX Modbus version is 2025.04.1. I will see if anything will come back.

MarekKubacak, 2025-04-23:
Another update, and the error is back :( 

[image]

closed 2026-08-28 by chalfontchubby