springfall2008/batpred #2193: Givtcp REST API Path

dulochpark, 2025-04-03:
In the Predbat documentation and yaml file, the URL for the Givtcp REST API is http://homeassistant.local:6345

The URL for using Givtcp 3.0 in Home Assistant is http://homeassistant.local:8099/REST1

I've had Predbat installed in Home Assistant since September and it was only by chance that I read the documentation for Givtcp and I saw that the 6345 port is for docker containers and 8099/REST1 was for Home Assistant.

Can you please put this into the YAML file and documentation? It would help people like me that rarely RTFM and I guess the majority of people run Predbat in Home Assistant rather Docker these days.

Thanks!
Craig

dulochpark, 2025-04-03:
Hello!

I found the details below in this Givtcp document - http://192.168.68.71:8123/hassio/addon/533ea71a_givtcp/info

GivTCP provides a REST API for data output and control. REST service is available at: http://ipaddress:8099/ Each available inverter exposes its REST interface on the following endpoints:

/REST1/
/REST2/
/REST3/
/REST4/
/REST5/ You can call REST1/api to discover what api calls are available to you.
If Docker is running in Host mode then the REST service is available on port 6345

springfall2008, 2025-04-04:
The 8099 port appears to work as well, maybe they supported the old and new methods

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

gcoan, 2026-08-20:
Use of the givtcp ports works fine, so no real benefit to change

closed 2026-08-20 by gcoan