springfall2008/batpred #2194: Validation of apps.yaml found 2 configuration errors - SolarEdge Inverter and GE AIO battery

kams19, 2025-04-03:
**Describe the bug**
Validation of apps.yaml found 2 configuration errors

**Expected behaviour**
no errors reported in apps.yaml

**Predbat version**

v8.17.4

**Environment details**

- SolarEdge Inverter and GE AIO battery
- Standard HAOS installer

**Screenshots**

[image]

[image]

[image]

[predbat_debug.yaml (2).txt](https://github.com/user-attachments/files/19583592/predbat_debug.yaml.2.txt)

**Log file**
2025-04-03 10:01:14.847358: Error: Validation of apps.yaml found 2 configuration errors
Watching ['./execute.py', './apps.yaml', './download.py', './gecloud.py', './inverter.py', './energydataservice.py', './predbat.py', './compare.py', './solcast.py', './db_manager.py', './web.py', './unit_test.py', './output.py', './ha.py', './predheat.py', './plan.py', './prediction.py', './config.py', './octopus.py', './utils.py', './userinterface.py', './hass.py', './futurerate.py', './alertfeed.py', './fetch.py'] for changes
2025-04-03 10:01:16.359360: Warn: Validation of apps.yaml found configuration item 'pv_power' element number.givtcp_chxxxxxxx_pv_power returned value None
2025-04-03 10:01:16.359452: Warn: Validation of apps.yaml found configuration item 'load_power' element number.givtcp_chxxxxx_load_power returned value None

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-04-04:
This appears to be an issue with your configuration, I believe these should be sensor. and not number.

kams19, 2025-04-04:
Thanks Trefor. I have updated the config to show them as sensor against
number.
However after the recent update (v8.17.5) I am getting two more errors:
2025-04-04 14:13:56.502726: Warn: Validation of apps.yaml found
configuration item 'car_charging_soc' element sensor.xxxxx_battery is not a
valid type float
2025-04-04 14:13:56.502861: Warn: Validation of apps.yaml found
configuration item 'car_charging_limit' element number.xxxxx_charge_limit
is not a valid type float

Could you advise please?

closed 2025-04-19 by springfall2008