springfall2008/batpred #2195: Battery charge/discharge curves require soc_kw and can't accept soc_percent

gcoan, 2025-04-03:
**Describe the bug**
Originally identified by @brickatius in https://github.com/springfall2008/batpred/issues/2184#issuecomment-2775400589

LuxPower inverter provides soc as a percentage, which predbat will accept for planning, but the battery charge/discharge function requires soc_kw (via a custom template sensor)

**Expected behaviour**
For consistency and expansion to other inverters, battery charge/discharge curves should accept soc either in kw or percentage

**Predbat version**
8.17

**Environment details**
Luxpower

springfall2008, 2025-04-04:
What do you mean by 'function' do you mean the service API?

tomkentpayne, 2025-04-04:
This may be related to [issue/enhancement 2026](https://github.com/springfall2008/batpred/issues/2026) that I raised for the same thing but to do with `charge_rate_percent` and `discharge_rate_percent` not being able to be used for charge/discharge curve prediction.

I'd actually already hit the `soc_kw` vs. `soc_percent` problem, but had worked around it by creating a template sensor. I think at the time I didn't know how to do template sensors for `charge/discharge_rate_percent`, assuming they were "outputs" from Predbat, as opposed to just a sensor.

Just want to tack on to this comment btw - thank you so much for all your efforts in creating and maintaining this project. Everyone I've shown Predbat's plan to and explained how it works is seriously impressed, and despite my wife having spent a couple of months being very sceptical about "the predator" being in control of our battery, she's definitely come around having seen it automatically signing up to Octopus saving sessions, exporting excess energy at key times, etc.

gcoan, 2025-04-04:
> battery charge/discharge function

Sorry very loose words here @springfall2008 

I meant to refer to the predbat routine that automatically creates charge/discharge curves.  At the moment it needs soc_kw for this, which not all inverters have (eg luxpower).

Proposed the curves could be created with soc_percent

springfall2008, 2025-04-05:
> Sorry very loose words here [@springfall2008](https://github.com/springfall2008)
>
> I meant to refer to the predbat routine that automatically creates charge/discharge curves. At the moment it needs [...]

Ah yes this could be possible

OneTwo3D, 2025-06-08:
Hi, 
adding to this topic: it seems the charge/discharge curve generation is also not possible for Solis inverters due to the missing soc_kw sensor but predbat also also complains about:
```
Warn: Validation of apps.yaml found configuration item 'charge_rate' element sensor.solis_battery_input_energy can not be modified
Warn: Validation of apps.yaml found configuration item 'discharge_rate' element sensor.solis_battery_output_energy can not be modified 
```

In the Solax integration this value is only showing the charge/discharge rate in W for the battery but doesn't allow it to be set by PredBat I assume. For the charge/discharge curve generation just reading these sensors' historic data should be sufficient I guess, so no writing to it required?

springfall2008, 2026-08-29:
**Automated first-pass triage** — a maintainer will review before any action is taken.

**Classification:** enhancement (already labelled)
**Priority:** not set (feature request)

## Investigation

Checked current main (`v8.53.5-33-gb6a713e2`). The requested capability — deriving the auto-computed battery charge/discharge curve from `soc_percent` instead of requiring `soc_kw` — has already been implemented:

- `Inverter.find_charge_curve()` in `apps/predbat/inverter.py:1039` now checks `if "soc_percent" in self.base.args` and, when present, reads history from the configured `soc_percent` sensor directly, converting it to kWh internally (`inverter.py:1084-1101`) rather than requiring a `soc_kw` template sensor.
- This landed in PR #2463 ("Add automatic unit detection for entity values"), first released in **v8.21.1** (2025-06-16) — after this issue was raised against v8.17.
- `apps/predbat/tests/test_find_charge_curve.py` exercises this path directly (`my_predbat.args["soc_percent"] = ["sensor.soc_percent"]`, no `soc_kw` set), and the registered `find_charge_curve` test passes on current main:

```
tools/triage_test.sh find_charge_curve
**** Test find_charge_curve PASSED in 0.29s ****
```

So a LuxPower user (or anyone else without a native `soc_kw` sensor) can now set `soc_percent` in `apps.yaml` and get the automatic curve computation without a custom template sensor — no `soc_kw` workaround needed. This looks resolved; suggest closing unless there's a remaining gap I'm missing.

**Related but distinct, still open:**
- #2026 — `charge_rate_percent`/`discharge_rate_percent` still can't be used for curve prediction (only `charge_rate`/`discharge_rate` in native units are read for the curve); that's a separate ask from this issue and doesn't appear to be fixed by the above.
- The Solis comment on this thread (from @OneTwo3D) about `charge_rate`/`discharge_rate` sensors being flagged as "cannot be modified" is a different symptom (a validation warning on read-only rate sensors) and would need its own investigation.

closed 2026-10-28 by springfall2008