springfall2008/batpred #2196: apps.yaml validation rejects charge/discharge start/end time when using time. sensors

gcoan, 2025-04-03:
**Describe the bug**
In #2184 have simplifying the LuxPower template to use time sensors which predbat accepts and works fine:

```yaml
  charge_start_time:
    - time.lux_ac_charge_start1
  charge_end_time:
    - time.lux_ac_charge_end1
  discharge_start_time:
    - time.lux_force_discharge_start1
  discharge_end_time:
    - time.lux_force_discharge_end1
```

These are however rejected as invalid by the new apps.yaml validation https://github.com/springfall2008/batpred/issues/2184#issuecomment-2775177854

**Expected behaviour**
time sensors not rejected as invalid in apps.yaml validation

**Predbat version**
8.17

**Environment details**
LuxPower

**Screenshots**
See https://github.com/springfall2008/batpred/issues/2184#issuecomment-2775177854

springfall2008, 2025-04-05:
Can you show the actual Warning reported?

springfall2008, 2025-04-05:
Fix on 'main' for what I think is the issue

gcoan, 2025-04-05:
@brickatius @raldred @owinwood 

can you test that this clears the apps.yaml validation errors you were getting

owinwood, 2025-04-05:
@gcoan. Just updated to v8.17.6 and it has cleared the apps.yaml validation errors for me.

Thank you

gcoan, 2025-04-05:
Great, thanks for confirming

closed 2025-04-05 by gcoan

brickatius, 2025-04-05:
@gcoan All cleared. Thank you.