springfall2008/batpred #2197: LuxPower - charge_freeze_service & discharge_freeze_service raising exceptions.

brickatius, 2025-04-03:
I'm trying to get the above services set up but an exception is raised each time these services are called. What am I doing wrong?
apps.yaml
has_service_api: True
support_charge_freeze: True
support_discharge_freeze: True

charge_freeze_service:
    - service: my_charge_freeze_service
      device_id: 53b47dc71e7cd328925306f3497
      power: number.lux_charge_current_limit
      target_soc: sensor.lux_battery
    - service: switch.turn_on
      entity_id: "switch.lux_charge_priority"
  discharge_freeze_service:
    - service: my_discharge_freeze_service
      device_id: 53b47dc71e7cd328925306f3497
      power: number.lux_discharge_current_limit
      target_soc: sensor.lux_battery
    - service: switch.turn_on
      entity_id: "switch.lux_charge_last"

Exceptions raised:
charge_freeze_service 

2025-04-03 15:05:23.940605: Inverter 0 Calling service charge_freeze_service domain charge service_name my_charge_freeze_service with data {'device_id': '53b47dc71e7cd328925306f3497', 'power': 'number.lux_charge_current_limit', 'target_soc': 'sensor.lux_battery'}
2025-04-03 15:05:23.940677: Error: Exception raised not enough values to unpack (expected 2, got 1)
2025-04-03 15:05:23.950155: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1307, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 632, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 442, in execute_plan
    inverter.adjust_charge_immediate(inverter.soc_percent, freeze=True)
  File "/config/inverter.py", line 2138, in adjust_charge_immediate
    if not self.call_service_template("charge_freeze_service", service_data, domain="charge", extra_data=extra_data):
  File "/config/inverter.py", line 2112, in call_service_template
    self.base.call_service_wrapper(service_name, **service_data)
  File "/config/predbat.py", line 252, in call_service_wrapper
    return self.ha_interface.call_service(service, **kwargs)
  File "/config/ha.py", line 502, in call_service
    domain, service = service.split("/")
ValueError: not enough values to unpack (expected 2, got 1)

discharge_freeze_service

2025-04-03 15:24:05.986117: Inverter 0 Calling service discharge_freeze_service domain discharge service_name my_discharge_freeze_service with data {'device_id': '53b47dc71e7cd328925306f3497', 'power': 'number.lux_discharge_current_limit', 'target_soc': 'sensor.lux_battery'}
2025-04-03 15:24:05.986258: Error: Exception raised not enough values to unpack (expected 2, got 1)
2025-04-03 15:24:05.986716: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1307, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 632, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 429, in execute_plan
    inverter.adjust_export_immediate(inverter.soc_percent, freeze=True)
  File "/config/inverter.py", line 2163, in adjust_export_immediate
    if not self.call_service_template("discharge_freeze_service", service_data, domain="discharge", extra_data=extra_data):
  File "/config/inverter.py", line 2112, in call_service_template
    self.base.call_service_wrapper(service_name, **service_data)
  File "/config/predbat.py", line 252, in call_service_wrapper
    return self.ha_interface.call_service(service, **kwargs)
  File "/config/ha.py", line 502, in call_service
    domain, service = service.split("/")
ValueError: not enough values to unpack (expected 2, got 1)

2025-04-03 15:24:06.028682: Info: record_status Error: Exception raised not enough values to unpack (expected 2, got 1)
2025-04-03 15:24:06.028826: Error: not enough values to unpack (expected 2, got 1)

Any help would be greatly appreciated.

gcoan, 2025-04-03:
I'm personally not a fan of device id's which are HA-install specific and I never use them.  I always just use entity id's.

Don't know if that helps fix your problem or not?

brickatius, 2025-04-03:
I agree. I only used device_id because it is what's mentioned in the documentation on the following page:-.
[https://springfall2008.github.io/batpred/inverter-setup/#service-api](url)
I'm no expert and I can't fathom out from the exception what the missing value could be.

brickatius, 2025-04-03:
@raldred It would be good to get this working properly especially ```discharge_freeze_service```  (Charge Last) since this is more likely to happen following update v18.16.7 @springfall2008 
I am hoping that this problem turns out to be nothing more than a schoolboy error on my part.

gcoan, 2025-04-03:
> charge_freeze_service:
> - service: my_charge_freeze_service
> device_id: 53b47dc71e7cd328925306f3497
> power: number.lux_charge_current_limit
> target_soc: sensor.lux_battery
> - service: switch.turn_on [...]

have you got a service defined in Home Assistant called my_charge_freeze_service?  The example in the documentation is just that, its an example, is expecting you to customise to the precise sequence of activities you need to invoke to do a charge freeze - this can either be manipulating HA controls OR calling an API, or both

If you look at the terminolgy Predbat uses for Freeze Charge mode https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-modes (and freeze discharge), you will probably need to manipulate the luxpower inverter controls to achieve the same result.

So freeze charging on my givenergy inverter sets the inverter into charge mode (set start and end time and turn AC charge enable on) BUT also sets the target_soc (charge-to) to the current SoC value

raldred, 2025-04-03:
Yeh so having a think about it, based on the terminology of the states @gcoan linked, I think we could do something like...

#### Freeze charging

1. Switch on `switch.lux_charge_last`
2. Switch off `switch.lux_feed_in_grid`
3. Set `number.lux_charge_current_limit` to `0`

 _PV or the grid will meet demand, If PV generation exceeds demand, it will charge the battery_ 

#### Freeze exporting

1. Switch on `switch.lux_charge_last`
_this should be enough, excess PV will be exported, if there is a short fall the battery will discharge to meet demand as normal_ 

We'll need an automation to watch for changes to `predbat.status` to switch off `switch.lux_charge_last` and switch on `switch.lux_feed_in_grid` when it's no longer in `freeze` state.

gcoan, 2025-04-03:
> We'll need an automation to watch for changes to predbat.status to switch off switch.lux_charge_last and switch on switch.lux_feed_in_grid when it's no longer in freeze state.

there isn't a stop_discharge_freeze service in predbat unfortunately - maybe there should be - but at the moment it calls the same stop_discharge service

Could the luxpower stop_discharge turn lux_charge_last off and switch on lux_feed_in_grid for every stop discharge activity - would that cause a problem (for ordinary export to grid discharges) ?

raldred, 2025-04-03:
If we setup an automation....
We can use conditions to trigger different things depending on the current state?

```
alias: Lux toggle discharge freeze
description: ""
triggers: []
conditions: []
actions:
  - if:
      - condition: state
        entity_id: switch.lux_charge_last
        state: "on"
    then:
      - action: switch.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: switch.lux_charge_last
    else:
      - if: 
          - condition: state
            entity_id: switch.lux_charge_last
            state: "off"
        then:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.lux_charge_last
mode: single

```

Then we can trigger from batpred with...

```
action: automation.trigger
data: {}
target:
  entity_id: automation.lux_toggle_discharge_freeze
```

gcoan, 2025-04-03:
I think discharge freeze service is only called at the start of a discharge freeze, its the ordinary stop discharge service that is caused at the end of BOTH a discharge freeze and a discharge period

So the flexibility needs to be in discharge_stop

raldred, 2025-04-03:
I see what you're saying, stopping freeze calls the regular `discharge_stop_service`
So I think we could do that with a similar automation?

raldred, 2025-04-03:
Ahh I remember why I gave on this, you can't set the charge/discharge current to zero. I wanted to use current control to prevent charging/discharging, but the minimum I can set on my inverter is 8A.

How about this...

When triggered
* If charge last is on, it will switch it off
* If charge last is off, it will disable the regular force discharge

```
alias: Lux batpred stop discharge
description: "Either stop freeze discharge or regular discharge"
triggers: []
conditions: []
actions:
  - if:
      - condition: state
        entity_id: switch.lux_charge_last
        state: "on"
    then:
      - action: switch.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: switch.lux_charge_last
    else:
      - if:
          - condition: state
            entity_id: switch.lux_charge_last
            state: "off"
        then:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.lux_force_discharge_enable
mode: single

```

Call with

```
discharge_stop_service:
  service: automation.trigger
  entity_id: automation.lux_batpred_stop_discharge
```

brickatius, 2025-04-03:
I don't know if this will be any help as a start but I made a couple of switch groups one ```switch.lux_charge_group``` which consists of Lux Charge Priority and Lux AC Charge Enable and ```switch.lux_discharge_group``` which consists of Lux Force Discharge Enable and Lux Charge Last. 
Calling 

``` 
charge_stop_service:
    service: switch.turn_off
    entity_id: "switch.lux_charge_group"
```

Would ensure that both were turned off. Likewise 

```
discharge_stop_service:
    service: switch.turn_off
    entity_id: "switch.lux_discharge_group"
```

That way I can't see a scenario where one member of the group would be on and the other would be off. All the entities can be turned on individually. 
Doesn't particularly help with defining the services properly though.

raldred, 2025-04-04:
Ok, my freeze export is working as expected it seems, 

[image]

It's been functioning normally with my import/export overnight

[image]

raldred, 2025-04-04:
Interesting switch groups might work freeze export but I don't think charge priority is going to give us what we want because we don't want the battery charging during freeze?

brickatius, 2025-04-04:
Sorry to disagree but I think it is Charge Priority, This is from the documentation:

Freeze charging - The battery is charging but the current battery level (SoC) is frozen (held). Think of it as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar power above the house load, the excess solar will be used to charge the battery,

raldred, 2025-04-04:
"Charge priority" on Lux inverters is intended to prioritise charging the bettery first, before any solar is used to supply load, it will only supply the load if PV exceeds your max charge rate.

Therefore I believe this goes against the ideal of what batpred is trying to do with "Charge Freeze", since it it's intending to hold the current SOC, and use PV / Grid for demand, you wont get that if you use "Charge priority".

brickatius, 2025-04-04:
Just a thought……Should we be looking at the logic of why and when freeze_charge happens? Is it that it’s only called when the battery SOC is less than 100% and it’s important to the plan that the battery is not discharged? Is it the intention that the battery gets up to 100% sooner than would otherwise have happened, even at the expense of having to import from the grid?  If that is the case I’m thinking that LuxPower might have a built-in advantage with Charge First/Priority. Given that the plan recalculates every 5 - 10 mins it may be that freeze_charge is needed for less time. 
I’ve never used Charge First/Priority and am away from home for a couple of days but I’d really like to see what actually happens when it is enabled. 
Have I got this completely wrong @springfall2008 ?

brickatius, 2025-04-05:
What if we used Charge Priority but had a custom service with a template sensor that reduced the battery_rate_max (with the proviso that it is >= 0) to the difference between the PV output live and the house load live when the freeze_charge_service was called? That way the house load is met, any excess charges the battery and hopefully if there’s a shortfall the battery won’t discharge and the difference is met by the grid. 
Apologies, I’m on my phone so giving a code example is painfully slow.

gcoan, 2025-04-05:
How and why predbat uses the different modes is a bit of a black art! 
One example of freeze charge that predbat is doing for me on Octopus Cosy is in the cheaper rate periods, Predbat will set a freeze charge to hold the current soc and ensure that the home runs off grid as this is cheaper (after losses) than letting the battery discharge and then be recharged next day from solar.
This can happen regardless of SoC, its more a function of import and export rates now and in the future.

[image]

I would be doubtful of changing battery charge rates based upon live PV values - PV changes all the time based on temperature, clouds, etc and so you won't get an accurate result.

Note that whilst predbat runs every 5 minutes it only recalculates the plan every 10 minutes.  Every other run it is just executing the previously calculated plan (which is at 5 minutes granularity)

brickatius, 2025-04-07:
DISCHARGE_FREEZE_SERVICE

From the documentation
Freeze exporting - The battery is in demand mode, but with charging disabled. The battery or solar covers the house load. As charging is disabled, if there is excess solar generated, the current SoC level will be held and the excess solar will be exported. If there is a shortfall of generated solar power to meet the house load, the battery will discharge to meet the extra load.

This is easy as it is the same as Charge Last

```
discharge_freeze_service:
    - service: switch.turn_on
      entity_id: "switch.lux_charge_last"
```

[image]

[image]
(As far as I know the 58w going to the battery is what is required to keep the BMS ticking over)

And a little later with a high load with FrzExp 

[image]

As mentioned before I made a switch group "switch.lux_discharge_group" with the following members -

[image]

```
discharge_stop_service:
    service: switch.turn_off
    entity_id: "switch.lux_discharge_group"
```

raldred, 2025-04-08:
Sounds good, yeh freeze export is as easy as enabling charge last, and ensuring is disabled by stop discharge.

I've done it by triggering an automation because it like the traces you get with automation triggers but the result is the same.

Freeze charge is more complicated.

brickatius, 2025-04-08:
CHARGE_FREEZE_SERVICE

From the documentation - 

Freeze charging - _**The battery is charging**_ but the current battery level (SoC) is frozen (held). Think of it as a charge to the current battery level. The grid or solar covers any house load. If there is a shortfall of Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar power above the house load, the excess solar will be used to charge the battery,
_(the bold italics are mine)_

Despite some admittedly confusing terminology I read this as being basically the same as 'Demand mode' but with with any shortfall being met from the grid, not the battery.
Note: if **_The battery is charging_** is correct then battery level (SoC) being frozen (held) isn't quite right, as the battery level can actually rise but it won't go below its present level. 

@raldred I tested your solution this morning and evening. It does work however, with respect, it is not quite doing what is described above. Battery is charging at the same rate as the solar output (less losses) and all of the house load is coming from the grid. Solar is not contributing to the house load. Unless I'm mistaken isn't this the same behaviour as Charge First?

In an attempt to mimic 'charge_freeze_service' ,as described, I removed the 'Charge Last' action from the @raldred automation as follows:- 

```
alias: Lux batpred start freeze charge
description: >-
  Start batpred charge freeze mode, turn on charge last, and set discharge
  current 0 to prevent discharging battery to satisfy load
triggers: []
conditions: []
actions:
  - action: number.set_value
    metadata: {}
    data:
      value: "0"
    target:
      entity_id: number.lux_discharge_current_limit
mode: single
``` 

This resulted in the inverter behaving as if it was in 'Charge First / Charge Priority' mode - all solar going to the battery and house load coming from the grid, again not matching the definition of charge_freeze_service.

Unless anyone knows otherwise I can't see a way to manipulate 'Demand' mode to replicate what 'charge_freeze_service is supposed to do. Does anyone else agree?

If we can't get anything to work, is it worth submitting a feature request specific to LuxPower inverters such that 'Charge First / Charge Priority' can be used by the Predbat plan if ever it proved to be cost-effective? @springfall2008 @gcoan

raldred, 2025-04-08:
There's nothing predbat can do more, we need to find the combination.

I think we need to do the following on charge freeze service call.

* Enable charge last, demand will be covered by solar first
* Disable grid feed in, prevents export
* Set discharge current to 0A

This will ensure that...

If there is a short fall of solar, the grid will supply the demand.
If there is an excess of solar, demand will be met by the solar, the extra will charge the battery

brickatius, 2025-04-15:
I think I've cracked it (almost)!
Given that the crucial thing about 'charge_freeze_service' is that the battery doesn't discharge to met any house load and any shortfall is drawn from the grid, how about this? 

Create a number helper to check whether the solar output is sufficient to meet the house load - Solar output minus Home consumption. 

```
{{ states("sensor.lux_solar_output_live") | float - states("sensor.lux_home_consumption_live") | float }}

number.lux_predbat_solar_home

```

Create an automation as follows to constantly monitor the state of the helper. If the value is above 0  do nothing - we are in normal 'Demand' mode.  If the value falls below 0, switch on 'Charge First' to prevent the battery from discharging and ensure house load is met from the grid. Any solar power will charge the battery.

```
alias: Lux Predbat Freeze Charge
description: >-
  Predbat charge_freeze_service switching. Turn on Charge Priority if Home
  Comsumption is greater than Solar Output.
triggers:
  - trigger: state
    entity_id:
      - number.lux_predbat_solar_home
    for:
      hours: 0
      minutes: 0
      seconds: 30
conditions: []
actions:
  - choose:
      - conditions:
          - condition: numeric_state
            entity_id: number.lux_predbat_solar_home
            below: 0.1
        sequence:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.lux_charge_priority
      - conditions:
          - condition: numeric_state
            entity_id: number.lux_predbat_solar_home
            above: -0.1
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.lux_charge_priority
mode: single

```
I've added a time delay to cope with any 'inverter lag' to confirm that the load is genuinely higher than the solar output, especially in borderline cases. 

Change from 'false' to 'true' in the inverter section of the default apps.yaml

```
inverter_type: LuxPower
  inverter:
   ..................................................
    support_charge_freeze: True
    support_discharge_freeze: True
```
Turn on the automation thus:

```
charge_freeze_service:
    service: automation.turn_on
    entity_id: "automation.lux_predbat_freeze_charge"
```
Turn the automation off as follows:
Note 'charge_stop_service' also turns off ac charge as well charge priority (needed in case it was turned on by the automation)

```
charge_stop_service:
    - service: switch.turn_off
      entity_id: "switch.lux_ac_charge_enable"
    - service: switch.turn_off
      entity_id: "switch.lux_charge_priority"
    - service: automation.turn_off
      entity_id: "automation.lux_predbat_freeze_charge"
```
Now for the (almost) bit - Charge First / Charge Priority will stop the battery from discharging and will pull the whole of the house load from the grid whilst using all of the solar output to charge the battery, whereas the Predbat charge_freeze_service will only draw any shortfall from the grid. This isn't a problem when there is no solar output or the battery SoC is 100% but there is a small extra cost implication otherwise.
 In the official description
 _If there is a shortfall of Solar power to meet house load, the excess house load is met from grid import, but if there is excess Solar power above the house load, the excess solar will be used to charge the battery,_

The extra cost equates to the difference between the solar output that is put into the battery and what would have been used for house load according to the Predbat definition.
  
Unless someone can come up with a better idea, I think that this is about as close as we're going to get but of course it doesn't exactly match what 'charge_freeze_service' does.

Is it worth submitting a Feature Request, that would be specific to LuxPower inverters,  a sort of lux_freeze_charge_service that would only be called if the conditions of 'Charge First' were of benefit to the plan?

closed 2025-11-17 by gcoan