springfall2008/batpred #2198: Many Small Linter Complaints

henrytdsimmons, 2025-04-03:
Not a bug or a feature request, just some housekeeping.

There are many small complaints from my linter:
- Unused imports
- Assigned but never used variables
- Some 'non pythonic' expressions: `if not x in y` rather than `if x not in y` (not PEP 8)

I'm happy to deal with some or all of these if there's any appetite.

gcoan, 2025-04-03:
Trefor always welcomes PR's to contribute to Predbat.

I've done a few minor bug fixes, but mainly do documentation.

Others have done code improvements such as the recent database option,  so feel free to assist !

springfall2008, 2025-04-04:
Sure, go for it :)

tomkentpayne, 2025-04-04:
Since there's already a .pre-commit-config.yaml, if you're fixing linting complaints @henrytdsimmons it may be worth (if @springfall2008 agrees) adding an agreed standard linting and formatting step to that, e.g. `ruff` and/or `black` etc.

In fact looking at the codebase, there's already config for `black`, `isort` and `autoflake`. It sounds like those may not have been used, if e.g. you're finding non- PEP 8 compliant things.

henrytdsimmons, 2025-04-09:
As a starting point, I have added a ruff check for unused imports and run the fix.

springfall2008, 2025-04-19:
I merged your PR, did this resolve the issue?

henrytdsimmons, 2025-04-22:
Yes, it has resolved one of the 3:

- Unused imports ✅ 
- Assigned but never used variables ❌ 
- Some 'non pythonic' expressions: if not x in y rather than if x not in y (not PEP 8) ❌ 

Number 3 seems a bit picky but number 2 seems like good practise to me. If there's any interest then I'm happy to take that too.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

gcoan, 2026-08-21:
@henrytdsimmons are you able to progress a PR to fix the remaining lint issues?

closed 2026-08-27 by springfall2008