springfall2008/batpred #2202: xx% SOC by xx:xx

gbenn324234, 2025-04-04:
Similar to the force charge/force export options, it would be great to be able to say I want my battery to have xx% SOC by a specific time and predbat works itself out as to whether it needs to export or import at achieve this, and work out what time it needs to start doing whatever it needs to do achieve the target.

-

Use case would be having an off-peak import starting at midnight but I want to ensure the battery is at 5% by 23:59 so export down to 5% so it can fill up from fresh on the off peak.

Or you know you have a large load coming coming up at a certain time and you want to ensure you have a full battery for it.

springfall2008, 2025-04-05:
>Use case would be having an off-peak import starting at midnight but I want to ensure the battery is at 5% by 23:59 so export down to 5% so it can fill up from fresh on the off peak.

This should happen anyhow

>Or you know you have a large load coming coming up at a certain time and you want to ensure you have a full battery for it.

It maybe possible to adapt the bad weather feature to do this, as it essentially takes a time range and a required battery level

gcoan, 2025-04-05:
> Or you know you have a large load coming coming up at a certain time and you want to ensure you have a full battery for it.

There is an example automation script to tell Predbat about anticipated changes to house load you might want to take a look at https://springfall2008.github.io/batpred/manual-api/#example-solution-to-over-ride-predicted-house-load

The feature does sound useful though, could probably help with ensuring the battery is empty to manage solar clipping #1206 as well

springfall2008, 2026-10-28:
This done now with manual soc

closed 2026-10-28 by springfall2008