springfall2008/batpred #2203: Charging during Freeze Export

gbenn324234, 2025-04-04:
When forced Freeze export is active and there's excess solar my AIO is still charging rather than exporting it. Or rather it splits half of the export between export and charge which is a little odd.

eg. at the moment in Freeze export there's 1.5kw solar, 500ish going to house load, 500w going to battery and 500w going to grid. I would have thought 500w should be going to house and 1kw to export.

Is this a bug or is there something wrong with my settings somewhere?

springfall2008, 2025-04-19:
Not sure, is pause becoming active as expected?

chalfontchubby, 2026-08-20:
Closing since there has been no response to the last request for a while. Please open another ticket if there is still a problem.
There have been some recent changes - but checking your export limit would be useful.

closed 2026-08-20 by chalfontchubby