springfall2008/batpred #2209: Clipping not being modelled on Giv Hybrid v2 - 3.68Kw

ColinSergeant, 2025-04-05:
**Describe the bug**
Clipping not being modelled on 3.68Kw inverter

**Expected behaviour**
Expect battery export

**Predbat version**
8.17.6 
(Has been in the previous few versions too, clipping onlt seen since mid March to evidence.)

**Environment details**

- Giv 3.6 Hybrid v2
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

[predbat_debug_17_25_00.yaml.txt](https://github.com/user-attachments/files/19617046/predbat_debug_17_25_00.yaml.txt)

gcoan, 2025-04-05:
See #1206

Predbat currently has limited support for managing clipping, but is something requested

ColinSergeant, 2025-04-05:
Plenty of people seem to have it working especially on Giv systems, Trefor had asked for a debug from FB. @gcoan

gcoan, 2025-04-05:
Ok.

I haven't seen Facebook.

Some screenshots and the plan would add to this issue to show the specific problem you have noticed.

Equally there are plenty of people (myself included) who have asked for enhancements to clipping handling in predbat as it doesn't do what we need. I have to control the charge rate and add lots of manual force commands to reduce clipping on my givenergy system.  For me predbat does not do any clipping management 😞

ColinSergeant, 2025-04-05:
Here's a screenshot of the plan for reference. Showing the 100% battery and 2.2kw solar... And no clipping 😬

[image]

springfall2008, 2025-04-05:
If you turn down 'metric_min_improvement_export' to something small like 0.1 then I get as expected

ColinSergeant, 2025-04-05:
Perfect that's looking like a good plan.
Is this just because the small difference in the clipping Vs cost per kWh? (It'll get more pronounced nearer June.)

springfall2008, 2025-04-05:
> Perfect that's looking like a good plan. Is this just because the small difference in the clipping Vs cost per kWh? (It'll get more pronounced nearer June.)

Yes

closed 2025-04-12 by springfall2008