springfall2008/batpred #2217: solax has no hold charge freeze charge or freeze discharge

adamrhunt, 2025-04-06:
when predbat goes into holdcharge mode on the plan it switches a solax inverter into charge mode causing it to charge on higher rate slots that you dont want it to. 
 Im not sure if there there is a way to make the inverter hold charge on solax could be added ? as the only commands solax have are is “force charge” (from grid) “force discharge” (to grid) “stop charge and discharge” (stops using battery) “back up mode” (using battery or charging battery from solar) “Feed in priority” ( prioritizes exporting excess solar energy to the grid, even if the battery could be charged,) Could there be a way to use these commands on apps yaml? I have the charge start service using “force charge” and then discharge service using “force discharge” and the stop services using “back up”. But for solax this does not make hold charge work it just makes the inverter charge switching to “force charge”

ive pasted my apps.yaml which all works apart from the hold service is there a way wit solax to make the hold service work?

 pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # XXX: Template configuration, delete this line once you have set up for your system
  

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #

  inverter_type: "SX4"
  num_inverters: 1
  #
  # Controls/status - must by 1 per inverter
  #

  # Max inverter power from battery
  battery_rate_max:
    - 8000
  # Battery capacity in kWh
  soc_max:
    - 14.87

  # Solis specific parameters (these are based on https://github.com/wills106/homeassistant-solax-modbus)

  load_today:
    - sensor.solax_today_house_load
  import_today:
    - sensor.solax_today_s_import_energy
  export_today:
    - sensor.solax_today_s_export_energy
  pv_today:
    - sensor.solax_today_s_solar_energy

  battery_voltage:
    - sensor.solax_battery_voltage_charge

   # This is disabled by default in the Solax integration so it must be manually enabled.
  inverter_time:
    - sensor.solax_rtc

  # This is disabled by default in the Solax integration so it must be manually enabled.
  battery_power:
    - sensor.solax_battery_power_charge
[... 280 lines cut ...]
  # Battery scaling makes the battery smaller (e.g. 0.9) or bigger than its reported
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
    - name: 'large'
      minutes: 60
      energy: 1.0
    - name: 'small'
      minutes: 15
      energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0

springfall2008, 2025-04-06:
Did you mean to close this issue without comment?

The code is meant to change between charging and discharging based on the current SOC level, can you share a logfile where it doesn't do this?

adamrhunt, 2025-04-06:
sorry no i didnt mean to close it 

[predbat.log](https://github.com/user-attachments/files/19620504/predbat.log)

jefft4, 2025-04-06:
Additonal config for investigation. Mine uses solax_modbus in modbus remote control mode (no EEPROM writes, instant control) and I think the implementations of the freeze service calls work when called.

```
 ## Need expert mode enabled to configure battery cycle cost and self-sufficiency bias, which are then set through the UI
  expert_mode: True

  # Inverter
  # JT - combined our two as one; trying to have two was tricky and we can't ignore the non-battery one else predbat doesn't see the true load
  inverter_type: "SX4"
  num_inverters: 1
  output_charge_control: "power"

  # solax_modbus_new: true

  ##JT shouldn't need this if using direct control rather than setting time periods, but it errors without it
  inverter_time: sensor.solax_rtc

  #
  # Controls/status - must by 1 per inverter
  #

  # Max inverter power from battery
  battery_rate_max:
  # per Solax docs for the T58
    - 4000
  # Battery capacity in kWh
  soc_max:
    - 5.8

  # Solis specific parameters (these are based on https://github.com/wills106/homeassistant-solax-modbus)

  load_today:
    - sensor.house_power_consumed_today
  import_today:
    - sensor.solax_today_s_import_energy
  export_today:
    - sensor.solax_today_s_export_energy
  pv_today:
    - sensor.solar_yield_today

  battery_voltage:
    - sensor.solax_battery_voltage_charge

  inverter_time:
    - sensor.solax_rtc

  battery_power:
    - sensor.solax_battery_power_charge
  pv_power:
    - sensor.solax_pv_power_total

  # For Solis inverters we need to add the bypass (backup) load to the main house load. If not used the load_power_1 can be commented out
  load_power:
    - sensor.solar_ac_power

  soc_percent:
    - sensor.solax_battery_capacity
  reserve:
    - number.solax_backup_discharge_min_soc
  battery_min_soc:
    - number.solax_feedin_discharge_min_soc

## JT - Predbat errors without this, so we have a dummy one when using services
#  charge_discharge_update_button:
#     - input_button.predbat_dummy_button

## JT - Services for start/stop - needed for SolaX as the supposed SX4 controls don't seem to achieve anything
  charge_start_service:
  # force charge from grid + solar
    service: input_select.select_option
[... 166 lines cut ...]
  car_charging_energy:
    - sensor.evc_charge_added
    # might need to be the cumulative one, not sure.
    # EVC bounces to zero quite a bit during charging. Possible need to smooth this out? See how it fares.
  car_charging_battery_size:
    - 58
  car_charging_planned:
  # A sensor that says 'we'd like to charge'
  # Could be simply, is there something attached to the charger... or some composite of the charger mode and the car 'charger connected' sensor?
  # The EVC seems to have nothing to say it's connected, so going with the car for now.
    - binary_sensor.id_3_charging_cable_connected
  car_charging_planned_reponse:
     - 'on'
     - 'true'
     # ID.3 state value is on/off, although shows up in UI as 'Plugged in'/'Unplugged', include just in case
     - 'Plugged in'
  car_charging_limit:
    - sensor.id_3_battery_target_charge_level
  car_charging_soc:
    - sensor.id_3_battery_level

```

The issue for me, is that I have a plan showing 'hold chrg', with a target to reduce the battery SoC, but predbat is calling the 'charge' service!
I wonder whether using the SX4 inverter type (Solax in the other control mode has no 'freeze' modes) stops it recognising that I have the freeze services available...

springfall2008, 2025-04-06:
The issue is there is no support charge hold option its always assumed to be working.

Can you add a logfile of the problem?

jefft4, 2025-04-06:
> The issue is there is no support charge hold option its always assumed to be working.
> 
> Can you add a logfile of the problem?
The situation I posted on the FB group around 20:20 on 05/04/25, should be in this log file... but plan debug wasn't on. If you'd like a log with plan debug, I'll create one next time I see a hold with a target SoC <100%

[predbat.log](https://github.com/user-attachments/files/19621164/predbat.log)

springfall2008, 2025-04-06:
So in the logfile it says '2025-04-05 20:20:03.625381: Current SOC 100% is greater than Target SOC 10. Grid Charge disabled.' 
At this point it should set timed_charge_current to a value of 0 which should hold the charge, but for some reason this is commented out in the template so it does nothing

jefft4, 2025-04-07:
> So in the logfile it says '2025-04-05 20:20:03.625381: Current SOC 100% is greater than Target SOC 10. Grid Charge disabled.' At this point it should set timed_charge_current to a value of 0 which should hold the charge, but for some reason this is commented out in the template so it does nothing

If timed_charge_current is a parameter to be sent to the inverter, then that would make sense - Solax in standard control mode doesn't support that. I'll have a look at whether I can adapt the SX4 to create a SX4-modbus-RC; the modbus remote control mode can set the power (not current).

jefft4, 2025-04-07:
Okay, I'm pretty confused now!
All evening until about 21:00, I've had a plan saying 'hold charge' with a reducing SoC target. Every 5 minutes, predbat has called first start_charge then immediately stop_charge (or possibly stop_discharge, can't which).
Now we're in the first of the 'freeze charge' periods - not sure why, we're already at min SoC so it makes no difference - and it called charge_freeze then immediately one of the stop_* services.
@springfall2008 Any idea what's causing that? I would only expect one service to be called each time, not two different ones.

[image]

[image]

[image]

rszemeti, 2025-04-09:
I beleive the Solis inverters use essentially the same software, certainly solax_modbus talks to my Solis S6 99% of the registers. 

On the Solis and I believe the Solax will be the same, "Charge Hold" is achieved by requesting "charge" and setting the target SOC to a value less than or equal to the current SOC,  this causes the inverter to stop charging the battery and the house load will be fed from the grid.

jimbaxter, 2025-04-17:
> So, I've set up my SX4 like this: config.py:
> 
> <snip>
> 
> The input_select changing triggers a HA script that sets the multiple modbus parameters needed to make things happen. If this works, I'll move the code of that script into the service call definitions in apps.yaml.
> 
That looks useful, what is in the HA script you used to set the modbus parameters?

I had managed to control it so far using these service calls to the modbus API's:

```
  energy_control_switch:
    - select.solax_remotecontrol_power_control

  charge_start_service:
    - service: input_number.set_value
      entity_id: input_number.battery_state
      value: "2"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_active_power
      value: "4000"
    - service: select.select_option
      entity_id: select.solax_remotecontrol_power_control
      option: Enabled Battery Control
    - service: number.set_value
      entity_id: number.solax_remotecontrol_duration
      value: "300"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_autorepeat_duration
      value: "28800"
    - service: button.press
      entity_id: button.solax_remotecontrol_trigger
      data: {}

  charge_stop_service:
    - service: number.set_value
      entity_id: input_number.battery_state
      value: "1"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_active_power
      value: "4000"
    - service: select.select_option
      entity_id: select.solax_remotecontrol_power_control
      option: Disabled
    - service: number.set_value
      entity_id: number.solax_remotecontrol_duration
      value: "300"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_autorepeat_duration
      value: "28800"
    - service: button.press
      entity_id: button.solax_remotecontrol_trigger
      data: {}

  discharge_start_service:
    - service: number.set_value
      entity_id: input_number.battery_state
      value: "3"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_active_power
      value: "-4000"
    - service: select.select_option
      entity_id: select.solax_remotecontrol_power_control
      option: Enabled Battery Control
    - service: number.set_value
      entity_id: number.solax_remotecontrol_duration
      value: "300"
    - service: number.set_value
      entity_id: number.solax_remotecontrol_autorepeat_duration
      value: "28800"
    - service: button.press
      entity_id: button.solax_remotecontrol_trigger
      data: {}
```

Though when I try to use the  {power} attribute the services fail to run so it is using an arbitrary value at the moment and I had to create a service to disable the charge if it reaches 100% so that the solar panels do not stop operating.

jefft4, 2025-04-18:
> > So, I've set up my SX4 like this: config.py:
> > 
> > The input_select changing triggers a HA script that sets the multiple modbus parameters needed to make things happen. If this works, I'll move the code of that script into the service call definitions in apps.yaml.
> 
> That looks useful, what is in the HA script you used to set the modbus parameters?
> 
> I had managed to control it so far using these service calls to the modbus API's:
> 
> ```
>   energy_control_switch:
>     - select.solax_remotecontrol_power_control
> 
>   charge_start_service:
>     - service: input_number.set_value
>       entity_id: input_number.battery_state
>       value: "2"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_active_power
>       value: "4000"
>     - service: select.select_option
>       entity_id: select.solax_remotecontrol_power_control
>       option: Enabled Battery Control
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_duration
>       value: "300"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_autorepeat_duration
>       value: "28800"
>     - service: button.press
>       entity_id: button.solax_remotecontrol_trigger
>       data: {}
> 
>   charge_stop_service:
>     - service: number.set_value
>       entity_id: input_number.battery_state
>       value: "1"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_active_power
>       value: "4000"
>     - service: select.select_option
>       entity_id: select.solax_remotecontrol_power_control
>       option: Disabled
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_duration
>       value: "300"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_autorepeat_duration
>       value: "28800"
>     - service: button.press
>       entity_id: button.solax_remotecontrol_trigger
>       data: {}
> 
>   discharge_start_service:
>     - service: number.set_value
>       entity_id: input_number.battery_state
>       value: "3"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_active_power
>       value: "-4000"
>     - service: select.select_option
>       entity_id: select.solax_remotecontrol_power_control
>       option: Enabled Battery Control
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_duration
>       value: "300"
>     - service: number.set_value
>       entity_id: number.solax_remotecontrol_autorepeat_duration
>       value: "28800"
>     - service: button.press
>       entity_id: button.solax_remotecontrol_trigger
[... 205 lines cut ...]
    entity_id: input_select.solar_control_mode
    option: "Charge freeze"
    repeat: true
  discharge_freeze_service:
  # Export freeze: ALLOW discharge to load (but don't force export), NO grid charge, NO solar charge.
    service: input_select.select_option
    entity_id: input_select.solar_control_mode
    option: "Discharge freeze"
    repeat: true
  discharge_start_service:
  # force discharge (i.e., force export)
    service: input_select.select_option
    entity_id: input_select.solar_control_mode
    option: "Discharge"
    repeat: true
  discharge_stop_service:
  # return to normal operation (self-use mode)
    service: input_select.select_option
    entity_id: input_select.solar_control_mode
    option: "Self-use"
    repeat: true

```

I think I have the right modbus actions for each predbat service call, although it may be that for some, predbat really wants to specify a SoC. What's bugging me, is that predbat appears to be calling the wrong service for what the plan says it wants to achieve!

rszemeti, 2025-04-18:
I did begin looking through PredBat's plan class with the idea of hooking it up to some dummy data and looking at the plan output and seeing if I could optimise the code further as it is proposing some weird things at times for me ... but I got lost in the code ... there are some incredibly loooooooong methods in the class, my mum always told me any method longer than a page needed refactoring, so I've made a start on that, but I really do need to hang a few more unit tests around it.

jefft4, 2025-04-21:
Another sequence of madness... Predbat seems to issue multiple service calls, some of them completely contrary to what it plans to be doing. Perhaps the inverters it was written for, need to be told 'stop this, do that' rather than just 'now do that', but some of these sequences of service calls don't make sense even then!

This is it trying to set charge mode:

[image]

And this is coming out of charge, back into demand/self-use:

[image]

Both are just way too many calls. The first wants to be simply 'charge' and the second, 'stop charge'.

jimbaxter, 2025-05-03:
> Another sequence of madness... Predbat seems to issue multiple service calls, some of them completely contrary to what it plans to be doing. Perhaps the inverters it was written for, need to be told 'stop this, do that' rather than just 'now do that', but some of these sequences of service calls don't make sense even then!
> Both are just way too many calls. The first wants to be simply 'charge' and the second, 'stop charge'.

I noticed the same thing, it results in the predbat Plan stating charge but the battery not charging. I guess it  is a bug.

It also shows in the logs:
```
2025-05-03 16:09:10.751486: Adjust demand (idle) time computed is 16:00:00-23:59:00
2025-05-03 16:09:10.751671: Inverter 0 current discharge rate is 2600.0W and new target is 5000W
2025-05-03 16:09:10.751739: Setting charging SOC to 100 as per target
2025-05-03 16:09:10.751809: Inverter 0 adjust target soc for charge to 100.0% based on requested all inverter soc 100%
2025-05-03 16:09:10.751882: Inverter 0 Current Target SOC is 100%, already at target
2025-05-03 16:09:10.751992: Current SOC 91% is less than Target SOC 100. Grid charging enabled with charge current set to 65.00
2025-05-03 16:09:10.752198: Inverter 0 Calling service discharge_stop_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.solar_control_mode', 'option': 'Self-use'}
2025-05-03 16:09:10.822819: Inverter 0 Calling service charge_start_service domain charge service_name input_select/select_option with data {'entity_id': 'input_select.solar_control_mode', 'option': 'Charge'}
2025-05-03 16:09:10.870794: Inverter 0 Calling service discharge_stop_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.solar_control_mode', 'option': 'Self-use'}
2025-05-03 16:09:10.935046: Inverter 0 count register writes 0
2025-05-03 16:09:11.092165: Total inverter register writes now 0
2025-05-03 16:09:41.273027: Wrote debug yaml to /addon_configs/6adb4f0d_predbat/debug/predbat_debug_15_55_00.yaml
2025-05-03 16:09:41.274036: Completed run status Charging
2025-05-03 16:09:41.320800: Info: record_status Charging target 91%-100%
```