springfall2008/batpred #2218: Adaptions for better Huawei support

Rhysgh, 2025-04-06:
Hi, i have been looking to resolve the clipping issue for Huawei inverters as they seem to operate different to most others. 

when you force charge or more importantly discharge the battery using the huawei_solar/forcible_discharge_soc service it take priority over the PV production so i dont want to use that in the case where clipping is the potental. 

before predbat what i did was change the battery mode between max_self_consumption and  fully_fed_to_grid. the latter sends as much power as possabel to the gird starting with PV and topping up with battery. 

i tried to trace how predbat worked and i came up with this that im trying to test on my end. i did have the select.select_option incorrect ly defined so i just fixed that now but i cant find a way to force this to try the code out.. 

in apps.yaml i added this, "battery_option_service: " to the template. 

`# Services to charge/discharge
charge_start_service: 
  service: huawei_solar/forcible_charge_soc
charge_stop_service: 
  service: huawei_solar/stop_forcible_charge
discharge_start_service: 
  service: huawei_solar/forcible_discharge_soc
discharge_stop_service: 
  service: huawei_solar/stop_forcible_charge

battery_option_service: 
  service: select.select_option
  target:
    entity_id: select.batteries_working_mode
  data:
    max_self_consumption: "maximise_self_consumption"
    fftg: "fully_fed_to_grid"`

and in the inverter.py file i added /amended this from line 2145

`    def adjust_export_immediate(self, target_soc, freeze=False):
        """
        Adjust from exporting or not exporting based on passed target soc
        """
        # Retrieve values from apps.yaml
        max_self_consumption = self.get_app_config('battery_option_service', 'data', 'max_self_consumption')
        fftg = self.get_app_config('battery_option_service', 'data', 'fftg')

        # Prepare basic service data for stopping services
        service_data_stop = {"device_id": self.base.get_arg("device_id", index=self.id, default="")}
        extra_data = {
            "discharge_start_time": self.base.get_arg("discharge_start_time", index=self.id, default="00:00:00"),
            "discharge_end_time": self.base.get_arg("discharge_end_time", index=self.id, default="00:00:00")
        }

        # Adjust based on target_soc
        if target_soc > 0 and target_soc < 100:
            service_data = {
                "device_id": self.base.get_arg("device_id", index=self.id, default=""),
                "target_soc": target_soc,
                "power": int(self.battery_rate_max_discharge * MINUTE_WATT),
            }

            # Stop charge
            if not self.call_service_template("battery_option_service", {"option": max_self_consumption}, domain="select"):
                self.call_service_template("charge_stop_service", service_data_stop, domain="charge")

            # Start discharge or discharge freeze
            if freeze:
                # First check for discharge_freeze_service
                if not self.call_service_template("discharge_freeze_service", service_data, domain="discharge", extra_data=extra_data):
                    # If discharge_freeze_service isn't available, check for battery_option_service
                    if not self.call_service_template("battery_option_service", {"option": fftg}, domain="select"):
                        # If neither discharge_freeze_service nor battery_option_service is available, start discharge
                        self.call_service_template("discharge_start_service", service_data, domain="discharge", extra_data=extra_data)
            else:
                # If not freeze, simply start the discharge
                if not self.call_service_template("discharge_freeze_service", service_data, domain="discharge", extra_data=extra_data):
                    if not self.call_service_template("battery_option_service", {"option": fftg}, domain="select"):
                        self.call_service_template("discharge_start_service", service_data, domain="discharge", extra_data=extra_data)

            # Before stopping discharge or charge, set the battery mode to max_self_consumption
            self.call_service_template("battery_option_service", {"option": max_self_consumption}, domain="select")

            # Stop discharge or switch to charge if necessary
            if not self.call_service_template("discharge_stop_service", service_data_stop, domain="discharge"):
                self.call_service_template("charge_stop_service", service_data_stop, domain="discharge")
`

do you have any input on this? can you help me refine this? once i have the matter mode working correctly i can address the freeze service as that is also something we can utalise

Rhysgh, 2025-04-06:
@springfall2008 i would appreciate your input on this one

Rhysgh, 2025-04-07:
i have tried hard coding the options but i still get this error ... Warn: Service call select/select_option data {'target': {'entity_id': 'select.batteries_working_mode'}, 'data': {'max_self_consumption': 'maximise_self_consumption', 'fftg': 'fully_fed_to_grid'}} failed with response None

Rhysgh, 2025-04-07:
> > i have tried hard coding the options but i still get this error
> 
> What happens if you call the service to set the inverter mode from developer tools rather than via predbat?

this works ```action: select.select_option
target:
  entity_id: select.batteries_working_mode
data:
  option: maximise_self_consumption```

springfall2008, 2025-04-12:
I don't really understand what you are doing here, without changing the predbat code why don't just call the select option server direct from discharge_start/freeze options e.g:

```
discharge_start_service:
   service: select.select_option
   entity_id: select.batteries_working_mode
   option: "maximise_self_consumption"
```

emilio-valdes, 2025-12-14:
I'm not sure my issue could be related to this, perhaps it's not. I have my huawei setup working really well with Predbat, thanks to @gcoan support. I use the script with the action: huawei_solar.forcible_charge_soc to 100%, when Predbat decides there's enough charge in the battery, it triggers the action: huawei_solar.stop_forcible_charge succesfully. When Predbat decides hold charge, nothing happens, no action is triggered. When predbat decides charge to 100% and hold the battery fully charged, then it triggers the action huawei_solar.forcible_charge_soc to 100%, and when the battery reaches soc+100% the forcible charge action stops, and the battery stars to discharge ignoring predbat instruction to hold the soc=100%. No errors are reported in the log, and the battery continues to deplet. 

THIS IS THE PLAN

[image]

THIS IS THE REALITY

[image]

springfall2008, 2026-08-29:
🤖 This is an automated first-pass triage. A maintainer will review before any labels/actions are treated as final.

**Classification:** enhancement / question (existing `question` label kept as-is)
**Priority:** not set (feature request / mixed thread, no single actionable bug remains)

## Investigated against current main (v8.53.5-21-g76fc2e0d)

This thread has drifted across two distinct asks and it's worth separating them:

**1. Original request (Rhysgh, Apr 2025) — a `battery_option_service` to switch the Huawei working-mode select (e.g. to "fully fed to grid") during discharge, to work around clipping.** This capability still doesn't exist in core Predbat — there's no built-in service hook for a working-mode select. The maintainer's original suggestion stands and requires no core change: wire the mode switch directly into your `discharge_start_service`/`discharge_freeze_service`/`charge_stop_service` templates in `apps.yaml` (calling `select.select_option` on your working-mode entity), rather than patching `inverter.py`. If first-class support for this is wanted, it'd need a new inverter capability (akin to `has_reserve_soc`/`has_timed_pause` in `config.py`'s `INVERTER_DEF`), which hasn't been scoped.

**2. Later derailment (emilio-valdes, Dec 2025; pedromrg, Mar 2026) — Huawei battery discharges immediately after a forced charge to 100% completes, ignoring the plan's "hold" period.** This is the exact symptom already reported, root-caused and fixed in #3106 (closed 2026-03-24). The fix landed in commit `7494167f` ("#3106: Huwaei slow rate charge and hold charge at 100%"), released in **v8.35.1** and present in all versions since (confirmed still in current main, `config.py:2014` sets `charge_discharge_with_rate: True` only for the `HU` inverter type). With that flag set, `execute.py:295-297` now proactively writes `adjust_discharge_rate(0)` (→ `number.batteries_maximum_discharging_power` = 0W) for the whole duration of any charge/hold window — independent of whether `huawei_solar.forcible_charge_soc` self-terminates at its target SoC. This was the missing piece: previously nothing held the battery once the forcible-charge service stopped itself at 100%.

I ran `apps/predbat/tests/test_execute.py` (`tools/triage_test.sh execute`) against current main — it passes, including the sub-tests added for `inv_charge_discharge_with_rate` (charge/export freeze and hold scenarios).

**For @emilio-valdes and @pedromrg:** if you're still seeing the discharge-after-full-charge behaviour on a Predbat version ≥ v8.35.1 with an up-to-date `templates/huawei.yaml` (using `power: "{power}"`, not a fixed wattage), please open that as its own bug report with your current version and a fresh log — it would be a regression, not the original #3106 issue. Otherwise this specific symptom looks resolved on main.

The original working-mode-switch enhancement (item 1) remains open and unimplemented in core.