springfall2008/batpred #2219: apps.yaml entity linked to an Incorrect entity in predbat

SwiftRR, 2025-04-06:
**Describe the bug**
A clear and concise description of what the bug is.

This morning the predbat UI apps.yam; checker came up with this error which I had never seen before.
Last night, I upgraded from v8.17.5 to v8.17.6 
[image]
```
My apps.yaml does have this entity

In my apps.yaml, I had the lines below:
```
 inverter_type:
    - "GE"
  num_inverters: 1
```
I checked my apps.yaml and the related 'schedule' entities are both givTCP switches, whereas the error line points to a predbat sensor.
apps.yaml:
``` 
  scheduled_charge_enable:
    - switch.givtcp_{geserial}_enable_charge_schedule
  scheduled_discharge_enable:
    - switch.givtcp_{geserial}_enable_discharge_schedule
```
I commented out the GE line in my apps.yaml and the error message went away. But the incorrect entity was still there.

So 
```
  scheduled_discharge_enable:
    - switch.givtcp_{geserial}_enable_discharge_schedule
```
is linking to:
```
  scheduled_discharge_enable:
    - sensor.predbat_ge_0_scheduled_discharge_enable

```
Both entities have the same friendly name.

The related 'scheduled charge' line in apps.yaml is linked to the correct givTCP entity.

The correct givTCP entity is there and the switch is changing from time to time. There is no activity with the 'ge' entity.

In Devices & Services/entities, I have found 3 'ge' entities which may all be dummy entities.

<img width="566" alt="Image" src="https://github.com/user-attachments/assets/90651da8-ee90-4877-bc98-5053bc35cf55" />

**Expected behaviour**
Entities in apps.yaml being linked to correct entities in running code. 

**Predbat version**
v8.17.6

**Environment details**

- Inverter and battery setup GE AC3.0 with two 8.2 kWh batteries
- Standard HAOS installer HA 2025.3.4 and 15.0 on Raspberry Pi5

In Devices & Services/Entities I have found 3 'ge' entities which could all be 'dummy entities'

**Screenshots**
See above

**Log file**

**Predbat debug yaml file**
I can provide log files and debug file but I do not think they will help. Predbat is picking up or creating the wrong entities.

Rob

gcoan, 2025-04-06:
@springfall2008 same on mine Trefor

apps.yaml points to givenergy sensors:
[image]

but predbat web console says that its using dummy predbat entities for this config item:
[image]

both the predbat and the givtcp entities exist ok and are available:
[image]

Looks like the GE inverter type is incorrectly believing that the inverter doesn't have a discharge enable switch and so is ignoring the apps.yaml config

gcoan, 2025-04-10:
if you manually set sensor.predbat_GE_0_scheduled_discharge_enable = off then the apps.yaml warning will go away (which is what I have)

still means that Predbat isn't controlling the scheduled discharge enable control in givtcp, but as far as I have observed predbat is controlling the inverter absolutely fine like this

springfall2008, 2025-04-12:
The dummy entity is as expected, as Predbat doesn't use scheduled discharge enable instead it uses the GivTCP inverter mode setting.

The only question is why is it highlighted as an error, in mine its just fine?

matttheotter, 2025-04-12:
Just updated to 1.2.7 and its no longer highlighted 🙃

gcoan, 2025-04-12:
> The dummy entity is as expected, as Predbat doesn't use scheduled discharge enable instead it uses the GivTCP inverter mode setting.

The givtcp apps.yaml configures predbat to use the scheduled discharge enable switch, which I'd always assumed it used.

So if predbat doesn't use this then we should take it out of the template apps.yaml

> The only question is why is it highlighted as an error, in mine its just fine?

Seems to be linked to using inverter type of GE in apps.yaml?  That's what @SwiftRR seemed to find

johnleeshore, 2025-04-13:
I’ve had the same issue but didn’t think it was a problem as everything is working fine

LeeFarm11, 2025-04-13:
For 3 phase inverter, changing the mode is not an option. There are no modes on the 3 phase inverters. Changing mode does not work, so this switch is needed. I did raise it previously here -

https://github.com/springfall2008/batpred/issues/2057#issuecomment-2692441839

and here -

https://github.com/springfall2008/batpred/issues/2057#issuecomment-2695586515

Here might be the best post to start from -

https://github.com/springfall2008/batpred/issues/2057#issuecomment-2705980221

closed 2026-08-28 by chalfontchubby