springfall2008/batpred #2223: XLOAD not in documentation

pacemaker82, 2025-04-08:
I have added xload-column to Predbat Table Card some time back, but after refactoring my documentation, I find no documentation on this column anymore:

https://springfall2008.github.io/batpred/search.html?q=XLoad

Did this disappear from the plan at some point? In my code I am referencing "XLoad" as the column name in the Predbat plan.

cc @gcoan

springfall2008, 2025-04-12:
Good point, XLoad is the external/extra load added to Predbat rather than the internally calculated load

pacemaker82, 2025-04-13:
small type `Xload wWh` should be Xload kWh` on https://springfall2008.github.io/batpred/predbat-plan-card/#understanding-the-predbat-plan

gcoan, 2025-04-13:
> small typo `Xload wWh` should be `Xload kWh` on https://springfall2008.github.io/batpred/predbat-plan-card/#understanding-the-predbat-plan

I have fixed this in my PR #2220

closed 2025-04-13 by gcoan