springfall2008/batpred #2227: Predbat is setting a potentially 'unsafe' register during freezes

mpartington, 2025-04-10:
**Describe the bug**
As I've reported in #2124 , Predbat is constantly switching freeze export on and off. A previously unnoticed consequence is Predbat also changes the SOC Target.

There appears to be no need to do this for GivEnergy inverters, as if you command the same pause natively through GivTCP it does not change this register.

The issue is that this has not been announced as a safe register in the RTC firmware, so I am racking up unsafe writes to flash at much higher rates. In fact Mark Cracknell has just confirmed it is 'unsafe'

Now most of this is to do with the bug I reported on the other issue, but ultimately there is no need to be changing this

**Expected behaviour**
SOC Target not to be changed when pause charge or pause export are commanded on inverters that support Pause Mode

**Predbat version**

V8.18

**Environment details**

- GivEnergy AC3 on A210 D210 firmware
- Standard HAOS installer
- Anything else

**Screenshots**

[image]

[image]

mpartington, 2025-04-11:
I also see Predbat sets soc target to 4% on forced export.  Again GivTCP does not do this, it leaves it at 100% (which works). 
Chart shows it going back into demand and soc changing from 4 to 100% then back to 4%. SOC reset was off

[image]

springfall2008, 2025-04-12:
The reason for this is some inverters use the target SOC for the target export, I can probably configure this out for GE

mpartington, 2025-04-12:
> The reason for this is some inverters use the target SOC for the target export, I can probably configure this out for GE

Is this the same reason it changes for a freeze export as well? Hopefully the register is 'safe'. I think the bigger issue is why are freeze export slots so unstable. Seems to be a lot reporting similar (referenced.issue). Setting freeze export only helps a lot

springfall2008, 2025-04-12:
Fix on main for this

WyndStryke, 2025-04-12:
@springfall2008 

> The reason for this is some inverters use the target SOC for the target export, I can probably configure this out for GE

Would it be possible to make this configurable (perhaps apps.yaml)?

The alternative is to just run without target_soc being defined.   So a secondary question might be to ask what it is used for & is it needed?  I have  reserve_soc configured (`reserve_soc` = `number.min_soc_on_grid` + `has_reserve_soc: True`).  Reserve Soc seems to be necessary for freeze charge / hold charge on the Fox, but isn't in the template.

Setting target_soc during the day during export periods on the Fox inverters is a problem because (if configured as `charge_limit` = `number.max_soc`), number.max_soc is a hard upper limit which blocks all charging, including clipping.  I tried setting `switch.predbat_inverter_soc_reset` on, but it didn't help.

This is the configuration that I was trying.  The max_soc seems to be the most appropriate way to configure target_soc on the Fox inverters, as far as I can see, but the fact that it blocks clipping being stored in the battery is a headache.  Haven't managed to find an alternative implementation for target_soc, if any fox user could suggest one that would be great ...    

```
...
    has_target_soc: True
...
  # charge_limit needed if target_soc is enabled (for freeze-charging?).  
  charge_limit:
    - number.max_soc

```

mpartington, 2025-04-12:
Working perfectly for me for both forced export and pause export

springfall2008, 2025-04-12:
> [@springfall2008](https://github.com/springfall2008)
> 
> > The reason for this is some inverters use the target SOC for the target export, I can probably configure this out for GE
> 
> Would it be possible to make this configurable (perhaps apps.yaml)?
> 
> The alternative is to just run without target_soc being defined. So a secondary question might be to ask what it is used for & is it needed? I have reserve_soc configured (`reserve_soc` = `number.min_soc_on_grid` + `has_reserve_soc: True`). Reserve Soc seems to be necessary for freeze charge / hold charge on the Fox, but isn't in the template.
> 
> Setting target_soc during the day during export periods on the Fox inverters is a problem because (if configured as `charge_limit` = `number.max_soc`), number.max_soc is a hard upper limit which blocks all charging, including clipping. I tried setting `switch.predbat_inverter_soc_reset` on, but it didn't help.
> 
> This is the configuration that I was trying. The max_soc seems to be the most appropriate way to configure target_soc on the Fox inverters, as far as I can see, but the fact that it blocks clipping being stored in the battery is a headache. Haven't managed to find an alternative implementation for target_soc, if any fox user could suggest one that would be great ...
> 
> ```
> ...
>     has_target_soc: True
> ...
>   # charge_limit needed if target_soc is enabled
>   charge_limit:
>     - number.max_soc
> ```

I've effectively done that by adding a new option called 'target_soc_used_for_discharge' but I changed the defaults for GE to False and the other inverters to True

closed 2025-04-15 by mpartington