springfall2008/batpred #2235: Predbat keeps stalling.

philicibine, 2025-04-13:
**Describe the bug**
It keeps stalling and restarting. I keep seeing this in the logs when it happens 
```Warn: Failed to decode response from http://192.168.0.59:8123/addons/self/info```
Although i have setup the ha_url and ha_key parameters.  Looking at the log, it doesnt seem to complete a run.  It's in some sort of loop.

**Predbat version**
8.18.0

**Environment details**
Hassos on RPi 4 2gb

- Inverter and battery setup - Solax modbus with Solis hybrid inverter and Dowell 10kw battery 
- Standard HAOS installer - Standard

**Log file**
[predbat.log](https://github.com/user-attachments/files/19724652/predbat.log)

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19724815/predbat_debug.yaml.txt)

gcoan, 2025-04-13:
I have a fix for this in PR #2220 

You can either pull ha.py from my PR, or comment ha_url and ha_key out in apps.yaml and it should stop this happening.

The problem is that when you set ha_url the call to the HA supervisor doesn't work and it will error.  Strange that its just started looping your predbat, mine is OK.

springfall2008, 2025-04-13:
I put the PR onto main earlier today for testing

philicibine, 2025-04-13:
Thanks.  Will test and if anything is wrong will let you know.

philicibine, 2025-04-14:
Hi,  I did still get a couple of occurrences of stalling overnight.  Will un-comment ha_key and ha_url for now.

gcoan, 2025-04-14:
> Hi,  I did still get a couple of occurrences of stalling overnight.  Will comment out ha_key and ha_url for now.

looking again at the errors you are getting (assume they are the same), its that the web socket connection to HA can't be made, that the connection is closed.

I'm wondering whether the issue is the 2Gb memory on your Pi4 and whether you are just running out of memory running Predbat, Solax modbus, plus whatever else you are running?

Are there any addons that you could stop or integrations you could disable to see if this reduces the errors you get?

Is there any errors in the supervisor or host logs?

philicibine, 2025-04-14:
I just caught it when it error'd and i see this in the supervisor logs 
```ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:5052 ssl:default [Connect call failed ('172.30.33.4', 5052)]```

Seeing this in the host logs
```2025-04-14 15:29:32.608 homeassistant kernel: hassio: port 10(veth064c2f7) entered blocking state
2025-04-14 15:29:32.608 homeassistant kernel: hassio: port 10(veth064c2f7) entered disabled state
2025-04-14 15:29:32.609 homeassistant kernel: veth064c2f7: entered allmulticast mode
2025-04-14 15:29:32.612 homeassistant kernel: veth064c2f7: entered promiscuous mode
2025-04-14 15:29:32.612 homeassistant kernel: hassio: port 10(veth064c2f7) entered blocking state
2025-04-14 15:29:32.612 homeassistant kernel: hassio: port 10(veth064c2f7) entered forwarding state
2025-04-14 15:29:32.612 homeassistant NetworkManager[507]: <info>  [1744644572.6114] manager: (veth064c2f7): new Veth device (/org/freedesktop/NetworkManager/Devices/41)
2025-04-14 15:29:32.688 homeassistant kernel: eth0: renamed from veth8354981
2025-04-14 15:29:32.703 homeassistant NetworkManager[507]: <info>  [1744644572.7035] device (veth064c2f7): carrier: link connected```

gcoan, 2025-04-14:
a pi4 with 4gb should be absolutely fine, so that's not the issue.

The host and supervisor log don't really give much of a clue, other than the connection has failed

springfall2008, 2025-04-15:
Does Predbat restart correctly after this error and connect again?

philicibine, 2025-04-15:
It seems to connect again.. but then loses connection again after a short time.  Like I said, it seems to get stuck in a loop

closed 2026-08-28 by chalfontchubby