springfall2008/batpred #2237: 8.18.0 not producing 'good stable' plans

gcoan, 2025-04-13:
**Describe the bug**
I know it's a vague title but since upgrading to 8.18.0 I have been concerned by some of the plan behaviours Predbat has devised.

On 8.17.10 I was getting pretty good plans, exporting remaining charge before/at sunrise (to reduce clipping, although the exports were not always big enough), and then charge during the day.

8.18.10's plan last night for this morning was to export the battery and then export all morning, relying on charging in the afternoon (which turned out to be overcast and raining):
[image]

And then a minute later the plan changed to just a first thing export:
[image]

This morning it appeared that Predbat was executing this latter plan:
[image]

But during the afternoon a big chunk of freeze export was applied, resulting in the batteries not charging fully

Tonight Predbat planned to fully export what remained in the battery, then partially re-fill,  but looking at the rates including losses, this would lose money:
[image]

And then half an hour later the plan looks more sensible:
[image]

**Expected behaviour**
Hard to describe, but more sensible plans, not exporting unnecessarily, and in particular holding onto the stored battery charge as long as possible before exporting it.  I'm ideally only going to see an export first thing in the morning before the sun gets up to empty the batteries to reduce clipping the next day.

**Predbat version**
8.18.0

**Environment details**
HAOS 2024.3.4
2x Gen 1 hybrid inverters each with their own battery

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-04-13:
I had a look but the debug yaml shows the good plan, can you capture a bad one? Ideally update to 'main' first and see if it helps too.

springfall2008, 2025-04-13:
This is what I got:

[image]

gcoan, 2025-04-13:
I'll try to capture a debug file from a 'bad' plan

My current plan from 8.18.0 I was quite happy with, it had freeze charging through the cheap cosy period then force export before the start of the solar day.
I forgot to paste it here before upgrading to main:

[image]

This plan is OK as well.

I then changed back to 8.18.0 to get a screenshot, but the plan wasn't as good as it originally was:

[image]

The charge at 22:00 to later on export isn't profitable, and debateable if the 6:30am charge is needed also

gcoan, 2025-04-13:
Maybe spoke too soon about main being 'good'

Increased load_filter_threshold to see if that reduced the number of load history errors, and the plan produced on main, well is not 'good' - the charge at 06:00 from 4% to 10% isn't required as it freeze exports afterwards and then discharges from 10 to 7%.

[image]

[predbat_debugmainv2.txt](https://github.com/user-attachments/files/19727239/predbat_debugmainv2.txt)

gcoan, 2025-04-14:
Plan execution overnight looked OK, mixture of demand and exporting

[image]

The charge and then export didn't happen and there's none of that planned for this morning

gcoan, 2025-04-14:
> I ran this plan myself and got a different result, the change I made to main at around 7pm last night seemed to have an impact

Your plan looks more like what I'd expect; possibly could have exported a bit more in the morning, but no unexpected charges.

I swapped back and forth between main and 8.18.0, but this was all at around 10pm so should have included your latest changes. 

I'll leave it on this version and see what's planned for tonight/ tomorrow

gcoan, 2025-04-16:
I've given it a bit, to decide how I feel about the plans being produced.

Have upgraded to 8.18.2 (latest version)

This is the plan earlier this evening, not a good plan.  Despite the battery being full, planning a charge this evening and then to export all morning which will produce a load of clipping for me:
[image]
[image]

Looking at the plan now, its better:
[image]

But why export down to empty at 4am, OK then run on cheap import is fine enough, but why charge at 5:30, freeze export and freeze charge over the following slots.  It's just weird behaviour; the charge and freeze charge then have to be exported again.  Surely better to just freeze export or leave exporting all the way through to the end of the cheap rate period.
Similarly 9:30 charges then exports again

As I was typing this in, the plan has just changed, a bit better, no charging now,  but still a 6:30 freeze charge that then has to be exported again

gcoan, 2025-04-17:
Another example of strange charge then export.  I wonder if this one is due to the rate export override from 4pm (to reduce potential DFS period exporting)?

[image]

[predbat_debug3.txt](https://github.com/user-attachments/files/19791273/predbat_debug3.txt)

[predbat.log](https://github.com/user-attachments/files/19791275/predbat.log)

springfall2008, 2025-04-19:
Can you feedback on version v8.18.7 which includes a new swap optimisation?

gcoan, 2025-06-03:
Closing this one as there’s been quite a few versions since

closed 2025-06-03 by gcoan