springfall2008/batpred #2241: Extra SoC accuracy on predbat plan

gcoan, 2025-04-15:
**Describe the bug**
Noticed the debug rate SoC target/limit for this slow rate export is 16.3 and 11.5%
[image]

A bit too precise?

**Expected behaviour**
Probably just missing a round(0) to show no decimal place, making this consistent with everywhere else SoC is displayed 

I also noticed that the debug version of the SoC target isn't explained in the documentation https://springfall2008.github.io/batpred/predbat-plan-card/#debug-mode-for-predbat-plan

**Predbat version**
main, 8.18.1

**Environment details**
HAOS
Givenergy gen 1

**Log file**
Can supply if required

**Predbat debug yaml file**
Can supply if required

springfall2008, 2025-04-15:
Interesting .3 means export at 70% rate while .5 is at 50% speed....

gcoan, 2025-04-15:
> Interesting .3 means export at 70% rate while .5 is at 50% speed....

ah ok, so this is intentional?

probably better document it then as I thought it was a bug

thanks

springfall2008, 2025-04-19:
Good point

gcoan, 2025-09-21:
Done, documentation updated. Will be in my next PR