springfall2008/batpred #2247: 8.18.2 - Completely Broken

ProximusAl, 2025-04-15:
Predbat crashed. Restarting in 5 seconds...
Starting Predbat standalone...
starting startup.py...........
Predbat files are installed correctly for version v8.18.2
**** Starting Standalone Predbat ****
2025-04-15 23:11:09.538331: Loading apps.yaml
2025-04-15 23:11:09.547298: Error: Exception raised can only concatenate str (not "NoneType") to str
2025-04-15 23:11:09.547825: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1189, in initialize
    self.ha_interface = HAInterface(self)
                        ^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 86, in __init__
    res = self.api_call("/addons/self/info", core=False, silent=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 532, in api_call
    "Authorization": "Bearer " + key,
                     ~~~~~~~~~~^~~~~
TypeError: can only concatenate str (not "NoneType") to str
2025-04-15 23:11:09.548003: Error: set_state_wrapper - No HA interface available
2025-04-15 23:11:09.548015: Info: record_status Error: Exception raised can only concatenate str (not "NoneType") to str
Error: Failed to initialize predbat can only concatenate str (not "NoneType") to str
Traceback (most recent call last):
  File "/config/hass.py", line 49, in main
    p_han.initialize()
  File "/config/predbat.py", line 1238, in initialize
    raise e
  File "/config/predbat.py", line 1189, in initialize
    self.ha_interface = HAInterface(self)
                        ^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 86, in __init__
    res = self.api_call("/addons/self/info", core=False, silent=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 532, in api_call
    "Authorization": "Bearer " + key,
                     ~~~~~~~~~~^~~~~
TypeError: can only concatenate str (not "NoneType") to str
2025-04-15 23:11:09.548362: Stopping Predbat
Bootstrap Predbat
Startup
Shutdown, sleeping 30 seconds before restarting
Predbat crashed. Restarting in 5 seconds...

gcoan, 2025-04-15:
Please can you fill in the bug template, it makes it a lot easier to work out what is happening!

My guess is that you are running Predbat in a docker container and it's falling foul of this bit of code I have recently changed in ha.py:

```
            # make an API call to the Supervisor using the supervisor token
            url = "http://supervisor" + endpoint
            key = os.environ.get("SUPERVISOR_TOKEN", None)

        headers = {
            "Authorization": "Bearer " + key,
            "Content-Type": "application/json",
            "Accept": "application/json",
```

I had asked people with docker installs to test this, and didn't get any failures reported, but I'm guessing that SUPERVISOR_TOKEN isn't set in docker.

If you can set this environment variable, even temporarily it will stop the crash

otherwise, edit ha.py and change the line 'key = ' to:

```
           key = os.environ.get("SUPERVISOR_TOKEN", "notoken")
```

and that should fix it as a quick hack

@springfall2008 this is an unexpected path through the code change I made to determine the slug id and fix the web socket issue.  Setting a dummy supervisor token isn't elegant but will stop the string concatenation failure

better fix, change to something like this to log the issue and return:

```
            # make an API call to the Supervisor using the supervisor token
            url = "http://supervisor" + endpoint
            key = os.environ.get("SUPERVISOR_TOKEN", None)

        if not key
            if not silent 
                self.log("Warn: api_call failure, key not set")
                self.api_errors += 1

            data = None
            return data
```

ProximusAl, 2025-04-16:
To be fair, it was 11.30pm at night, my car was draining my batteries, and with predbat crashing, it wasnt going to get them charged either, so my priorities were restoring predbat.  I copied the error, pasted it in, and immediately restored a predbat backup to get me out of a mess.

Yes, I use everything as dockers:
HA is a docker, running CORE.
PredBat is a docker

HA Core doesnt have supervisor, so perhaps this may be the issue?

Predbat basically runs my entire house, so when it crashes out like this, it's a major issue for me.

Apologies for not filling out a bug report, but hopefully I can give you any info you need now.

ProximusAl, 2025-04-16:
I should add, I restored back to 8.18.0 and it's working again....with no changes....

gcoan, 2025-04-16:
> I should add, I restored back to 8.18.0 and it's working again....with no changes....

the changes to ha.py were introduced in 8.18.1, so this and .2 will probably crash on docker

would be helpful if you could confirm the fix I suggested does work on docker. I think it will, but don't have a test rig myself

ProximusAl, 2025-04-16:
So just to be clear you want me to upgrade to 8.12.2 again so it breaks and modify ha.py:
           key = os.environ.get("SUPERVISOR_TOKEN", "notoken")

Right?

gcoan, 2025-04-16:
> So just to be clear you want me to upgrade to 8.12.2 again so it breaks and modify ha.py:
>            key = os.environ.get("SUPERVISOR_TOKEN", "notoken")
> 
> Right?

if you could, it would help confirm that this fix (bodge) works on docker.

Thanks

nipar4, 2025-04-16:
Tried to upgrade to 8.18.2 this morning and got this error as well

Setting the environment variable resolves this and gets the addon running again

@ProximusAl do you build your own docker image or use the one on the Docker hub?  If so try the 'test' tag and see if it resolves the issue for you

ProximusAl, 2025-04-16:
> > So just to be clear you want me to upgrade to 8.12.2 again so it breaks and modify ha.py:
> > key = os.environ.get("SUPERVISOR_TOKEN", "notoken")
> > Right?
> 
> if you could, it would help confirm that this fix (bodge) works on docker.
> 
> Thanks

Doesnt appear to fix it:

025-04-16 09:00:37.978940: Error: set_state_wrapper - No HA interface available

2025-04-16 09:00:37.978952: Info: record_status Error: Exception raised HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7c0542453a10>: Failed to resolve 'supervisor' ([Errno -2] Name or service not known)"))

Error: Failed to initialize predbat HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7c0542453a10>: Failed to resolve 'supervisor' ([Errno -2] Name or service not known)"))

Traceback (most recent call last):

  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 203, in _new_conn

    sock = connection.create_connection(

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 60, in create_connection

    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):

               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3.12/socket.py", line 963, in getaddrinfo

    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):

               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

socket.gaierror: [Errno -2] Name or service not known

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 791, in urlopen

    response = self._make_request(

               ^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 497, in _make_request

    conn.request(

  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 395, in request

    self.endheaders()

  File "/usr/lib/python3.12/http/client.py", line 1331, in endheaders

    self._send_output(message_body, encode_chunked=encode_chunked)

  File "/usr/lib/python3.12/http/client.py", line 1091, in _send_output

    self.send(msg)

  File "/usr/lib/python3.12/http/client.py", line 1035, in send

    self.connect()

  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 243, in connect

    self.sock = self._new_conn()

                ^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 210, in _new_conn

    raise NameResolutionError(self.host, self, e) from e

urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPConnection object at 0x7c0542453a10>: Failed to resolve 'supervisor' ([Errno -2] Name or service not known)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 486, in send

    resp = conn.urlopen(

           ^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 845, in urlopen

    retries = retries.increment(

              ^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 517, in increment

    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]

    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7c0542453a10>: Failed to resolve 'supervisor' ([Errno -2] Name or service not known)"))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/config/hass.py", line 49, in main

    p_han.initialize()

  File "/config/predbat.py", line 1238, in initialize

    raise e

  File "/config/predbat.py", line 1189, in initialize

    self.ha_interface = HAInterface(self)

                        ^^^^^^^^^^^^^^^^^

  File "/config/ha.py", line 86, in __init__

    res = self.api_call("/addons/self/info", core=False, silent=True)

          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/config/ha.py", line 545, in api_call

    response = requests.get(url, headers=headers, timeout=TIMEOUT)

               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/requests/api.py", line 73, in get

    return request("get", url, params=params, **kwargs)

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/requests/api.py", line 59, in request

    return session.request(method=method, url=url, **kwargs)

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 589, in request

    resp = self.send(prep, **send_kwargs)

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 703, in send

    r = adapter.send(request, **kwargs)

        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 519, in send

    raise ConnectionError(e, request=request)

requests.exceptions.ConnectionError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x7c0542453a10>: Failed to resolve 'supervisor' ([Errno -2] Name or service not known)"))

2025-04-16 09:00:37.979981: Stopping Predbat

ProximusAl, 2025-04-16:
Just FYI, I've rolled back to 8.18.0 again.

I'm at work so have limited time for debugs....

nipar4, 2025-04-16:
use nipar44/predbat_addon:test

ProximusAl, 2025-04-16:
> use nipar44/predbat_addon:test
> 
> Edit: I've only built it for amd at the moment, so if your on a arm processor (Rasberrypi) let me know

Doesnt work either:

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/urllib3/connection.py", line 198, in _new_conn

    sock = connection.create_connection(

        (self._dns_host, self.port),

    ...<2 lines>...

        socket_options=self.socket_options,

    )

  File "/usr/local/lib/python3.13/site-packages/urllib3/util/connection.py", line 60, in create_connection

    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):

               ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.13/socket.py", line 977, in getaddrinfo

    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):

               ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

socket.gaierror: [Errno -2] Name does not resolve

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/urllib3/connectionpool.py", line 787, in urlopen

    response = self._make_request(

        conn,

    ...<10 lines>...

        **response_kw,

    )

  File "/usr/local/lib/python3.13/site-packages/urllib3/connectionpool.py", line 493, in _make_request

    conn.request(

    ~~~~~~~~~~~~^

        method,

        ^^^^^^^

    ...<6 lines>...

        enforce_content_length=enforce_content_length,

        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

    )

    ^

  File "/usr/local/lib/python3.13/site-packages/urllib3/connection.py", line 445, in request

    self.endheaders()

    ~~~~~~~~~~~~~~~^^

  File "/usr/local/lib/python3.13/http/client.py", line 1333, in endheaders

    self._send_output(message_body, encode_chunked=encode_chunked)

    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.13/http/client.py", line 1093, in _send_output

    self.send(msg)

    ~~~~~~~~~^^^^^

  File "/usr/local/lib/python3.13/http/client.py", line 1037, in send

    self.connect()

    ~~~~~~~~~~~~^^

  File "/usr/local/lib/python3.13/site-packages/urllib3/connection.py", line 276, in connect

    self.sock = self._new_conn()

                ~~~~~~~~~~~~~~^^

  File "/usr/local/lib/python3.13/site-packages/urllib3/connection.py", line 205, in _new_conn

    raise NameResolutionError(self.host, self, e) from e

urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPConnection object at 0x78c044ea7e00>: Failed to resolve 'supervisor' ([Errno -2] Name does not resolve)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "/usr/local/lib/python3.13/site-packages/requests/adapters.py", line 667, in send

    resp = conn.urlopen(

        method=request.method,

    ...<9 lines>...

        chunked=chunked,

    )

  File "/usr/local/lib/python3.13/site-packages/urllib3/connectionpool.py", line 841, in urlopen

    retries = retries.increment(

        method, url, error=new_e, _pool=self, _stacktrace=sys.exc_info()[2]

    )

  File "/usr/local/lib/python3.13/site-packages/urllib3/util/retry.py", line 519, in increment

    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]

    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x78c044ea7e00>: Failed to resolve 'supervisor' ([Errno -2] Name does not resolve)"))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/addon/hass.py", line 49, in main

    p_han.initialize()

    ~~~~~~~~~~~~~~~~^^

  File "/addon/predbat.py", line 1238, in initialize

    raise e

  File "/addon/predbat.py", line 1189, in initialize

    self.ha_interface = HAInterface(self)

                        ~~~~~~~~~~~^^^^^^

  File "/addon/ha.py", line 86, in __init__

    res = self.api_call("/addons/self/info", core=False, silent=True)

  File "/addon/ha.py", line 545, in api_call

    response = requests.get(url, headers=headers, timeout=TIMEOUT)

  File "/usr/local/lib/python3.13/site-packages/requests/api.py", line 73, in get

    return request("get", url, params=params, **kwargs)

  File "/usr/local/lib/python3.13/site-packages/requests/api.py", line 59, in request

    return session.request(method=method, url=url, **kwargs)

           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.13/site-packages/requests/sessions.py", line 589, in request

    resp = self.send(prep, **send_kwargs)

  File "/usr/local/lib/python3.13/site-packages/requests/sessions.py", line 703, in send

    r = adapter.send(request, **kwargs)

  File "/usr/local/lib/python3.13/site-packages/requests/adapters.py", line 700, in send

    raise ConnectionError(e, request=request)

requests.exceptions.ConnectionError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x78c044ea7e00>: Failed to resolve 'supervisor' ([Errno -2] Name does not resolve)"))

2025-04-16 09:14:23.013548: Stopping Predbat

ProximusAl, 2025-04-16:
I've gone back to the :latest docker with 8.18.0 and it now works again:

All files OK

modified apps.yaml present

Starting Predbat standalone...

starting startup.py...........

Predbat files are installed correctly for version v8.18.0

**** Starting Standalone Predbat ****

2025-04-16 09:16:00.203883: Loading apps.yaml

2025-04-16 09:16:00.217608: Info: Connected to Home Assistant at http://192.168.76.2:8123

2025-04-16 09:16:00.218968: Warn: Failed to decode response <Response [404]> from http://192.168.76.2:8123/addons/self/info

2025-04-16 09:16:00.219060: Creating task: <coroutine object HAInterface.socketLoop at 0x7abcda290e50>

2025-04-16 09:16:00.219366: Info: Web Socket task started

2025-04-16 09:16:00.219499: Info: Start socket for url http://192.168.76.2:8123/api/websocket

2025-04-16 09:16:00.219989: Creating task: <coroutine object WebInterface.start at 0x7abcda6bf100>

Web interface started

2025-04-16 09:16:00.222604: Info: Web Socket active

Watching ['./unit_test.py', './config.py', './utils.py', './ha.py', './userinterface.py', './octopus.py', './execute.py', './gecloud.py', './inverter.py', './alertfeed.py', './startup.py', './predheat.py', './download.py', './fetch.py', './solcast.py', './hass.py', './apps.yaml', './plan.py', './db_manager.py', './output.py', './futurerate.py', './prediction.py', './web.py', './energydataservice.py', './predbat.py', './compare.py'] for changes

asnaedae, 2025-04-16:
@nipar4, looks like I'm in the same place as @ProximusAl - 

Added SUPERVISOR_TOKEN: notatoken and errors change from
```
File "/config/ha.py", line 532, in api_call
"Authorization": "Bearer " + key,
```

To

```
File "/usr/local/lib/python3.13/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /addons/self/info (Caused by NameResolutionError("<urllib3.connection.HTTPConnection object at 0x76ecfc04f770>: Failed to resolve 'supervisor' ([Errno -3] Try again)"))
```

Switching between your latest image and the test tag didn't change the behaviour for me

ProximusAl, 2025-04-16:
It's clearly trying to connect to host supervisor on port 80, which my docker is HA Core.  Supervisor *does not exist*.

asnaedae, 2025-04-16:
This looks to be related to _Originally posted by @springfall2008 in https://github.com/springfall2008/batpred/pull/2220#discussion_r2040623571_

If I remove the change from @gcoan within api_call() (force use of ha_url) predbat starts up correctly - will reread the discussion around the change and see if I can spot the issue

nipar4, 2025-04-16:
OK, I might have got a bit happy and not upgraded to 8.18.2 - its not working for me either

@gcoan I have an environment variable set SUPERVISOR_TOKEN = notoken which doesn't resolve the issue

I also tried editing has.py to change the line to "key = os.environ.get("SUPERVISOR_TOKEN", "notoken")"
which makes no difference

gcoan, 2025-04-16:
> OK, I might have got a bit happy and not upgraded to 8.18.2 - its not working for me either
>
> @gcoan I have an environment variable set SUPERVISOR_TOKEN = notoken which doesn't resolve the issue
>
> I [...]

Thanks for trying guys

This is all trying to surpress the error line 

2025-04-16 09:16:00.218968: Warn: Failed to decode response <Response [404]> from http://192.168.76.2:8123/addons/self/info

That you currently get with docker installs 

What I think needs changing is (as outlined above), in HA.PY, if the supervisor token returns None then just log it and return from the api_call routine

jamesremuscat, 2025-04-16:
Manually implementing the change described at the end of https://github.com/springfall2008/batpred/issues/2247#issuecomment-2807752271 has allowed Predbat to start up again (at least for me).

gcoan, 2025-04-16:
> Manually implementing the change described at the end of https://github.com/springfall2008/batpred/issues/2247#issuecomment-2807752271 has allowed Predbat to start up again (at least for me).

Excellent thank you 

Can share the revised bit of code and the startup logfile then we can get this fixed properly

asnaedae, 2025-04-16:
PR raised to solve (works for me)

```
Predbat files are installed correctly for version v8.18.2
**** Starting Standalone Predbat ****
2025-04-16 11:51:51.439517: Loading apps.yaml
2025-04-16 11:51:51.463461: ERROR: ha_key set and just called for supervisor None
2025-04-16 11:51:51.471760: Info: Connected to Home Assistant at http://IP_ADDRESS:8123
2025-04-16 11:51:51.471809: Creating task: <coroutine object HAInterface.socketLoop at 0x77bb813b2f20>
2025-04-16 11:51:51.472024: Info: Web Socket task started
2025-04-16 11:51:51.472224: Creating task: <coroutine object WebInterface.start at 0x77bb801ba5c0>
2025-04-16 11:51:51.472632: Info: Start socket for url http://IP_ADDRESS:8123/api/websocket
Web interface started
2025-04-16 11:51:51.478905: Info: Web Socket active
```

ProximusAl, 2025-04-16:
> PR raised to solve (works for me)
>
> ```
> Predbat files are installed correctly for version v8.18.2
> **** Starting Standalone Predbat ****
> 2025-04-16 11:51:51.439517: Loading apps.yaml
> 2025-04-16 [...]

I can confirm the PR works for me.

ProximusAl, 2025-04-16:
Here is my output, I note I dont get the error you do:

Starting Predbat standalone...

starting startup.py...........

Predbat files are installed correctly for version v8.18.2

**** Starting Standalone Predbat ****

2025-04-16 12:10:51.113330: Loading apps.yaml

2025-04-16 12:10:51.126494: Info: Connected to Home Assistant at http://192.168.76.2:8123

2025-04-16 12:10:51.126524: Creating task: <coroutine object HAInterface.socketLoop at 0x7b74082b0fe0>

2025-04-16 12:10:51.126776: Info: Web Socket task started

2025-04-16 12:10:51.126879: Info: Start socket for url http://192.168.76.2:8123/api/websocket

2025-04-16 12:10:51.127056: Creating task: <coroutine object WebInterface.start at 0x7b74086e23e0>

2025-04-16 12:10:51.129934: Info: Web Socket active

Web interface started

Watching ['./unit_test.py', './config.py', './utils.py', './ha.py', './userinterface.py', './octopus.py', './execute.py', './gecloud.py', './inverter.py', './alertfeed.py', './startup.py', './predheat.py', './download.py', './fetch.py', './solcast.py', './hass.py', './apps.yaml', './plan.py', './db_manager.py', './output.py', './futurerate.py', './prediction.py', './web.py', './energydataservice.py', './predbat.py', './compare.py'] for changes

nipar4, 2025-04-16:
Works for me as well, once its released I'll rebuild all the images

gcoan, 2025-04-16:
> PR raised to solve (works for me)

Thanks f raising this PR, I've had a Quick Look at it and the indentation isn't right and won't work as it's supposed to on HAOS

Have left a review comment

gcoan, 2025-04-16:
> I used chatgpt to fix the indentation :)

Slightly worrying!

ProximusAl, 2025-04-17:
Closing this now as it has been fixied in 8.18.3

closed 2025-04-17 by ProximusAl