springfall2008/batpred #2249: 8.18.2 readonly mode not working, inverter mode being changed

WyndStryke, 2025-04-16:
**Describe the bug**
After updating to 8.18.2 and also installing HAos 15.2, something appears to be changing the inverter mode unexpectedly despite predbat having readonly mode turned on.  Inverter write count not going up, log says that it is in readonly mode. 

For example, work mode switched to backup around 1am, backup mode is used for freeze charge and hold charge. 

Setting  template flag to on in apps.yaml appears to have stopped the unexpected work mode changes.

Note that the plan HTML looks odd with rounded cells, I think I have seen that before when predbat was crashing during proceeding, however have not spotted a crash in the logs yet. 

Will attach logs etc in a while. 

**Expected behaviour**
Inverter work mode to not change in read only mode

**Predbat version**

8.18.2

**Environment details**

- Fox KH7, ECS4300 H4 battery
- add on, fresh HAos 15.2 upgrade
- Anything else?

**Screenshots**
To follow

**Log file**
To follow

**Predbat debug yaml file**
To follow

Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-04-20:
I don't see anything in the log to suggest controls are being changed. 

Can you try the latest version?

WyndStryke, 2025-04-20:
@springfall2008 
Hi Trefor,

I rolled back HAos 15.2, and I've been keeping up with the latest versions for a few days.  No re-occurrence since then and everything has been behaving.  So it was either a random glitch, or an interaction between HAos and the modbus integration.  I wanted to run for a few days to confirm.  4 days is probably long enough.

Cheers

closed 2025-04-20 by WyndStryke