springfall2008/batpred #2255: 8.18.3 has completely broken table card

pacemaker82, 2025-04-16:
Hi @springfall2008 

I am not sure what HTML change you made in 8.18.3 but it has completely broken the table card I made.

[This fix](https://github.com/springfall2008/batpred/pull/2253) seems to have been the culprit

Now my card has all the columns out of whack - I can't really figure out why yet.

[image]

You also seem to have removed the header information from the top of the HTML plan, so no last updated or other info. But separate less urgent issue...

Can you revert this change or describe exactly what you changed so I can look into it?

pacemaker82, 2025-04-16:
OK I figured it out

colspan = 2 on the state header cell was unexpected, and not what used to be there. All my data collection assumed it didn't have that, so threw out my calculations (I used indexes and number of columns was now different in the table header).

Second change I dont see here in latest version, but did affect my code was the table headers are now <th> and used to be <td> previously. Maybe that happened in the last few updates you made and only manifested when I needed to see car column data.

gcoan, 2025-04-16:
> Second change I dont see here in latest version, but did affect my code was the table headers are now <th> and used to be <td> previously

this was a change in 8.18.3

I tend to [look at the release details](https://github.com/springfall2008/batpred/compare/v8.18.2...v8.18.3) so I can see what has actually changed

Whack-a-mole again

[image]

pacemaker82, 2025-04-16:
8.18.3 only references the colspan change https://github.com/springfall2008/batpred/commit/464897b0d7f48aa736bd4928170843ef93df017f

I guess the `th` change was 8.18.x and I just didn't pick it up until now. That didn't break the columns like we have today, that was definitely caused by the colspan change. Anyway I have rushed out a fix and will figure out backwards compatibility tomorrow

gcoan, 2025-04-16:
> 8.18.3 only references the colspan change 464897b
> 
> I guess the th change was 8.18.x

the th change was 8.18.3, can see it in the release details if you look at what files are changed (not just the release description):

[image]

ProximusAl, 2025-04-17:
I am seeing the same.  Every row has a Planned Export column even though I'm set to control charge only..

gcoan, 2025-04-17:
> I am seeing the same.  Every row has a Planned Export column even though I'm set to control charge only..

@ProximusAl have you upgraded to the latest table card https://github.com/pacemaker82/PredBat-Table-Card/releases/tag/v1.5.7.12 

this cures the extra exports for me

ProximusAl, 2025-04-17:
> [@ProximusAl](https://github.com/ProximusAl) have you upgraded to the latest table card https://github.com/pacemaker82/PredBat-Table-Card/releases/tag/v1.5.7.12
>
> this cures the extra exports for me

That indeed cured my issue....now I wonder why HACS didnt inform me an update was availabe? 
I was many verison out.

Thank you for letting me know...

springfall2008, 2025-04-17:
Sorry about that, I do wonder if it would be better to create a JSON output for use with the plan card otherwise it will keep breaking every time the HTML is updated?

pacemaker82, 2025-04-17:
@springfall2008 something was worked on related to that https://github.com/springfall2008/batpred/pull/1288 - I am not sure if its ready for primetime yet though but I agree - JSON would be 100x safer.

pacemaker82, 2025-04-17:
To close this one out, I added a new release today that supports both the old and new way of displaying the original plan HTML

https://github.com/pacemaker82/PredBat-Table-Card/releases/tag/v1.5.7.14

gcoan, 2025-04-17:
> I wonder why HACS didnt inform me an update was available?

HACS only polls for changes once every day or so,  so you have to manually tell it to redownload

gcoan, 2025-04-17:
> Sorry about that, I do wonder if it would be better to create a JSON output for use with the plan card otherwise it will keep breaking every time the HTML is updated?

can we look at this @springfall2008 eitehr #1288 or your own json structure.  It was suggested ages ago as being the way forward and would stop the wack-a-mole with the table card and predbat

springfall2008, 2025-04-18:
Heads up I made a new release with a total row at the bottom, hopefully that won't break anything

SwiftRR, 2025-04-18:
Now on v8.18.5 and Table card is broken for me with this error.
I missed v8.18.4 so updated straight from v8.18.3 to .5

[image]

Rob

pacemaker82, 2025-04-18:
> Now on v8.18.5 and Table card is broken for me with this error.

@SwiftRR please update to latest https://github.com/pacemaker82/PredBat-Table-Card/releases/tag/v1.5.9.5 - this issue will be resolved in there.

pacemaker82, 2025-04-18:
> Heads up I made a new release with a total row at the bottom, hopefully that won't break anything

@springfall2008 it did unfortunately but I fixed it in an update quickly. I had made an assumption you always set a <TD> background colour as that was consistent in the rest of the table rows - but in this new row you didn't do that, which broke my code when I try to read and convert the colour. Anyway fixed now.

SwiftRR, 2025-04-18:
@pacemaker82 
Well done with such a rapid response.
The table card is now working on Mac OSX and on iOS mobile devices.

Rob

closed 2025-04-19 by springfall2008