springfall2008/batpred #2257: DIscharge happening very early in the morning, risking running out of battery

campbellcopeland, 2025-04-17:
**Describe the bug**
After the offpeak period, there seems to be a shift as of v8.18.x to discharge a lot of the battery early in the day, increasing hte risk of running out of battery if the solar prediction is not accurate. I have an AC couples AIO with the hybrid option disabled so don't need to discharge to prevent clipping.

**Expected behaviour**
The old behaviour of exporting from 11pm onwards worked well

**Predbat version**
v8.18.x

**Environment details**

- GivEnergy single AIO & Gateway
- Standard HAOS installer

**Screenshots**
to be attached

**Log file**
to be attached

[predbat_log.txt](https://github.com/user-attachments/files/19794350/predbat_log.txt)

**Predbat debug yaml file**
will need to enable and check tomorrow if early export happens again - has done for past two+ days though.

gcoan, 2025-04-17:
its worth you looking at the clipping column in the plan, Predbat may be exporting in an attempt to reduce clipping ?

although for AC coupled AIO this wouldn't make any difference, but worth ruling it out

campbellcopeland, 2025-04-17:
Debug enabled for early morning export  planned for 18th

[predbat_debug_14_05_00.yaml.txt](https://github.com/user-attachments/files/19794853/predbat_debug_14_05_00.yaml.txt)

[image]

campbellcopeland, 2025-04-17:
> its worth you looking at the clipping column in the plan, Predbat may be exporting in an attempt to reduce clipping ?
> 
> although for AC coupled AIO this wouldn't make any difference, but worth ruling it out

I don't have that column - I've got 8kW of microinverters, 6kW AIO and 14.2kW G99 so not clipped at any point

ma9mwah, 2025-04-17:
I've had something similar for quite a while now. I would expect predbat to move the dump during the day to as late as possible so that there wasn't a possibility of running out of battery in the evening before the cheap rate at 00:30

[image]

campbellcopeland, 2025-04-18:
Additional files from this morning where discharge is happening before sun rise

[predbat.log](https://github.com/user-attachments/files/19808492/predbat.log)
[apps.yaml.txt](https://github.com/user-attachments/files/19808491/apps.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19808489/predbat_debug.yaml.txt)
[predbat_plan.html.txt](https://github.com/user-attachments/files/19808490/predbat_plan.html.txt)

[image]

springfall2008, 2025-04-18:
New release to try to fix this issue

campbellcopeland, 2025-04-18:
> New release to try to fix this issue

Thanks - will check tomorrow morning but plan so far looks good 👍

campbellcopeland, 2025-04-19:
Looks fixed, export now happening in the evening

[image]
[image]
[image]

springfall2008, 2025-04-19:
Great thanks

closed 2025-04-19 by springfall2008