springfall2008/batpred #2265: Dashboard apps.yaml lists an invalid non-exixtent sensor

nbullus, 2025-04-18:
**Describe the bug**
Apps.yaml on Predbat dashboard reports non-existent sensor

**Expected behaviour**
Dashboard Apps.yaml should reflect the actual content of the file.

**Predbat version**

8.18.2

**Environment details**

- GE AIO
- Standard HAOS installer

**Screenshots**

The following screenshot shows an incorrect sensor reported under 'scheduled_discharge_enable'

[image]

Apps.yaml in File Editor shows that this contains the correct switch entity and not a sensor as reported on the dashboard:

[image]

**Log file**

Not needed.

**Predbat debug yaml file**

Not applicable.

gcoan, 2025-04-18:
This was raised before. 

It seems predbat doesn't use the scheduled discharge enable in givtcp so it can be removed from your apps.yaml

nbullus, 2025-04-19:
> This was raised before.
> 
> It seems predbat doesn't use the scheduled discharge enable in givtcp so it can be removed from your apps.yaml

Cheers this has gone now. Might be worth removing in the next round of documentation updates as its still listed.

nbullus, 2025-04-19:
You can close this issue down.

gcoan, 2025-04-19:
> You can close this issue down.

I'll allocate it to me to do the template documentation update

cheers

kiisukene, 2025-04-20:
Thanks guys - I just discovered this is v8.18.7 and simply commenting out the "scheduled_discharge_enable" entry in apps.yaml worked for me too

gcoan, 2025-05-13:
removed scheduled discharge enable from givtcp template

closed 2025-05-13 by gcoan