springfall2008/batpred #2267: FoxESS setting Force Charge for "Hold Charging"

AndySmithers, 2025-04-19:
I saw this behaviour yesterday but didn't have debug on.
Today I watched the problem happening in real time.

As background, I have implemented all of Tony Hoyle's templates and automations for the Fox H1 G2 inverter.
All the functions of BottleCapDave's Fox integration work perfectly.

The behaviour I saw this morning is as follows:

Inverter was in Force Charge mode until 8:30Am (taking advantage of a cheap-ish Agile slot and lack of sun forecast).
The slot after 8:30 was marked as a "Freeze Charge" slot.
At 8:30, Predbat came to life and, after a minute or so, changed the 8:30 slot to "Hold Charge" (presumably because the charging had taken the SOC slightly higher than planned?)
Now, I have the entity for the FoxESS workmode selector visible on the PredBat page.
What I saw was the Workmode selector first changed to the (correct) "Back-Up" mode.
Then, a few 10s of seconds later changed back to "Force Charge", where it remained after Predbat had finished.

My guess is that "Freeze Charge" is mapped to the correct Fox function (Back-Up) but "Hold Charge" is not but I'm not sure.

This integration setup has been running fine for a few weeks but it's been sunny and so I haven't seen many (or even any) Force Charge cycles.

I'm attaching the last bit of the relevant log file (after all the calculations) and my apps.yaml.

[FoxESSLog.txt](https://github.com/user-attachments/files/19820228/FoxESSLog.txt)

[apps.yaml.txt](https://github.com/user-attachments/files/19820233/apps.yaml.txt)

springfall2008, 2025-04-19:
Okay, can you try the version on 'main' where I have included a fix for this, and I also rolled the Fox inverter settings into the Predbat config.py

AndySmithers, 2025-04-19:
[image]

springfall2008, 2025-04-19:
Fix pushed to main, please try again

AndySmithers, 2025-04-19:
Yep, that fixes the Exception.
It's difficult to test if it fixes the original problem but a scan through the differences in the python files suggest to my (very amateur!) analysis that it will!

Fantastic responses, Trefor - and on a Saturday evening!

springfall2008, 2025-04-19:
Let me know how you get on, I'll include this in the next release if its not totally broken

AndySmithers, 2025-04-20:
Looks OK so far. I've tried all the Force manual options and they work fine.
Of course, what I can't actually simulate is the case where it fell over before, charging to a target SOC where the next calculation cycle saw an SOC higher than the target.

closed 2025-05-02 by springfall2008