springfall2008/batpred #2268: 'Hold for car' not working when Pred bat status is 'hold charging target'.

dpe926, 2025-04-19:
**Describe the bug**
When 'hold for car' is enabled and car charging is in progress the battery output should reduce to as near zero as possible.
This always used to be the case however recently I have noticed that when the predbat mode is 'hold charging target' and the car charges, the battery responds to meet it with its maximum output. I have checked that when the predbat mode is 'demand' and the car charges the battery output is reduced as expected. 
To check this I forced the car charging indictor I have in HA to 'on' during demand mode and checked the battery output, and then did the same during a predbat 'hold charging target' This confirmed what I had seen during the previous two evenings

**Expected behaviour**
When 'hold for car' is enabled and car charging is in progress the battery output should reduce to as near zero as possible.

**Predbat version**
8.18.5

**Environment details**

- Inverter AC3 software D0.210-A0.210
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-04-19:
I probably need more details here, maybe a logfile?

dpe926, 2025-04-20:
OK, I have repeated the test last night. At 23:00 (predbat demand mode) I forced the car charging indicator to charging, and the battery output dropped to near zero, I then returned the car charging indicator to its normal setting at about 23:05 and the battery supplied the house load. 
At 01:00 the car started to charge, the indicator changed to  car charging and the battery discharged at is maximum rate. I downloaded the logfile below at about 01:30

[predbat_debug (1).yaml.txt](https://github.com/user-attachments/files/19824429/predbat_debug.1.yaml.txt)

[image]

springfall2008, 2025-04-20:
I think I found the issue and a fix has been made on main

dpe926, 2025-04-20:
Great, thanks so much, will give it a go

closed 2025-04-22 by dpe926