springfall2008/batpred #2271: 48 hour plan in reality much less

pacemaker82, 2025-04-19:
In apps.yaml I have set for 48 hours forecast:

```
  # Number of hours forward to forecast, best left as-is unless you have specific reason
  forecast_hours: 48
```

Unless I misunderstand this property, I am expecting a 48 hour plan (from NOW) in the predbat plan. 

Instead, I see today, and "most" of tomorrow. Most of the time my plan ends at 23:00 tomorrow

[image]

Sometimes I see the full day tomorrow, including the final 23:30 slot, it randomly appears and disappears.

**Expectation:**
I am expecting a plan with 96 half hour slots if I set it to 48 hour plan... ?

Predbat Version 8.18.6

pacemaker82, 2025-04-19:
If I set the apps YAML to

```
  # Number of hours forward to forecast, best left as-is unless you have specific reason
  forecast_hours: 24
```

I correctly get exactly 24 hours

[image]

So seems my expectations of a 48 hour plan are correct?

springfall2008, 2025-04-19:
There's a setting called 'input_number.predbat_forecast_plan_hours' in Predbat which defaults to 24, this means the plan will be 24 hours from the next charge period, to align with the cycle of energy rates. If you change this to 48 (while leaving forecast_hours in apps.yaml as 48) you will get a complete plan. The reason to cut it short however is to ignore end cases where you dump your battery for the "end of the world"....

pacemaker82, 2025-04-20:
I wont pretend to understand situation where YAML is 48 and config is 24 being a random "tomorrow" cutoff (23:00 or 23:30), but yes it works if I set both YAML and config to the same numbers at least, thanks.

closed 2025-04-20 by pacemaker82