springfall2008/batpred #2275: nordpool per 15 minutes as of june 11th

eddysteurs, 2025-04-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
nordpool pricing will per 15 minutes a of june 11th, will predbat be following this
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
predbat pricing for nordpool per 15 min

gcoan, 2025-04-20:
There would be no benefit to Predbat moving to 15 minute nordpool predicted prices as the energy suppliers (e.g. Octopus Agile) price on 30 minute periods.

It might be that Predbat can average the two 15 minute slots to produce a blended 30 minute slot price

Or it might be that the predicted prices don't work when nordpool change to 15 minute granularity

springfall2008, 2025-04-21:
I think at the very least Predbat will need to average the new datapoints, but I can't do this until the change occurs

kukumagi, 2025-05-01:
There is a demo endpoint for 15min resolution. https://data.nordpoolgroup.com/auction/day-ahead/price-indices?deliveryDate=2025-05-02&currency=EUR&resolutionInMinutes=15&indexNames=EE
The returned json can be seen here: https://dataportal-api.nordpoolgroup.com/api/DayAheadPriceIndices?date=2025-05-02&market=DayAhead&indexNames=EE&currency=EUR&resolutionInMinutes=15

d01mf, 2025-06-19:
> There would be no benefit to Predbat moving to 15 minute nordpool predicted prices as the energy suppliers (e.g. Octopus Agile) price on 30 minute periods.

From what I understand the EU does mandate that electricity markets (day-ahead, intraday, and balancing) operate on 15-minute intervals. However, extending that to household-level billing and metering is not currently mandatory across the board, but strongly encouraged and planned for in several EU directives and national implementations.

Even if Octopus Agile stick to 30 minute pricing for now, others are likely to jump to 15 minutes sooner rather than later. Having the interval configurable would be beneficial in supporting all users.

gcoan, 2025-06-19:
I suggest that there is little benefit in extending predbat price granularity to 15 minutes at this point in time.

The big dependency is getting smart meters to record consumption every 15 minutes instead of every 30 minutes. Looks like whilst the UK has had businesses on 30 minutes readings, for consumers its been optional https://www.ofgem.gov.uk/blog/how-half-hourly-settlement-will-help-cut-energy-bills  and the UK is only just starting to standardise on every meter recording 30 minute data - that rollout won't complete until 2026 https://www.tritility.com/uncategorized/the-market-wide-half-hourly-settlement-which-meter-should-your-business-choose/#:~:text=options%20are%20available.-,What%20is%20the%20Market%2DWide%20Half%2DHourly%20Settlement?,switch%20to%20a%20new%20meter. 

Ofgem made a cost benefit analysis of moving to 15 minute settlement in 2020 and decided not to move due to it costing a lot https://www.ofgem.gov.uk/decision/decision-grant-exemption-applying-15-minute-imbalance-settlement-period-gb-synchronous-area

There may be more recent plans in the UK, couldn't find anything.
Situation may be different in other countries of course.

eddysteurs, 2025-10-04:
i noticed a problem this morning with predbat probably because of the 15 min price change
Because of negative prices I had entered a manual charge period from 01:00 till 07:00 hours (predbat did not plan any charge for today)
But I noticed the following

changed to Charging
05:01:24 - 5 hours ago - Supervisor
changed to Demand
04:46:31 - 6 hours ago - Supervisor
changed to Charging
04:31:25 - 6 hours ago - Supervisor
changed to Demand
03:46:45 - 7 hours ago - Supervisor
changed to Charging
03:01:32 - 7 hours ago - Supervisor
changed to Demand
02:46:41 - 8 hours ago - Supervisor
changed to Charging
02:31:29 - 8 hours ago - Supervisor
changed to Demand
02:16:34 - 8 hours ago - Supervisor
changed to Charging
02:02:26 - 8 hours ago - Supervisor

So it seems it starts each time charing on the hour and half hour but switches to demand after 15 min.
Never had this before so my guess is because the sensor state of nordpool gives another price for the next 15 min period.
I stopped predbat at 05:10 so my battery would charge as planned, restarted this morning with another price provides (Frank Energies which for this months still gives hourly prices but is switching on Nov 1st also to 15 min)

eddysteurs, 2025-10-07:
I found a fix on the nordpool integration (custom) to put in some code for writing the nordpool sensor again as an hourly source (averaging the 4 quarter values), it is a dirty fix direct in the code on members individual installations so it is definitely not a good permanent one, there is a request to get this with a switch in the code but not sure if this is coming, maybe this could be done in predbad and then on a 30 min timeframe as predbat works with half hours already, so we could at least profit partially form the price differences