springfall2008/batpred #2284: Battery Prediction and Actual Charge very different

discodancerstu, 2025-04-22:
I'm unsure if this is a bug or setup issue, but grateful for some guidance please.

Setup - dual 3.6 inverters and dual 9.5 batteries
GivTCP - v3 main
Octopus tarrif - Go/Fixed

The Home Battery Prediction always looks like this:

[image]

And is as I would expect; charge the batteries to full during the cheap import (0030 - 0530) up to 100% and then discharge throughout the day.  Finally, an export with anything left before midnight.

However, when looking at the battery SOCs over the last week, at no point have the batteries got to above 65%, and they certainly are not charged to max during the cheap early morning period even though PredBat has this as the prediction.

[image]

I just can't get my head around this, have I got it setup incorrectly?  Apps.yaml attached

[Apr25apps.yaml.txt](https://github.com/user-attachments/files/19851058/Apr25apps.yaml.txt)

gcoan, 2025-04-22:
The only obvious thing I see in apps.yaml is you have inverter_limit set to 5000 not 3600 for your inverters

Would be worth creating a battery charge curve to improve the accuracy of the predbat plan, but otherwise looks OK.

Can you capture a debug file please and add it to the issue

discodancerstu, 2025-04-22:
Apologies, the original post was slightly wrong, I have dual 5k inverters, so believe apps.yaml is correct for 2 inverters.

Can you advise on how to create the battery charge curve to improve the plan please? 

Debug attached. 

[predbat_debug.bin.txt](https://github.com/user-attachments/files/19856116/predbat_debug.bin.txt)

gcoan, 2025-04-23:
> Apologies, the original post was slightly wrong, I have dual 5k inverters, so believe apps.yaml is correct for 2 inverters.

Yes that is correct

> Can you advise on how to create the battery charge curve to improve the plan please?

Details of how to create the charge curve is in the documentation https://springfall2008.github.io/batpred/apps-yaml/#battery-chargedischarge-curves

You need to ensure that 3 sensors are available to predbat, and restart predbat, and it should determine the curve and write the details to the logfile

discodancerstu, 2025-04-23:
Thanks. 

Once again last night, Predbat plan was for battery to charge to full overnight, but it did not. 

Yes I will create the charge curve, but will this really make the big change? 

What I can't understand is why the plan has the battery to charge to full, but it then does not. 

Can you explain please? 

[image]
[image]

discodancerstu, 2025-04-25:
> Details of how to create the charge curve is in the documentation https://springfall2008.github.io/batpred/apps-yaml/#battery-chargedischarge-curves
>
> You need to ensure that 3 sensors are available [...]

Hi Geoffrey,

Thanks for the info.  My logfile does not have the battery charge curves within.  I am using REST and the appropriate entities are uncommented as per the documentation, please correct me if I have anything wrong.

Grateful if you can advise what I'm missing please?

[25Apr apps.yaml.txt](https://github.com/user-attachments/files/19911834/25Apr.apps.yaml.txt)

2025-04-25 15:51:21.590250: Find charge curve with sensors sensor.givtcp_inv1__soc_kwh and number.givtcp_inv1__battery_charge_rate and predbat.status and sensor.givtcp_inv1__battery_power
2025-04-25 15:51:41.888737: Find charge curve has 8.0 days of data, max days 8
2025-04-25 15:51:41.948511: Note: Cannot find battery charge curve (no final curve), one of the required settings for predbat.status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml

[25Apr predbat.log.txt](https://github.com/user-attachments/files/19911851/25Apr.predbat.log.txt)

closed 2026-08-28 by chalfontchubby