springfall2008/batpred #2287: Predbat load scaling

Tristram69, 2025-04-23:
Need to adjust load scaling so that guaranteed SOC of 60% between hours of 22,00 and 00.00 please. 

[predbat_debug (1).txt](https://github.com/user-attachments/files/19875903/predbat_debug.1.txt)

gcoan, 2025-04-23:
could you explain a bit more what you are trying to do and why you need this specific SoC level

Predbat doesn't really work to specific target SoC's, it will manage your battery charging and discharging based on solar and house load predictions so that it minimises your electricity spend (income-export) and makes best use of your battery.  Neither charging too much nor too little to meet your house load.

Tristram69, 2025-04-24:
I have quite a bit of electric demand during the night from my property . Based on manual timer reads I set before installing Predbat 55/60% SOC between my off peak times of 22.00 to 00.00 got me through the night without going majorly low on my SOC. Since install of Predbat I find that it will charge at between these times but slightly too low. This therefore means my battery depletes to minimum level forcing my house to use peak electric before my next off peak period (04.00-07.00) . This in turn causes my batteries to unbalance.
I do notice in my Predbat plan though that even if it states SOC level to achieve by a certain time my batteries don’t quite make it to that level

springfall2008, 2025-04-24:
If I reverse your load scaling setting (back to 1.0) and re-run your plan then it still says you have plenty of energy left only going to 38% minimum.

Are you saying the max charge rate you set of 4.6kw is actually incorrect, maybe you need to adjust that so the predictions are accurate?

[image]

Tristram69, 2025-04-24:
Yes this appears to be the case . Despite it being a 3.6ke Sunsynk eco inverter it can charge my batteries at a higher rate using solar and grid electric combined

springfall2008, 2025-04-24:
Check in Home Assistant how fast the battery actually charges

springfall2008, 2025-04-24:
Oh wait maybe this is the issue:

  inverter_limit:
  - 4600

Should be 3600, and maybe battery_rate_max is correct at 4600 when charging from DC?

springfall2008, 2025-04-24:
I helped to change the inverter_limit to 3600 to match the real rate, set load scaling back to 1.1 and best_soc_keep to 3 kWh

closed 2026-08-28 by chalfontchubby