springfall2008/batpred #2288: changed to ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType'

eddysteurs, 2025-04-24:
**Describe the bug**
A clear and concise description of what the bug is.
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.4.3
 Home Assistant Supervisor: 2025.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[16:33:07] INFO: Configuring timezone (Europe/Brussels)...
s6-rc: info: service base-addon-log-level successfully started
Predbat files are installed correctly for version v8.18.10
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
**** Starting Standalone Predbat ****
2025-04-24 16:33:07.870819: Loading apps.yaml
2025-04-24 16:33:07.915613: Info: Add-on slug is 6adb4f0d_predbat
2025-04-24 16:33:07.927557: Info: Connected to Home Assistant at http://supervisor/core
2025-04-24 16:33:07.927703: Creating task: <coroutine object HAInterface.socketLoop at 0x7f0fd34e4e50>
2025-04-24 16:33:07.929358: Info: Start socket for url http://supervisor/core/api/websocket
2025-04-24 16:33:07.929392: Info: Web Socket task started
2025-04-24 16:33:07.930511: Creating task: <coroutine object WebInterface.start at 0x7f0fd354e7a0>
Web interface started
2025-04-24 16:33:07.949738: Info: Web Socket active
Watching ['./execute.py', './solcast.py', './web.py', './octopus.py', './predbat.py', './userinterface.py', './gecloud.py', './compare.py', './energydataservice.py', './apps.yaml', './utils.py', './inverter.py', './unit_test.py', './hass.py', './ha.py', './predheat.py', './output.py', './fetch.py', './alertfeed.py', './prediction.py', './download.py', './db_manager.py', './config.py', './plan.py', './futurerate.py'] for changes
2025-04-24 16:34:52.056590: Info: record_status Charging target 38%-41%
2025-04-24 16:35:36.131502: Info: record_status Charging target 38%-41%
2025-04-24 16:40:33.340725: Info: record_status Charging target 38%-41%
2025-04-24 16:46:38.391777: Info: record_status Charging target 40%-41%
2025-04-24 16:50:34.993722: Info: record_status Charging target 40%-41%
2025-04-24 16:55:41.678410: Info: record_status Hold charging target 41%-41%
2025-04-24 17:01:47.996166: Info: record_status Hold charging target 43%-41%
2025-04-24 17:05:33.367946: Info: record_status Hold charging target 43%-41%
2025-04-24 17:10:33.914764: Info: record_status Hold charging target 43%-41%
2025-04-24 17:16:30.076998: Info: record_status Hold charging target 44%-41%
2025-04-24 17:20:33.060691: Info: record_status Hold charging target 44%-41%
2025-04-24 17:25:34.446186: Info: record_status Hold charging target 44%-41%
2025-04-24 17:31:44.320792: Info: record_status Hold charging target 44%-43%
2025-04-24 17:35:34.813735: Info: record_status Hold charging target 44%-43%
2025-04-24 17:40:35.138325: Info: record_status Hold charging target 44%-43%
2025-04-24 17:48:08.660794: Info: record_status Hold charging target 44%-43%
2025-04-24 17:51:02.349611: Info: record_status Hold charging target 44%-43%
2025-04-24 17:56:05.502277: Info: record_status Hold charging target 44%-43%
2025-04-24 18:02:47.383424: Info: record_status Demand
2025-04-24 18:05:33.463495: Info: record_status Demand
2025-04-24 18:10:35.078900: Info: record_status Demand
2025-04-24 18:16:48.065402: Info: record_status Demand
2025-04-24 18:20:37.595909: Info: record_status Demand
2025-04-24 18:25:39.043166: Info: record_status Demand
2025-04-24 18:31:37.959795: Info: record_status Demand
2025-04-24 18:35:37.676479: Info: record_status Demand
2025-04-24 18:40:39.024229: Info: record_status Demand
2025-04-24 18:40:48.206387: ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType'
2025-04-24 18:40:48.228179: Info: record_status ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType'
2025-04-24 18:40:48.228288: Error: '<' not supported between instances of 'float' and 'NoneType'
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predheat.py", line 660, in run_time_loop
    raise e
  File "/config/predheat.py", line 656, in run_time_loop
[... 275 lines cut ...]
A clear and concise description of what you expected to happen.

**Predbat version**

V 8.18.10

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?
  HAOS
**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

If I disable predheat, the error goes away and predbat runs normal

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/19896746/predbat_debug.yaml.txt)

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

springfall2008, 2025-04-27:
What is 'flow_temp' set to in apps.yaml in the Predheat section?

eddysteurs, 2025-04-27:
45
Eddy Steurs

springfall2008, 2025-04-28:
Can you share your apps.yaml?

gcoan, 2025-04-28:
@eddysteurs you will have to go onto github in the website or app, you can't attach apps.yaml via an email reply

closed 2026-08-20 by chalfontchubby