springfall2008/batpred #2289: Predbat 8.18.* occasionally doesn't wait for cheap charging slots after peak export ends

kamenmackay, 2025-04-24:
**Describe the bug**
Predbat 8.18.* charges the battery right after peak export rather than waiting for cheap slot.

**Expected behaviour**
With versions previous to 8.18.*, it would wait for the cheap slot from 2am - 530 (Octopus Flux) to charge the battery as needed

**Predbat version**

8.18.10

**Environment details**

- Givenergy 3.6 kw inverted Gen2 with 9.5 kwh battery
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**

[image]

[predbat_debug_20_30_00.yaml.txt](https://github.com/user-attachments/files/19898396/predbat_debug_20_30_00.yaml.txt)

[image]

springfall2008, 2025-04-26:
There's something odd here, the debug yaml is corrupted in that its missing the energy rate data, can you try to re-generate it again?

springfall2008, 2025-04-26:
Looking at this the target target is 6% meaning its not charging?

kamenmackay, 2025-04-26:
Hmm, I reverted 8.17.6 but I suppose I can reinstall 8.18.10 to help out. I'll do that and re-generate the yaml if I'm lucky enough to catch it in time.

kamenmackay, 2025-04-28:
@springfall2008 So when I woke up this morning, Predbat was charging from the grid at 06:30 and it looked like it planned to do that all day from the brief glance I took at it (forgot to capture the plan). I quickly changed it to demand only until I got back and also generated the debug file which is attached here.
I suppose I should say the expected behaviour (as per pre 8.18.*) was for the battery to charge to a certain SOC during the cheap Octopus Flux hours and then let solar do the rest of the work to get it up to 100% by 1600 in time to export it when the export rate was higher.

[debug_yaml_0428.txt](https://github.com/user-attachments/files/19935856/debug_yaml_0428.txt)

kamenmackay, 2025-04-29:
8.18.10 charged the battery to 100% (low power charge) during the cheap slots with an expected day of 22 kwh ahead of it. I anticipated that it would maintain a charge of 11% like it did with 8.17.* and then use solar to charge the battery to 100% using solely solar.

gcoan, 2025-04-29:
> 8.18.10 charged the battery to 100% (low power charge) during the cheap slots with an expected day of 22 kwh ahead of it. I anticipated that it would maintain a charge of 11% like it did with 8.17.* and then use solar to charge the battery to 100% using solely solar.

What is the rate for cheap rate charging and what is the export rate you have?  Can you share the logfile please

kamenmackay, 2025-04-29:
I'm on Octopus Flux so the cheap rate is 2am - 5am at @ £0.1549 and the export rate from 5am - 4pm is £0.0955. 
Let me know if you need anything else.
[predbat_20250429.log](https://github.com/user-attachments/files/19954902/predbat_20250429.log)

springfall2008, 2025-05-12:
I believe this is fixed now in the latest release, I ran your debug YAML and there was no charging

[image]

kamenmackay, 2025-05-12:
I haven't seen this behavior as of recent versions so thanks!

closed 2025-05-12 by kamenmackay